gazebosim/gz-sim #1871: Failing to import plugins in gazebo 7.2.0

Rezenders, 2023-01-27:
## Environment
* OS Version: Ubuntu 22.04
* Source: Gazebo Sim, version 7.2.0

## Description
After upgrading ubuntu 22.04, gz sim got upgraded to version 7.2.0 and now it is falling to import plugins.

I get the following error:
```
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-hydrodynamics-system] : couldn't find shared library.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-thruster-system] : couldn't find shared library.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-thruster-system] : couldn't find shared library.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-thruster-system] : couldn't find shared library.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-thruster-system] : couldn't find shared library.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-thruster-system] : couldn't find shared library.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [gz-sim-thruster-system] : couldn't find shared library.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:78] Failed to load system plugin [libArduPilotPlugin.so] : couldn't find shared library.
```

It seems that I need to go back to using `ignition-gazebo` and `ignition::gazebo` instead of `gz-sim` and `gz::sim`, respectively.
And to find custom plugins I need to back to `IGN_GAZEBO_SYSTEM_PLUGIN_PATH` instead of `GZ_SIM_SYSTEM_PLUGIN_PATH`. However, the custom plugin I am using doesn't work when I do this.

I get the following error:
```
[ruby $(which ign) gazebo-1] Library [/home/gus/ardupilot_gazebo/build/libArduPilotPlugin.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.
[ruby $(which ign) gazebo-1] [Err] [SystemLoader.cc:87] Failed to load system plugin [libArduPilotPlugin.so] : couldn't load library on path [/home/gus/ardupilot_gazebo/build/libArduPilotPlugin.so].
```

Is this behavior desirable? Should we go back to using ignition names? It doesn't seem like this is desirable.

## Steps to reproduce

Install this [repo](https://github.com/kas-lab/suave) and its deps (including [ardupilot_gazebo](https://github.com/ArduPilot/ardupilot_gazebo))
And run:
```
ros2 launch suave simulation.launch.py
```

azeey, 2023-01-27:
I was able to install ardupilot_gazebo and run the examples without a problem using a ubuntu 22.04. Installing all of the `suave` seems quite involved. Would you be able to provide a minimal example showing the problem you're facing?

Rezenders, 2023-01-28:
Hello, thanks for the reply.

> Installing all of the suave seems quite involved. Would you be able to provide a minimal example showing the problem you're facing?

Yes, I added a launch file called `simulation.launch.py` to this [repo](https://github.com/Rezenders/bluerov2_ignition). You only need to build this individual repo to test it, by running:

```
ros2 launch bluerov2_ignition simulation.launch.py
```

I tested starting the simulation directly with `gz sim underwater.world -v3` and it works. So I guess the problem is actually related to `ros_gz_sim`.

By the way, I am using ROS humble, and I am building `ros_gz` from source with the humble branch commit [74dcf50](https://github.com/gazebosim/ros_gz/commit/74dcf50afc0c487e0d6ef5da8c412ddb14762b30).

azeey, 2023-01-30:
Okay, I tested running `simulation.launch.py` and everything seems to work for me. My suspicion is you have some ABI compatibility issues. So here are some follow-up questions:

* Do you have `ros-gz` packages installed from debs that might be interfering? 
* Is `ros_gz` built with `GZ_VERSION="garden"` environment variable set?
* Can you run `ldd /home/gus/ardupilot_gazebo/build/libArduPilotPlugin.so` and see which `gz` libraries it's linked against?

Rezenders, 2023-01-31:
> Is ros_gz built with GZ_VERSION="garden" environment variable set?
This did the trick. Now it works. Thanks.

Rezenders, 2023-01-31:
Actually, now the simulation starts but it is crashing when I run ardupilot which connects to the libArduPilotPlugin, but I don't know if it is related to this issue anymore.

the output of ldd /home/gus/ardupilot_gazebo/build/libArduPilotPlugin.so

```
linux-vdso.so.1 (0x00007ffc7c7f6000)
	libgz-sim7.so.7 => /lib/x86_64-linux-gnu/libgz-sim7.so.7 (0x00007ff4fb91f000)
	libgz-common5.so.5 => /lib/x86_64-linux-gnu/libgz-common5.so.5 (0x00007ff4fb8c2000)
	libgz-transport12.so.12 => /lib/x86_64-linux-gnu/libgz-transport12.so.12 (0x00007ff4fb843000)
	libgz-msgs9.so.9 => /lib/x86_64-linux-gnu/libgz-msgs9.so.9 (0x00007ff4fb4bd000)
	libsdformat13.so.13 => /lib/x86_64-linux-gnu/libsdformat13.so.13 (0x00007ff4fb279000)
	libgz-math7.so.7 => /lib/x86_64-linux-gnu/libgz-math7.so.7 (0x00007ff4fb222000)
	libprotobuf.so.23 => /lib/x86_64-linux-gnu/libprotobuf.so.23 (0x00007ff4faf5f000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007ff4fad35000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007ff4fac4e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007ff4fac2c000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007ff4faa04000)
	libgz-fuel_tools8.so.8 => /lib/x86_64-linux-gnu/libgz-fuel_tools8.so.8 (0x00007ff4fa97f000)
	libgz-plugin2-loader.so.2 => /lib/x86_64-linux-gnu/libgz-plugin2-loader.so.2 (0x00007ff4fa968000)
	libgz-plugin2.so.2 => /lib/x86_64-linux-gnu/libgz-plugin2.so.2 (0x00007ff4fa95c000)
	libgz-common5-events.so.5 => /lib/x86_64-linux-gnu/libgz-common5-events.so.5 (0x00007ff4fa955000)
	libtinyxml2.so.9 => /lib/x86_64-linux-gnu/libtinyxml2.so.9 (0x00007ff4fa93b000)
	libgz-transport12-parameters.so.12 => /lib/x86_64-linux-gnu/libgz-transport12-parameters.so.12 (0x00007ff4fa919000)
	/lib64/ld-linux-x86-64.so.2 (0x00007ff4fbba0000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007ff4fa910000)
	libzmq.so.5 => /lib/x86_64-linux-gnu/libzmq.so.5 (0x00007ff4fa874000)
	libgz-utils2.so.2 => /lib/x86_64-linux-gnu/libgz-utils2.so.2 (0x00007ff4fa86f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007ff4fa851000)
	libjsoncpp.so.25 => /lib/x86_64-linux-gnu/libjsoncpp.so.25 (0x00007ff4fa81c000)
	libyaml-0.so.2 => /lib/x86_64-linux-gnu/libyaml-0.so.2 (0x00007ff4fa7fb000)
	libzip.so.4 => /lib/x86_64-linux-gnu/libzip.so.4 (0x00007ff4fa7dd000)
	libcurl.so.4 => /lib/x86_64-linux-gnu/libcurl.so.4 (0x00007ff4fa737000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007ff4fa71d000)
	libsodium.so.23 => /lib/x86_64-linux-gnu/libsodium.so.23 (0x00007ff4fa6c5000)
	libpgm-5.3.so.0 => /lib/x86_64-linux-gnu/libpgm-5.3.so.0 (0x00007ff4fa674000)
	libnorm.so.1 => /lib/x86_64-linux-gnu/libnorm.so.1 (0x00007ff4fa549000)
	libgssapi_krb5.so.2 => /lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007ff4fa4f5000)
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007ff4fa4e0000)
	libcrypto.so.3 => /lib/x86_64-linux-gnu/libcrypto.so.3 (0x00007ff4fa09e000)
	libnghttp2.so.14 => /lib/x86_64-linux-gnu/libnghttp2.so.14 (0x00007ff4fa074000)
	libidn2.so.0 => /lib/x86_64-linux-gnu/libidn2.so.0 (0x00007ff4fa053000)
	librtmp.so.1 => /lib/x86_64-linux-gnu/librtmp.so.1 (0x00007ff4fa034000)
	libssh.so.4 => /lib/x86_64-linux-gnu/libssh.so.4 (0x00007ff4f9fc7000)
	libpsl.so.5 => /lib/x86_64-linux-gnu/libpsl.so.5 (0x00007ff4f9fb1000)
	libssl.so.3 => /lib/x86_64-linux-gnu/libssl.so.3 (0x00007ff4f9f0d000)
	libldap-2.5.so.0 => /lib/x86_64-linux-gnu/libldap-2.5.so.0 (0x00007ff4f9eae000)
	liblber-2.5.so.0 => /lib/x86_64-linux-gnu/liblber-2.5.so.0 (0x00007ff4f9e9d000)
	libzstd.so.1 => /opt/ros/humble/lib/libzstd.so.1 (0x00007ff4f9e0e000)
	libbrotlidec.so.1 => /lib/x86_64-linux-gnu/libbrotlidec.so.1 (0x00007ff4f9e00000)
	libmd.so.0 => /lib/x86_64-linux-gnu/libmd.so.0 (0x00007ff4f9df1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007ff4f9dec000)
	libkrb5.so.3 => /lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007ff4f9d21000)
	libk5crypto.so.3 => /lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007ff4f9cf2000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007ff4f9cec000)
	libkrb5support.so.0 => /lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007ff4f9cde000)
	libunistring.so.2 => /lib/x86_64-linux-gnu/libunistring.so.2 (0x00007ff4f9b32000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007ff4f9947000)
	libhogweed.so.6 => /lib/x86_64-linux-gnu/libhogweed.so.6 (0x00007ff4f98ff000)
	libnettle.so.8 => /lib/x86_64-linux-gnu/libnettle.so.8 (0x00007ff4f98b9000)
	libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007ff4f9837000)
	libsasl2.so.2 => /lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007ff4f981a000)
	libbrotlicommon.so.1 => /lib/x86_64-linux-gnu/libbrotlicommon.so.1 (0x00007ff4f97f7000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007ff4f97f0000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007ff4f97dc000)
	libp11-kit.so.0 => /lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007ff4f96a1000)
	libtasn1.so.6 => /lib/x86_64-linux-gnu/libtasn1.so.6 (0x00007ff4f9687000)
	libffi.so.8 => /lib/x86_64-linux-gnu/libffi.so.8 (0x00007ff4f967a000)

```

[... 39 lines cut ...]
[ruby $(which gz) sim-1] #18   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f49314ec492, in 
[ruby $(which gz) sim-1] #17   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f49314efe2d, in 
[ruby $(which gz) sim-1] #16   Object "/usr/lib/x86_64-linux-gnu/libgz-sim7-gz.so.7.2.0", at 0x7f4930a09e35, in runServer
[... cut ...]
[ruby $(which gz) sim-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f4935228475, in raise
[ruby $(which gz) sim-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f493527ca7c, in pthread_kill
[ruby $(which gz) sim-1] Aborted (Signal sent by tkill() 363028 1000)
[ruby $(which gz) sim-1] libEGL warning: DRI2: failed to create dri screen
[ruby $(which gz) sim-1] libEGL warning: DRI2: failed to create dri screen
[INFO] [ruby $(which gz) sim-1]: process has finished cleanly [pid 362952]

```

azeey, 2023-02-03:
That looks like a separate issue. Usually the error `[ruby $(which gz) sim-1] ODE INTERNAL ERROR 1: assertion "aabbBound >= dMinIntExact && aabbBound < dMaxIntExact" failed in collide() [collision_space.cpp:460]` happens when an object has a translation that's too big. Is something accelerating too fast? 

I'll go ahead and close this issue. If you think the ODE ERROR is a bug different from https://github.com/gazebosim/gz-sim/issues/15 or other similar issues already captured in a ticket, feel free to open a new ticket.

closed 2023-02-03 by azeey

Rezenders, 2023-02-06:
Ok, thanks for the reply.

> Is something accelerating too fast?

I don't really know, it is crashing immediately when ArduPilot is connected, I don't know if the plugin I am using might be generating accelerations too fast. My code doesn't even start running before it crashes. Funny thing is that it was working before. I will open an issue in the plugin repo.

srmainwaring, 2023-02-07:
@azeey and @ahcorde - not sure whether this should be reopened or it's for a separate issue. I carried out an investigation in the linked ardupilot_gazebo issue raised above. The quadratic damping calculation changed in the hydrodynamics plugin in https://github.com/gazebosim/gz-sim/commit/fde6ec608bbd1679178c70c62bdc08db41eb38d6. The change (removing an abs() from the state multiplying the coefficients) results in a positive feedback which causes the magnitude of the calculated force-torque to increase without bound given an initial perturbation.

azeey, 2023-02-16:
@srmainwaring probably best if it's a separate issue, but it might not be needed since https://github.com/gazebosim/gz-sim/pull/1888 is open.

gregoriomarchesini, 2025-06-11:
Sorry I also have this question since I am pretty new to gazebo

I am trying to create a custom made plugin. To do this I have forked the gz-sim library locally on my computer and I have created my plugin in the /src/systems/ folder of gz-sim.

Then I have built the code and it seems that everything is recognized. So my build output is

```
-- gz-sim8 version 8.9.0
-- Operating system is Linux
-- Found CPack generators: DEB
-- Looking for display capabilities
--  + found a display available ($DISPLAY is set)
-- Could not find glxinfo. Trying with 'gl-test.py' which may not work.If 'gl-test.py' fails to find DRI, try installng the package 'mesa-utils' for glxinfo
--  ! valid dri display not found (no glxinfo or pyopengl)
-- Looking for sdformat14 -- found version 14.7.0
-- Searching for dependencies of sdformat14
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 10.0.0
-- Looking for gz-math7 -- found version 7.5.1
-- Searching for dependencies of gz-math7
-- Looking for gz-utils2 -- found version 2.2.1
-- Searching for dependencies of gz-utils2
-- Looking for gz-utils2 -- found version 2.2.1
-- Searching for dependencies of gz-utils2
-- Searching for <gz-utils2> component [cli]
-- Looking for gz-utils2-cli -- found version 2.2.1
-- Searching for dependencies of gz-utils2-cli
-- Looking for sdformat14 - found

-- Looking for gz-plugin2 -- found version 2.0.4
-- Searching for dependencies of gz-plugin2
-- Looking for gz-utils2 -- found version 2.2.1
-- Searching for <gz-plugin2> component [loader]
-- Looking for gz-plugin2-loader -- found version 2.0.4
-- Searching for dependencies of gz-plugin2-loader
-- Searching for <gz-plugin2> component [register]
-- Looking for gz-plugin2-register -- found version 2.0.4
-- Searching for dependencies of gz-plugin2-register
-- Looking for gz-plugin2 - found

-- Looking for gz-transport13 -- found version 13.4.1
-- Searching for dependencies of gz-transport13
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq, version 4.3.5
-- Checking for module 'uuid'
--   Found uuid, version 2.39.3
-- Looking for gz-utils2 -- found version 2.2.1
-- Looking for gz-msgs10 -- found version 10.3.2
-- Searching for dependencies of gz-msgs10
-- Looking for gz-utils2 -- found version 2.2.1
-- Looking for gz-math7 -- found version 7.5.1
-- Searching for <gz-transport13> component [log]
-- Looking for gz-transport13-log -- found version 13.4.1
-- Searching for dependencies of gz-transport13-log
-- Searching for <gz-transport13> component [parameters]
-- Looking for gz-transport13-parameters -- found version 13.4.1
-- Searching for dependencies of gz-transport13-parameters
-- Looking for gz-transport13 - found

-- Looking for gz-msgs10 -- found version 10.3.2
-- Looking for gz-msgs10 - found

-- Looking for gz-common5 -- found version 5.7.1
-- Searching for dependencies of gz-common5
-- Looking for gz-utils2 -- found version 2.2.1
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <gz-common5> component [av]
-- Looking for gz-common5-av -- found version 5.7.1
[... 425 lines cut ...]
but neither of them worked for me. The [plugin tutorial](https://gazebosim.org/api/sim/9/createsystemplugins.html) is a bit vague on this point (or better say that I am not expert enough to have it clear).

If you have any hint on how to help me I would be very grateful. 

The structure folder of my plugin looks like

```
src/systems/clohessy_wiltshire_model/
├── ClohesseyWiltshireModel.cc
├── ClohesseyWiltshireModel.hh
└── CMakeLists.txt
```

and the CMakeLists.txt looks like

```
gz_add_system(clohessy-wiltshire-model
  SOURCES
  ClohesseyWiltshireModel.cc
  PUBLIC_LINK_LIBS
    gz-common${GZ_COMMON_VER}::gz-common${GZ_COMMON_VER}
)
```

Thanks in advance for the help!