gazebosim/gz-sim #1879: Perpetually Downloading Resources (Even for empty Worlds)

Bezalelohim, 2023-02-01:
***OS Version: Ubuntu 22.04 LTS***
***Gazebo Garden Version 7.2.0***
***Intel Core i3***
**Integrated Graphic Card**

- The GUI is unreponsive when opened using gz command taking a look using the `verbose` command.
- I found an error log from the terminal. 
- One line keeps repeating, " the requesting list of world names and the server maybe busy downloading them" . 
- Even tried this with a locally downloaded world and same result.

**Measures Taken**
- Tried reopening with and without network connectivity [same result] .
- Reinstalled gazebo garden and ros2 humble multiple times.

**ERROR LOG FROM TERMINAL**
```
joerops@joerops:~$ gz sim empty.sdf -v 4
[Msg] Gazebo Sim GUI    v7.2.0
[Dbg] [gz.cc:162] Subscribing to [/gazebo/starting_world].
[Dbg] [gz.cc:164] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:260] Waiting for subscribers to [/gazebo/starting_world]...
[Wrn] [Gui.cc:268] Waited for 10s for a subscriber to [/gazebo/starting_world] and got none.
[Dbg] [Application.cc:96] Initializing application.
[Dbg] [Application.cc:122] Qt using OpenGL graphics interface
[GUI] [Dbg] [Application.cc:596] Create main window
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Dbg] [PathManager.cc:67] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:241:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
```

azeey, 2023-02-06:
I'm unable to reproduce this error locally. " The server may be busy downloading resources...." message can be misleading because it is printed any time `gz sim` GUI (the client) is trying to communicate with the `gz sim` server and fails. The majority of the time, this communication failure happens because the server is downloading models from Fuel, but that's probably not the case here because you're loading `empty.sdf`.

Is it possible you have other instances of `gz sim` running? In my logs I see both "[Msg] Gazebo Sim GUI " and "[Msg] Gazebo Sim Server", but in yours I only see the former.

Bezalelohim, 2023-02-07:
I have deleted gazebo sim and command gz is not recognized in the terminal now
Also is there a process  I should kill in system monitor/htop ?

Bezalelohim, 2023-02-07:
This seems to be the start of error from error log ```[GUI] [Dbg] [Gui.cc:260] Waiting for subscribers to [/gazebo/starting_world]...

[GUI] [Wrn] [Gui.cc:268] Waited for 10s for a subscriber to [/gazebo/starting_world] and got none.``` Usually the message should be that subsribers should be recieved. And the world (empty.sdf)  just begins loading

Bezalelohim, 2023-02-07:
> Is it possible you have other instances of `gz sim` running? In my logs I see both "[Msg] Gazebo Sim GUI " and "[Msg] Gazebo Sim Server", but in yours I only see the former.

I have made sure to kill all other instances gone deleted all traces of gazebo (to my knowledge) and still the error remains

Bezalelohim, 2023-02-07:
> In my logs I see both "[Msg] Gazebo Sim GUI " and "[Msg] Gazebo Sim Server", but in yours I only see the former.

Yeah probably as the gazebo gui doesn't connect with gazebo server and tries to initialize that connection with the server over and over again

ShevaDavid, 2023-03-02:
Have the same problem on mac OS Ventura 
Gazebo Sim GUI    v7.4.0
[image]
 gz sim -v4 -r -g shapes.sdf

[Msg] Gazebo Sim GUI    v7.4.0
[Dbg] [Application.cc:96] Initializing application.
[Dbg] [Application.cc:104] Qt using Metal graphics interface
[GUI] [Dbg] [Application.cc:596] Create main window
[GUI] [Wrn] [Application.cc:845] [QT] Populating font family aliases took 129 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Dbg] [PathManager.cc:67] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:241:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be

azeey, 2023-03-02:
@Bezalelohim  can you run just the server with `-s` and see the logs, i.e., `gz sim -v4 -s -r empty.world`? Once that starts, you can inspect the services advertised by the server with `gz service -l`.

@ShevaDavid not sure why you're running with `-g`. That runs just the gui without the server. You need both to be running. Can you try without the `-g`?

azeey, 2023-03-13:
Any update on this @Bezalelohim ?

Bezalelohim, 2023-03-16:
When running  ```gz service -l``` while the above gz sim command runs yields no services

Bezalelohim, 2023-03-16:
@azeey Sorry for the late response.

azeey, 2023-03-16:
It looks like you have a network configuration issue. Do you have a firewall that disables multicast messages? Can you try disabling it?

azeey, 2023-05-01:
Any luck testing it with my suggestion above? If it is working for you, please go ahead and close the issue.

Bezalelohim, 2023-05-02:
Yes @azeey the Ubuntu firewall was the culprit 
Solution was simply
```~$ sudo ufw disable```
Thank You @azeey its fixed :)

closed 2023-05-02 by Bezalelohim

Bezalelohim, 2023-05-02:
Can I use gz sim without disabling ufw ?
Is there a fix for that ?

azeey, 2023-05-02:
I think it's possible to allow multicast without disabling ufw. See https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#enable-multicast

vudala, 2023-09-29:
Running Gazebo Garden on a fresh Ubuntu 22.04 causes this issue.
As suggested by @azeey:

> I think it's possible to allow multicast without disabling ufw. See https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#enable-multicast

Doing these steps will solve the problem.

jnomikos, 2025-03-10:
> Yes [@azeey](https://github.com/azeey) the Ubuntu firewall was the culprit Solution was simply `~$ sudo ufw disable` Thank You [@azeey](https://github.com/azeey) its fixed :)

This fixed it for me too, but I will look into enabling multicast instead because I don't like the idea of disabling my whole firewall