gazebosim/gz-sim #1886: :man_farmer: Flaky test: `UNIT_Gui_clean_exit_TEST` in gz-sim7 focal

Crola1702, 2023-02-08:
## Environment
* OS Version: Ubuntu 20.04
* Binary Gazebo Garden

## Description
* Expected behavior: UNIT_Gui_clean_exit_TEST should pass
* Actual behavior: UNIT_Gui_clean_exit_TEST exceeds the 240 seconds timeout and fails

## Steps to reproduce

1. Run a build in [gz-sim7-focal](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-focal-amd64/)
2. See UNIT_Gui_clean_exit_TEST if lucky.

## Output

Reference build: https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-focal-amd64/67/

Failing test: [UNIT_Gui_clean_exit_TEST.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-focal-amd64/67/testReport/junit/(root)/UNIT_Gui_clean_exit_TEST/test_ran/) (Timeout)

Log output:

```
test 65
        Start  65: UNIT_Gui_clean_exit_TEST

65: Test command: /home/jenkins/workspace/ignition_gazebo-ci-gz-sim7-focal-amd64/build/bin/UNIT_Gui_clean_exit_TEST "--gtest_output=xml:/home/jenkins/workspace/ignition_gazebo-ci-gz-sim7-focal-amd64/build/test_results/UNIT_Gui_clean_exit_TEST.xml"
65: Test timeout computed to be: 240
65: [==========] Running 2 tests from 1 test suite.
65: [----------] Global test environment set-up.
65: [----------] 2 tests from WorldFiles/GazeboDeathTest
65: [ RUN      ] WorldFiles/GazeboDeathTest.CleanExit/0
65: [Msg] Gazebo Sim GUI    v7.3.0
65: [Msg] Loading SDF world file[/home/jenkins/workspace/ignition_gazebo-ci-gz-sim7-focal-amd64/ign-gazebo/test/worlds/empty.sdf].
65: [Dbg] [Application.cc:96] Initializing application.

...

65: [GUI] [Dbg] [MarkerManager.cc:171] Advertise /marker/list service.
65: [GUI] [Dbg] [MarkerManager.cc:181] Advertise /marker/list.
 65/267 Test  #65: UNIT_Gui_clean_exit_TEST ................................***Timeout 240.10 sec
```

Failing test is defined in [this file](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/gui/Gui_clean_exit_TEST.cc). It has not been changed since 2022-09-01. And its first appearance was in 2022-09-28.

Fun fact: It has failed only on Drogon. 

Times it has happened:
| Job Name | Build number | Date | Agent name |
| -- | -- | -- | -- |
|ignition_gazebo-ci-gz-sim7-focal-amd64|67|2023-02-06 |linux-drogon.nv.focal-cba5e3fd|
|ignition_gazebo-ci-gz-sim7-focal-amd64|59|2023-01-20 |linux-drogon.nv.focal-cba5e3fd|
|ignition_gazebo-ci-gz-sim7-focal-amd64|58|2023-01-20 |linux-drogon.nv.focal-cba5e3fd|
|ignition_gazebo-ci-gz-sim7-focal-amd64|48|2022-12-15 |linux-drogon.nv.focal-cba5e3fd|
|ignition_gazebo-ci-gz-sim7-focal-amd64|20|2022-10-20 |linux-drogon.nv.focal-cba5e3fd|
|ignition_gazebo-ci-gz-sim7-focal-amd64|18|2022-10-18 |linux-drogon.nv.focal-cba5e3fd|
|ignition_gazebo-ci-gz-sim7-focal-amd64|10|2022-09-28 |linux-drogon.nv.focal-cba5e3fd|

Maybe Drogon is a slower agent and sometimes needs more than 240s to run the tests?

As this test sometimes fails, this is a Flaky test

azeey, 2023-02-16:
Looking at [this](https://build.osrfoundation.org/job/ignition_gazebo-ci-pr_any-ubuntu_auto-amd64/10185/consoleFull#console-section-6) build on optimus, I see that it's loading the `gui.config` in `WorldFiles/GazeboDeathTest.CleanExit/1`.

<details>

```
65: Test command: /home/jenkins/workspace/ignition_gazebo-ci-pr_any-ubuntu_auto-amd64/build/bin/UNIT_Gui_clean_exit_TEST "--gtest_output=xml:/home/jenkins/workspace/ignition_gazebo-ci-pr_any-ubuntu_auto-amd64/build/test_results/UNIT_Gui_clean_exit_TEST.xml"
65: Test timeout computed to be: 240
65: [==========] Running 2 tests from 1 test suite.
65: [----------] Global test environment set-up.
65: [----------] 2 tests from WorldFiles/GazeboDeathTest
65: [ RUN      ] WorldFiles/GazeboDeathTest.CleanExit/0
65: [Msg] Gazebo Sim GUI    v7.4.0
65: [Msg] Loading SDF world file[/home/jenkins/workspace/ignition_gazebo-ci-pr_any-ubuntu_auto-amd64/ign-gazebo/test/worlds/empty.sdf].
65: [Dbg] [Application.cc:96] Initializing application.
65: [Dbg] [Application.cc:122] Qt using OpenGL graphics interface
65: [GUI] [Dbg] [Application.cc:596] Create main window
65: [GUI] [Msg] Serving entity system service on [/entity/system/add]
65: [GUI] [Dbg] [PathManager.cc:67] Requesting resource paths through [/gazebo/resource_paths/get]
65: [GUI] [Dbg] [Physics.cc:869] Loaded [gz::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/gz-physics-6/engine-plugins/libgz-physics-dartsim-plugin.so]
65: [GUI] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
65: [GUI] [Msg] Create service on [/world/empty/create]
65: [GUI] [Msg] Remove service on [/world/empty/remove]
65: [GUI] [Msg] Pose service on [/world/empty/set_pose]
65: [GUI] [Msg] Pose service on [/world/empty/set_pose_vector]
65: [GUI] [Msg] Light configuration service on [/world/empty/light_config]
65: [GUI] [Msg] Physics service on [/world/empty/set_physics]
65: [GUI] [Msg] SphericalCoordinates service on [/world/empty/set_spherical_coordinates]
65: [GUI] [Msg] Enable collision service on [/world/empty/enable_collision]
65: [GUI] [Msg] Disable collision service on [/world/empty/disable_collision]
65: [GUI] [Msg] Material service on [/world/empty/visual_config]
65: [GUI] [Msg] Material service on [/world/empty/wheel_slip]
65: [GUI] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
65: [GUI] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
65: [GUI] [Dbg] [Gui.cc:340] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
65: [GUI] [Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Contact] for entity [1]
65: [GUI] [Msg] Loaded level [3]
65: [GUI] [Msg] Serving world controls on [/world/empty/control], [/world/empty/control/state] and [/world/empty/playback/control]
65: [GUI] [Msg] Serving GUI information on [/world/empty/gui/info]
65: [GUI] [Msg] World [empty] initialized with [fast] physics profile.
65: [GUI] [Msg] Serving world SDF generation service on [/world/empty/generate_world_sdf]
65: [GUI] [Msg] Serving world names on [/gazebo/worlds]
65: [GUI] [Msg] Resource path add service on [/gazebo/resource_paths/add].
65: [GUI] [Msg] Resource path get service on [/gazebo/resource_paths/get].
65: [GUI] [Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
65: [GUI] [Msg] Resource paths published on [/gazebo/resource_paths].
65: [GUI] [Msg] Server control service on [/server_control].
65: [GUI] [Msg] Found no publishers on /stats, adding root stats topic
65: [GUI] [Msg] Found no publishers on /clock, adding root clock topic
65: [GUI] [Dbg] [SimulationRunner.cc:513] Creating PostUpdate worker threads: 3
65: [GUI] [Dbg] [SimulationRunner.cc:524] Creating postupdate worker thread (0)
65: [GUI] [Dbg] [SimulationRunner.cc:524] Creating postupdate worker thread (1)
65: [GUI] [Msg] Serving scene information on [/world/empty/scene/info]
65: [GUI] [Msg] Serving graph information on [/world/empty/scene/graph]
65: [GUI] [Msg] Serving full state on [/world/empty/state]
65: [GUI] [Msg] Serving full state (async) on [/world/empty/state_async]
65: [GUI] [Msg] Publishing scene information on [/world/empty/scene/info]
65: [GUI] [Msg] Publishing entity deletions on [/world/empty/scene/deletion]
65: [GUI] [Msg] Publishing state changes on [/world/empty/state]
65: [GUI] [Msg] Publishing pose messages on [/world/empty/pose/info]
65: [GUI] [Msg] Publishing dynamic pose messages on [/world/empty/dynamic_pose/info]
65: [GUI] [Dbg] [SignalHandler.cc:141] Received signal[15].
65: [GUI] [Dbg] [SignalHandler.cc:141] Received signal[15].
65: [GUI] [Dbg] [ServerPrivate.cc:115] Server received signal[15]
65: [GUI] [Dbg] [SignalHandler.cc:141] Received signal[15].
65: [GUI] [Dbg] [SimulationRunner.cc:540] [GUI] [Dbg] [SimulationRunner.cc:540] [GUI] [Dbg] [SimulationRunner.cc:540] Exiting postupdate worker thread (1Exiting postupdate worker thread (0))
65: 
65: [GUI] [Dbg] [Application.cc:167] Terminating application.
65: [       OK ] WorldFiles/GazeboDeathTest.CleanExit/0 (5204 ms)
65: [ RUN      ] WorldFiles/GazeboDeathTest.CleanExit/1
65: [Msg] Gazebo Sim GUI    v7.4.0
[... 38 lines cut ...]
65: [GUI] [Dbg] [SimulationRunner.cc:524] Creating postupdate worker thread (0)
65: [GUI] [Msg] Serving scene information on [/world/default/scene/info]
65: [GUI] [Msg] Serving graph information on [/world/default/scene/graph]
[... cut ...]
65: [GUI] [Dbg] [SimulationRunner.cc:540] Exiting postupdate worker thread (0)
65: [GUI] [Dbg] [Application.cc:167] Terminating application.
65: [       OK ] WorldFiles/GazeboDeathTest.CleanExit/1 (5189 ms)
65: [----------] 2 tests from WorldFiles/GazeboDeathTest (10394 ms total)
65: 
65: [----------] Global test environment tear-down
65: [==========] 2 tests from 1 test suite ran. (10394 ms total)
65: [  PASSED  ] 2 tests.
 65/269 Test  #65: UNIT_Gui_clean_exit_TEST ................................   Passed   10.42 sec
```
</details>

Blast545, 2023-02-16:
@azeey We are mainly comparing what the machines do differently for the specific test that's failing right now, which is `UNIT_Gui_clean_exit_TEST`. 

This is your reference build: [ignition_gazebo-ci-pr_any-ubuntu_auto-amd64#10185](https://build.osrfoundation.org/job/ignition_gazebo-ci-pr_any-ubuntu_auto-amd64/10185/consoleFull#console-section-6 )
This is a reference build with the problem: [ignition_gazebo-ci-gz-sim7-focal-amd64#67](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-focal-amd64/67)

It seems you may be right, the config file may be available in both scenarios if we look to the output from other tests. However, if you grep for the specific failing test, you can see that in the failing build in drogon, for some reason it's loading the config file (and several other components) that are not being loaded in the optimus build. I'll attach the diff of the two tests for reference.

[diff.txt](https://github.com/gazebosim/gz-sim/files/10758576/diff.txt)

Blast545, 2023-02-16:
@azeey I took some time to check more closely this problem with @Crola1702 just now, I'm not completely sure how the Server/Gui synchronization mechanism works for Gazebo, but it seems to me that the run that's passing (optimus) it's loading properly the `empty.sdf` file (and then not loading any plugins/rendering stuff) because it's an empty sdf, and for drogon the gui it's not finding the empty.sdf being listed by the worlds service, and then it's loading what it seems default gui config (which seems to contain various things that take lots of time to load).

@Crola1702 Can we see the result saved into the `WorldMsg` variable? https://github.com/gazebosim/gz-sim/blob/3592c201094b94b36a3edbd1f03f89d959f4e578/src/gui/Gui.cc#L342
Maybe add a `gzdbg` print statement, so we can take a closer look to the output from the two different nodes.

azeey, 2023-02-16:
The test in optimus shows that it received signal 15 (SIGTERM) which is expected. The test then immediately terminates without loading the rest of the plugins. I think on drogon, for some reason the signal is not being received and so gazebo continues running and loads all the plugins. I wonder if it's a timing issue in that gazebo is not ready to receive signals while it's loading. One thing to try would be to wait longer in the test before sending the signal so things have stabilized:

https://github.com/gazebosim/gz-sim/blob/7bc9b93b8c0ea3d8419c553dfec69cea7c92e34d/src/gui/Gui_clean_exit_TEST.cc#L72

It's currently set to `4s`, We could try `15s`? I made it short so the test wouldn't take too long, but if it's making it flaky, maybe increasing the delay would be worth it.

PR #1897 (2023-02-14): Increase timeout for UNIT_Gui_clean_exit_TEST
# 🦟 Bug fix

Fixes #1886

## Summary

This PR increases the default timeout for Gui_clean_exit_TEST (240 -> 300 seconds)

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] W [... cut]
PR #1902 (2023-02-17): Increase thread sleep delay
# 🦟 Bug fix

Fixes #1886 

## Summary

Check https://github.com/gazebosim/gz-sim/issues/1886#issuecomment-1430320323 and below

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting [... cut]
closed 2023-02-17 by azeey