gazebosim/gz-sim #1887: :farmer: `testFixture_TEST.py` failing in gazebo6 Homebrew

Crola1702, 2023-02-09:
## Environment
* OS Version: MacOS Bigsur
* Gazebo Fortress binary

## Description
* Expected behavior: .ign-gazebo.python.test.testFixture_TEST should pass
* Actual behavior: .ign-gazebo.python.test.testFixture_TEST fails

## Steps to reproduce

1. Run a build in [ignition_gazebo-ci-ign-gazebo6-homebrew-amd64](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/) job
2. See testFixture_TEST fail

## Output

Reference build: https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/180/#showFailuresLink

Failing test: [.ign-gazebo.python.test.testFixture_TEST](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/180/testReport/junit/(root)/(empty)/ign_gazebo_python_test_testFixture_TEST/)

Log output:

```
test 249
        Start 249: testFixture_TEST.py

249: Test command: /usr/local/Frameworks/Python.framework/Versions/3.10/bin/python3.10 "-m" "pytest" "/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/ign-gazebo/python/test/testFixture_TEST.py" "--junitxml" "/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/build/test_results/UNIT_testFixture_TEST.xml"
249: Working Directory: /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/build/python
249: Environment variables: 
249:  PYTHONPATH=/usr/local/Cellar/gz-sim6/HEAD/lib/python/
249:  LD_LIBRARY_PATH=/usr/local/Cellar/gz-sim6/HEAD/lib:
249: Test timeout computed to be: 1500
249: ============================= test session starts ==============================
249: platform darwin -- Python 3.10.9, pytest-7.2.1, pluggy-1.0.0
249: rootdir: /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64
249: collected 0 items / 1 error
249: 
249: ==================================== ERRORS ====================================
249: _________ ERROR collecting ign-gazebo/python/test/testFixture_TEST.py __________
249: ImportError while importing test module '/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/ign-gazebo/python/test/testFixture_TEST.py'.
249: Hint: make sure your test modules/packages have valid Python names.
249: Traceback:
249: /usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/importlib/__init__.py:126: in import_module
249:     return _bootstrap._gcd_import(name[level:], package, level)
249: ../../ign-gazebo/python/test/testFixture_TEST.py:20: in <module>
249:     from ignition.math import Vector3d
249: E   ModuleNotFoundError: No module named 'ignition.math'
249: - generated xml file: /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo6-homebrew-amd64/build/test_results/UNIT_testFixture_TEST.xml -
249: =========================== short test summary info ============================
249: ERROR ../../ign-gazebo/python/test/testFixture_TEST.py
249: !!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!
249: =============================== 1 error in 0.37s ===============================
249/249 Test #249: testFixture_TEST.py .....................................***Failed    0.68 sec
```

This test failure is consistently failing in gazebo6 Homebrew only since 2022-08-23:
| Build number | Date |
| -- | -- |
|180|2023-02-03 |
|179|2023-01-26 |
|173|2023-01-05 |
| ... | ... |
|122|2022-08-23 |
|123|2022-08-25 |
|124|2022-08-30 |

Crola1702, 2024-04-08:
This issue is still affecting consistently to gz-sim6-homebrew ([reference build](https://build.osrfoundation.org/job/gz_sim-ci-ign-gazebo6-homebrew-amd64/34/))

<details>
<summary>
Log output (almost the same as before):
</summary>

**Note the module error message:**
```
273:     from ignition.math import Vector3d
273: E   ModuleNotFoundError: No module named 'ignition.math'
```

---

```
test 273
        Start 273: testFixture_TEST.py

273: Test command: /usr/local/Frameworks/Python.framework/Versions/3.12/bin/python3.12 "-m" "pytest" "/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-ign-gazebo6-homebrew-amd64/gz-sim/python/test/testFixture_TEST.py" "--junitxml" "/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-ign-gazebo6-homebrew-amd64/build/test_results/UNIT_testFixture_TEST.xml"
273: Working Directory: /Users/jenkins/jenkins-agent/workspace/gz_sim-ci-ign-gazebo6-homebrew-amd64/build/python
273: Environment variables: 
273:  PYTHONPATH=/usr/local/Cellar/gz-sim6/HEAD/lib/python/
273:  LD_LIBRARY_PATH=/usr/local/Cellar/gz-sim6/HEAD/lib:
273: Test timeout computed to be: 1500
273: ============================= test session starts ==============================
273: platform darwin -- Python 3.12.2, pytest-8.1.1, pluggy-1.4.0
273: rootdir: /Users/jenkins/jenkins-agent/workspace/gz_sim-ci-ign-gazebo6-homebrew-amd64
273: collected 0 items / 1 error
273: 
273: ==================================== ERRORS ====================================
273: ___________ ERROR collecting gz-sim/python/test/testFixture_TEST.py ____________
273: ImportError while importing test module '/Users/jenkins/jenkins-agent/workspace/gz_sim-ci-ign-gazebo6-homebrew-amd64/gz-sim/python/test/testFixture_TEST.py'.
273: Hint: make sure your test modules/packages have valid Python names.
273: Traceback:
273: /usr/local/Cellar/python@3.12/3.12.2_1/Frameworks/Python.framework/Versions/3.12/lib/python3.12/importlib/__init__.py:90: in import_module
273:     return _bootstrap._gcd_import(name[level:], package, level)
273: ../../gz-sim/python/test/testFixture_TEST.py:20: in <module>
273:     from ignition.math import Vector3d
273: E   ModuleNotFoundError: No module named 'ignition.math'
273: - generated xml file: /Users/jenkins/jenkins-agent/workspace/gz_sim-ci-ign-gazebo6-homebrew-amd64/build/test_results/UNIT_testFixture_TEST.xml -
273: =========================== short test summary info ============================
273: ERROR ../../gz-sim/python/test/testFixture_TEST.py
273: !!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!
273: =============================== 1 error in 0.60s ===============================
273/273 Test #273: testFixture_TEST.py .....................................***Failed    0.90 sec
```

</details>

scpeters, 2024-04-15:
> This issue is still affecting consistently to gz-sim6-homebrew ([reference build](https://build.osrfoundation.org/job/gz_sim-ci-ign-gazebo6-homebrew-amd64/34/))
> 
> Log output (almost the same as before):

This is the same issue as https://github.com/gazebosim/gz-sim/issues/2249, which was closed after fixing Garden, Harmonic, and Ionic. Let's handle Fortress in this issue