gazebosim/gz-sim #1889: How to create model plugin in ignition?

mkschreder, 2023-02-09:
I was surprised to find just how little documentation there is available for the ignition gazebo.

I want to port a gazebo classic model plugin to ignition. How do I do this? Where are all the plugin interfaces? Which ones should be used?

Perhaps there are some examples of existing model plugins somewhere?

srmainwaring, 2023-02-09:
Hi @mkschreder, there are lots of examples of model plugins in the src/systems folder:

https://github.com/gazebosim/gz-sim/tree/gz-sim7/src/systems/diff_drive

and a fully worked example of migrating a Gazebo 11 plugin to Gazebo Sim here:

https://gazebosim.org/api/sim/7/ardupilot.html

mkschreder, 2023-02-09:
@srmainwaring very nice. Thanks.

closed 2023-02-09 by mkschreder

serkanMzlm, 2024-07-18:
> @srmainwaring very nice. Thanks.
> 
> PS. none of these links are readily discoverable from the main site. I ended up exploring the examples folder to make it work.

I agree, the documentation for gz sim is very bad. It would be much better if it was made like ROS2