gazebosim/gz-sim #1890: SDF editor

mkschreder, 2023-02-09:
What's the preferred way of editing SDF setups including joints, collision shapes etc?

I find that the current gz-sim editor is extremely limited without any possibility to move joints for example.

Is there a straightforward way to define these things in blender perhaps and then export to SDF? Or is there some other way?

azeey, 2023-02-28:
There might be some pending work on the model editor (see https://github.com/gazebosim/gz-sim/issues/1232), but I'm not sure that includes what you're looking for. If there's a specific feature you're looking for, would you mind creating a feature request.

There are other approaches, but I don't know of a fully featured graphical editor for SDF files. You could take a look at https://github.com/Rhoban/onshape-to-robot,  https://github.com/ros/solidworks_urdf_exporter or https://github.com/dfki-ric/phobos as possible options, but I have limited to no experience with any of them.

mkschreder, 2023-03-07:
An SDF editor is definitely needed. Manually typing coordinates is like manually typing vertices for a mesh model. Very old school.

SurafelTesfaye, 2023-03-11:
> There might be some pending work on the model editor (see #1232), but I'm not sure that includes what you're looking for. If there's a specific feature you're looking for, would you mind creating a [...]

Yes, I have tried the Solidworks plugin, it worked fine for me, also Solidworks gives the flexibility to work with different file formats for the model (I am not sure about Blender), we are also able to calculate different properties automatically from the model including moment of inertia, density etc depending on the material we defined. We can use the api provided by Solidworks to make a new plugin. This is interesting I'd be happy to work on this