gazebosim/gz-sim #1891: Unable to run ignition gazebo on Ubuntu 22.04

tadteo, 2023-02-10:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build:
 binary  (tried both gazebo fortress and garden)
* I'm using Intel CPU with an integrated GPU

Ignition Gazebo GUI    v6.14.0

Rendering system info:

- lspci:
```
      00:02.0 VGA compatible controller [0300]: Intel Corporation RocketLake-S GT1 [UHD Graphics 750] [8086:4c8a] (rev 04)
```
- Display:     ``` :0```
- glxinfo: 
```
direct rendering: Yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) Graphics (RKL GT1)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.2.5
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.2.5
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.2.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```
- ps aux | grep Xorg :
```
my-username      6818  0.0  0.0  12088  2684 pts/0    S+   14:36   0:00 grep --color=auto Xorg
```
- sudo env LANG=C X -version:
```
X.Org X Server 1.21.1.3
X Protocol Version 11, Revision 0
Current Operating System: Linux my-home 5.15.0-60-generic #66-Ubuntu SMP Fri Jan 20 14:29:49 UTC 2023 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-60-generic root=UUID=93337e43-f0f4-4079-bfb7-2a529dc3b8fd ro quiet splash vt.handoff=7
xorg-server 2:21.1.3-2ubuntu2.7 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
```

However, using the default wayland on Ubuntu 22.04

## Description
Running
``` ign gazebo```
the window to select the world appears, and when I select a world and press "run", the program crashes (tried different worlds and it does not work). Tried also disabling wayland and running ubuntu on X11 but still same errors.

## Steps to reproduce
1. Run ```ign gazebo```
2. Execute any world (for example empty)
3. press "run"

## Output
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QSocketNotifier: Can only be used with threads started with QThread
Stack trace (most recent call last) in thread 8035:
#9    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#8    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3a8acc59ff, in 
#7    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f3a8ac33b42, in 
#6    Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f3a84d8eca0, in 
#5    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7f3a8420ee06, in 
#4    Object "/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7f3a8420e61b, in 
#3    Object "/usr/lib/x86_64-linux-gnu/qt5/plugins/wayland-graphics-integration-client/libqt-plugin-wayland-egl.so", at 0x7f3a6c2b8b2a, in 
#2    Object "/usr/lib/x86_64-linux-gnu/qt5/plugins/wayland-graphics-integration-client/libqt-plugin-wayland-egl.so", at 0x7f3a6c2b91ca, in 
#1    Object "/lib/x86_64-linux-gnu/libQt5WaylandClient.so.5", at 0x7f3a7c6600e8, in QtWaylandClient::QWaylandAbstractDecoration::contentImage()
#0    Object "/lib/x86_64-linux-gnu/libQt5WaylandClient.so.5", at 0x7f3a7c6753e4, in QtWayland::wl_surface::damage(int, int, int, int)
Segmentation fault (Address not mapped to object [0x18])
Escalating to SIGKILL on [Ignition Gazebo Server]
```

BiggestRic, 2023-02-14:
I am having the exact same issue, have you found anything out?

iche033, 2023-02-14:
wayland probably won't work. You'll need to run gazebo with X server. It sounds like you're able to see the 3D scene in the GUI window but crashes only after hitting run?  You can get more verbose console output with `-v 4`, e.g. `ign gazebo -v 4 [world_name]` and see if there're any error messages

JaouadROS, 2023-03-02:
@iche033 I've got the same symptoms. When I run the tutorial examples `ros2 launch ros_gz_sim_demos camera.launch.py` it loads the 3D scene and once it starts publishing images it crushes. First it shows this message:
`[ruby $(which ign) gazebo-1] ../src/intel/isl/isl.c:2216: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.`
Then after few seconds it shutdown with segmentation fault
```
[ruby $(which ign) gazebo-1] Stack trace (most recent call last) in thread 7012:
[ruby $(which ign) gazebo-1] #24   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ruby $(which ign) gazebo-1] #23   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7effcf7269ff, in 
[ruby $(which ign) gazebo-1] #22   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7effcf694b42, in 
[ruby $(which ign) gazebo-1] #21   Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7effcaedc2b2, in 
[ruby $(which ign) gazebo-1] #20   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7effb01212bf, in ignition::gazebo::v6::systems::SensorsPrivate::RenderThread()
[ruby $(which ign) gazebo-1] #19   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/libignition-gazebo-sensors-system.so", at 0x7effb0120ba6, in ignition::gazebo::v6::systems::SensorsPrivate::RunOnce()
[ruby $(which ign) gazebo-1] #18   Object "/lib/x86_64-linux-gnu/libignition-sensors6.so.6", at 0x7effa7e70c6e, in ignition::sensors::v6::Manager::RunOnce(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
[ruby $(which ign) gazebo-1] #17   Object "/lib/x86_64-linux-gnu/libignition-sensors6.so.6", at 0x7effa7e70c05, in ignition::sensors::v6::Sensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
[ruby $(which ign) gazebo-1] #16   Object "/lib/x86_64-linux-gnu/libignition-sensors6-camera.so.6", at 0x7effa7ec580f, in ignition::sensors::v6::CameraSensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&)
[ruby $(which ign) gazebo-1] #15   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre.so", at 0x7effa452f2ab, in 
[ruby $(which ign) gazebo-1] #14   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-6/engine-plugins/libignition-rendering-ogre.so", at 0x7effa45aeb44, in ignition::rendering::v6::OgreRenderTarget::Copy(ignition::rendering::v6::Image&) const
[ruby $(which ign) gazebo-1] #13   Object "/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0", at 0x7eff974ac90a, in Ogre::RenderTexture::copyContentsToMemory(Ogre::PixelBox const&, Ogre::RenderTarget::FrameBuffer)
[ruby $(which ign) gazebo-1] #12   Object "/usr/lib/x86_64-linux-gnu/OGRE-1.9.0/RenderSystem_GL.so", at 0x7eff6df67a85, in Ogre::GLTextureBuffer::download(Ogre::PixelBox const&)
[ruby $(which ign) gazebo-1] #11   Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff76919cdd, in 
[ruby $(which ign) gazebo-1] #10   Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff769186ae, in 
[ruby $(which ign) gazebo-1] #9    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff769184b0, in 
[ruby $(which ign) gazebo-1] #8    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff76776975, in 
[ruby $(which ign) gazebo-1] #7    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff76919376, in 
[ruby $(which ign) gazebo-1] #6    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff767740a8, in 
[ruby $(which ign) gazebo-1] #5    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff76797f19, in 
[ruby $(which ign) gazebo-1] #4    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff7736bf37, in 
[ruby $(which ign) gazebo-1] #3    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7eff7669c8b6, in 
[ruby $(which ign) gazebo-1] #2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7effcf6287f2, in abort
[ruby $(which ign) gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7effcf642475, in raise
[ruby $(which ign) gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7effcf696a7c, in pthread_kill
[ruby $(which ign) gazebo-1] Aborted (Signal sent by tkill() 6929 1000)
[ruby $(which ign) gazebo-1] Stack trace (most recent call last):
[ruby $(which ign) gazebo-1] #30   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
[ruby $(which ign) gazebo-1] #29   Object "ign gazebo gui", at 0x5592b4f941c4, in _start
[ruby $(which ign) gazebo-1] #28   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7effcf629e3f, in __libc_start_main
[ruby $(which ign) gazebo-1] #27   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7effcf629d8f, in 
[ruby $(which ign) gazebo-1] #26   Object "ign gazebo gui", at 0x5592b4f9417e, in 
[ruby $(which ign) gazebo-1] #25   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7effcfaa8e19, in ruby_run_node
[ruby $(which ign) gazebo-1] #24   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7effcfaa5317, in 
[ruby $(which ign) gazebo-1] #23   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7effcfc3a30c, in rb_vm_exec
[ruby $(which ign) gazebo-1] #22   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7effcfc34c96, in 
[ruby $(which ign) gazebo-1] #21   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7effcfc31fc5, in 
[ruby $(which ign) gazebo-1] #20   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7effcfc2fc34, in 
[... cut ...]
[ruby $(which ign) gazebo-1] #1    Object "/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7effca291281, in ignition::gazebo::v6::GuiRunner::~GuiRunner()
[ruby $(which ign) gazebo-1] #0    Object "/lib/x86_64-linux-gnu/libignition-gazebo6-gui.so.6", at 0x7effca29fadb, in void ignition::utils::detail::DefaultDelete<ignition::gazebo::v6::GuiRunner::Implementation>(ignition::gazebo::v6::GuiRunner::Implementation*)
[ruby $(which ign) gazebo-1] Segmentation fault (Address not mapped to object [0x7eff7c97cf18])

```
Is it related to the intel gpu? can I set gazebo to use m eGPU?

iche033, 2023-03-02:
The `isl` error maybe it's related to your graphics driver. Try updating your mesa driver?

The backtrace shows that it's crashing in the rendering. You can take a look at `~/.ignition/rendering/ogre2.log` and see if there are any ogre exception logs.

You can run everything without ros for easier debugging

```
ign gazebo -v 4 shape.sdf.
```

You can also try running with the ogre1.x render engine and see if it makes any difference:

```
ign gazebo -v 4 shapes.sdf --render-engine ogre
```

JaouadROS, 2023-03-03:
@iche033 Thank you, that was helpful.
Also [Troubleshooting](https://github.com/gazebosim/docs/blob/master/fortress/troubleshooting.md) here is useful.

AnishMathewOommen, 2024-04-15:
**Found the Solution for the above problem**
Please type the below commands in the terminal the issue will be solved( Updating Mesa driver)

```
sudo add-apt-repository ppa:kisak/kisak-mesa
sudo apt update
sudo apt upgrade

```

Reference document: https://itsfoss.com/install-mesa-ubuntu/

Farraj007, 2024-04-27:
i did update my graphics driver it didn't help

husanboyqodirov, 2025-02-25:
> **Found the Solution for the above problem** Please type the below commands in the terminal the issue will be solved( Updating Mesa driver)
> 
> ```
> sudo add-apt-repository ppa:kisak/kisak-mesa
> sudo apt update
> sudo apt upgrade
> ```
> 
> Reference document: https://itsfoss.com/install-mesa-ubuntu/

This worked for me.