gazebosim/gz-sim #1903: MULTICOPTER MOTOR MODEL PLUGIN

omr-web, 2023-02-17:
hello everyone,
I want to make an attitude and altitude controller in the ignition gazebo. I use the multicopter motor model plugin for this. As a result of my experiments, I realized the following: Although the velocity value I give to the motors is constantly increasing, the drone constantly falls to the ground after a certain height. 

What is the reason of this? Is it a feature originating from the gazebo world?

Thank you for your help.

azeey, 2023-03-01:
@omr-web this is a question best suited for answers.gazebosim.org or robotics.stackexchange.com since it doesn't appear to be a bug/feature request on gz-sim. Could you please post your question there?

closed 2023-03-01 by azeey

vbalaji21, 2024-04-12:
I am curious on this question and answer. If it is solved can you please let me know, how it is solved. If the question is asked in robotics stack exchange, share the link here.

aarsht7, 2025-05-12:
Hello @azeey 

I faced a similar issue with gzsim. I have asked the [question on the roobtics se](https://robotics.stackexchange.com/questions/115587/gzsim-drone-crashes-after-certain-time-of-flight). Will it be possible for you to have a look whenever you get some time?

Thanks!