gazebosim/gz-sim #1904: Reset/Delete services crash when spawning model with ackermann plugin

retinfai, 2023-02-20:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? Binary Garden 

## Description
* Expected behavior:
Resetting or Deleting a model does so gracefully

* Actual behavior:
When spawning a model with the ackermann steering plugin after already starting the simulation, then resetting/deleting the model, results in crash.

Resetting simulation results in segmentation fault
Deleting model causes error spams

## Steps to reproduce

Prereq:
extract the ackermann_steering example car into it's own sdf, so we can spawn after starting the simulation. For this purpose, this file is referred to as `car.sdf`
1. `gz sim empty.sdf`
2. `gz service -s /world/empty/create --reqtype gz.msgs.EntityFactory --reptype gz.msgs.Boolean --timeout 1000 --req 'sdf_filename: "path/to/car.sdf", name: "blue-car"'
`
3. `gz service -s /world/empty/control --reqtype gz.msgs.WorldControl --reptype gz.msgs.Boolean --timeout 3000 --req 'reset: {all: true}'` 
or 
`gz service -s world/empty/remove --reqtype gz.msgs.Entity --reptype gz.msgs.Boolean --timeout 1000 --req 'type: 2, name: "blue-car"'`

## Output

For resetting simulation:
```
Stack trace (most recent call last) in thread 1493792:
#6    Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#5    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6b89c2a9ff, in 
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f6b89b98b42, in 
#3    Object "/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f6b8538d2b2, in 
#2    Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f6b8474c82d, in 
#1    Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-ackermann-steering-system.so", at 0x7f6b550d391d, in non-virtual thunk to gz::sim::v7::systems::AckermannSteering::PostUpdate(gz::sim::v7::UpdateInfo const&, gz::sim::v7::EntityComponentManager const&)
#0    Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-ackermann-steering-system.so", at 0x7f6b550cac4c, in gz::sim::v7::systems::AckermannSteeringPrivate::UpdateVelocity(gz::sim::v7::UpdateInfo const&, gz::sim::v7::EntityComponentManager const&)
Segmentation fault (Address not mapped to object [(nil)])

```
For deleting the model:
Identical errors to previous issue #1107 
```
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [32], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [34], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [33], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [35], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [30], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [31], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [8319580315957903596] attached to entity [32], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [8319580315957903596] attached to entity [33], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [8319580315957903596] attached to entity [30], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [8319580315957903596] attached to entity [31], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [32], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [34], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [33], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [35], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [30], but this entity does not exist. This create component request will be ignored.
[Err] [EntityComponentManager.cc:1047] Trying to create a component of type [10078773228734069488] attached to entity [31], but this entity does not exist. This create component request will be ignored.
```

bperseghetti, 2023-02-27:
@retinfai can you include your "blue car" model?

retinfai, 2023-02-28:
Sorry for not putting this in the issue. 

Here is`car.sdf`. Taken from ackermann steering gazebo examples

```
<?xml version="1.0" ?>

<sdf version="1.6">
	<model name='vehicle_blue'>
	<pose>0 2 0.325 0 -0 0</pose>

      <link name='chassis'>
        <pose>-0.151427 -0 0.175 0 -0 0</pose>
        <inertial>
          <mass>1.14395</mass>
          <inertia>
            <ixx>0.126164</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>0.416519</iyy>
            <iyz>0</iyz>
            <izz>0.481014</izz>
          </inertia>
        </inertial>
        <visual name='visual'>
          <geometry>
            <box>
              <size>2.01142 1 0.568726</size>
            </box>
          </geometry>
          <material>
            <ambient>0.5 0.5 1.0 1</ambient>
            <diffuse>0.5 0.5 1.0 1</diffuse>
            <specular>0.0 0.0 1.0 1</specular>
          </material>
        </visual>
        <collision name='collision'>
          <geometry>
            <box>
              <size>2.01142 1 0.568726</size>
            </box>
          </geometry>
        </collision>
      </link>

      <link name='front_left_wheel'>
        <pose>0.554283 0.625029 -0.025 -1.5707 0 0</pose>
        <inertial>
          <mass>2</mass>
          <inertia>
            <ixx>0.145833</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>0.145833</iyy>
            <iyz>0</iyz>
            <izz>0.125</izz>
          </inertia>
        </inertial>
        <visual name='visual'>
          <geometry>
            <cylinder>
              <length>0.15</length>
              <radius>0.3</radius>
            </cylinder>
          </geometry>
          <material>
            <ambient>0.2 0.2 0.2 1</ambient>
            <diffuse>0.2 0.2 0.2 1</diffuse>
            <specular>0.2 0.2 0.2 1</specular>
          </material>
[... 312 lines cut ...]
        <steering_limit>0.5</steering_limit>
        <wheel_base>1.0</wheel_base>
        <wheel_separation>1.25</wheel_separation>
        <wheel_radius>0.3</wheel_radius>
        <min_velocity>-1</min_velocity>
        <max_velocity>3</max_velocity>
        <min_acceleration>-3</min_acceleration>
        <max_acceleration>3</max_acceleration>
      </plugin>
      
      <plugin
        filename="gz-sim-pose-publisher-system"
        name="gz::sim::systems::PosePublisher">
        <publish_link_pose>false</publish_link_pose>
        <publish_collision_pose>false</publish_collision_pose>
        <publish_visual_pose>false</publish_visual_pose>
        <publish_nested_model_pose>true</publish_nested_model_pose>
      </plugin>

    </model> 

</sdf>
```

The gazebo version is `7.3.0`

retinfai, 2023-04-10:
I found this in the System Manager:

```
  /// \brief Perform a reset on all systems
  ///
  /// If a system implements the ISystemReset interface, it will be called.
  //
  /// Otherwise, if a system does not have the ISystemReset interface
  /// implemented, and was created via loading a plugin,
  /// that plugin will be reloaded.
  ///
  /// Otherwise, if a system is created from in-memory rather than a plugin,
  /// that system will remain unaffected.
```

Just wanted to ask what the last part is talking about. If you spawn an entity, is this creating the systems that the entity has from in-memory?

azeey, 2023-05-03:
> I found this in the System Manager:
>
> ```
>   /// \brief Perform a reset on all systems
>   ///
>   /// If a system implements the ISystemReset interface, it will be called.
>   //
>   /// Otherwise, if a [...]

It's possible to add systems from `shared_ptr` instances of a class that implements the `System` interfaces as opposed to being loaded from a shared library (plugin). This was added in https://github.com/gazebosim/gz-sim/pull/936 for reference.