gazebosim/gz-sim #1905: Python bindings to interact with gz-sim

gnanukoth, 2023-02-22:
Hello, may I know if there are any existing python bindings or library that is suitable for interacting and the controlling gazebo simulations? If yes, could you point me to the same? I would like to be able to publish robot control commands and corresponding  parameter values from python and also be able to subscribe to the sensor data. The current available option seems to be through ROS. I was wondering if that could be bypassed and direct python-gazebo interface could be setup.
Thanks!

ahcorde, 2023-02-23:
there are python binding for
 - `gz-math` https://github.com/gazebosim/gz-math/tree/gz-math7/src/python_pybind11\
 - `sdformat` https://github.com/gazebosim/sdformat/tree/sdf13/python
 - `gz-sim` https://github.com/gazebosim/gz-sim/tree/gz-sim7/python

`gz-math` and `sdformat` is almost fully supported, but for `gz-sim` there is some preliminary work but nothing really stable.

gnanukoth, 2023-02-24:
Noted, Thank you!

closed 2023-02-24 by gnanukoth