gazebosim/gz-sim #1909: Port LensFlareSensorPlugin from gazebo-classic to a LensFlare system

scpeters, 2023-02-28:
## Desired behavior

There is support for lens flares in gazebo-classic (see the [lensflare_plugin.world](https://github.com/gazebosim/gazebo-classic/blob/gazebo11/worlds/lensflare_plugin.world) example world). Support for lens flares has been added to gz-rendering's harmonic (`main`) branch in https://github.com/gazebosim/gz-rendering/issues/730, but in order to use it, we should port the [LensFlareSensorPlugin](https://github.com/gazebosim/gazebo-classic/blob/gazebo11/plugins/LensFlareSensorPlugin.cc) to a gz-sim. It should not be too difficult to implement.

## Alternatives considered

## Implementation suggestion

## Additional context

jasmeet0915, 2023-03-07:
Hi @scpeters, I would like to work on this issue. 

From what I understood one would have to create a `LensFlare` system plugin in the `gz-sim` repo which adds the `LensFlarePass` as a `RenderPass` for the camera after setting the values (scale, color, and occlusion steps) in the `Configure Interface` of the plugin. Something like shown in the `Test it` code snippet of the LensFlare pass [PR](https://github.com/gazebosim/gz-rendering/pull/752).

Am I headed in the right direction with this implementation?

iche033, 2023-03-07:
yes that's the right idea. One difference is that the LensFlarePass must be added to the camera in the render thread instead of `Configure`. As an example, you can take a look at the [camera_video_recorder](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/camera_video_recorder/CameraVideoRecorder.cc) system which is also a sensor plugin. 

In this example, the system [connects to a render event](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/camera_video_recorder/CameraVideoRecorder.cc#L196-L201) so all rendering operations are done in the [callback function](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/camera_video_recorder/CameraVideoRecorder.cc#L276) in the render thread. 

It [retrieves the name of the camera](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/camera_video_recorder/CameraVideoRecorder.cc#L436-L438) based on its entity id (you should be able to do this in `Configure`) then [use the name to get the rendering camera ptr](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/camera_video_recorder/CameraVideoRecorder.cc#L296-L303). You just then need to apply Lens Flare pass to that camera. Afterwards, you are free to [disconnect from the render event](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/camera_video_recorder/CameraVideoRecorder.cc#L423-L424).

scpeters, 2023-03-07:
> Hi @scpeters, I would like to work on this issue.

Hooray! Thanks for volunteering, and please follow Ian's advice as he knows more than I do about this topic.

jasmeet0915, 2023-03-11:
@scpeters @iche033 I was able to add the [LensFlare System](https://github.com/jasmeet0915/gz-sim/tree/jasmeet/lens_flare_system/src/systems/lens_flare) to gz-sim. I have also ported the [example sdf ](https://github.com/jasmeet0915/gz-sim/blob/jasmeet/lens_flare_system/examples/worlds/camera_lens_flare.sdf)from `gazebo-classic` over to this repo. Here is a snapshot:

[image]

I am not sure why but the effect was somewhat more prominent in the Wide Angle Camera as compared to others. I'll create a PR soon after adding the integration tests!

iche033, 2023-03-13:
awesome thanks for the work! I'll take a look at the PR once it's up

jasmeet0915, 2023-03-16:
@iche033 the [PR](https://github.com/gazebosim/gz-sim/pull/1933) is up!

scpeters, 2023-03-20:
lens_flare system added in https://github.com/gazebosim/gz-sim/pull/1933