gazebosim/gz-sim #1923: Unbounded velocity growth observed with hydrodynamics plugin

laughlinbarker, 2023-03-09:
## Environment
* OS Version: Ubuntu 22.04.2 LTS
* Source or binary build: binary, version 7.4.0
    - Rendering plugin: ogre2
      - [X] GUI rendering error. 
      - [X] running in Docker/Singularity
    - Rendering system info:
    ```
ubuntu@coyote:/opt/mr/nadir$ lspci -nn | grep VGA
2d:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP107 [GeForce GTX 1050 Ti] [10de:1c82] (rev a1)
ubuntu@coyote:/opt/mr/nadir$  echo "$DISPLAY"
:1
ubuntu@coyote:/opt/mr/nadir$ glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)' 
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 510.108.03
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 510.108.03
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 510.108.03
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
ubuntu@coyote:/opt/mr/nadir$ ps aux | grep Xorg
ubuntu    180400  0.0  0.0   4020  2052 pts/1    S+   13:21   0:00 grep --color=auto Xorg
          ```

<details>

There appears to be a bug, which I believe to be related to the hydrodynamics plugin, which results in a model world velocity run-away. I've experienced this with my personal model, and it is reproducible in the AUV example provided by `[auv_controls.sdf](https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/auv_controls.sdf#L139)`

</details>

## Description
* Expected behavior: Vehicle velocity should decrease upon commanding zero prop speed
* Actual behavior: Vehicle velocity grows unbounded until sim crash

In my own simulation, the problem is not exhibited when the added mass terms are 0, or significantly smaller than the vehicle mass. I have adjusted the sign of the added mass terms in my own model (non-example LRAUV system) and this does not change the outcome.

I haven't had time yet to check the implementation of Fossen's Eqns in [HydrodynamicsPlugins.cc](https://github.com/gazebosim/gz-sim/blob/206def391d6ba23259ce8596e7256c2d705a9e53/src/systems/hydrodynamics/Hydrodynamics.cc#L492).

I believe M_A (added mass matrix) should be positive definite, symmetric. Yet the terms that exist in `[auv_controls.sdf](https://github.com/gazebosim/gz-sim/blob/206def391d6ba23259ce8596e7256c2d705a9e53/examples/worlds/auv_controls.sdf#L144)` result in a decidedly negative definite matrix.

## Steps to reproduce
1. Launch the auv sim: `gz sim auv_controls.sdf -v 4`
2. Run the prop momentarily: `gz topic -t /model/tethys/joint/propeller_joint/cmd_thrust -m gz.msgs.Double -p 'data: -31'`
3. Stop the prop: `gz topic -t /model/tethys/joint/propeller_joint/cmd_thrust -m gz.msgs.Double -p 'data: -0'` 

## Output
You can see a video of the problem manifesting here: [https://youtube.com/shorts/zExnMM8K-L8](https://youtube.com/shorts/zExnMM8K-L8)
Notice the world velocity of the AUV growing unbounded even once the prop has been commanded to stop.

[ogre2.log](https://github.com/gazebosim/gz-sim/files/10936595/ogre2.log)

laughlinbarker, 2023-03-09:
Of note: when commanding the prop in reverse, the vehicle backs up as expected. Upon stopping the prop, the vehicle decelerates (towards zero velocity), but the sim still crashes after a few seconds.

azeey, 2023-03-10:
There was a behavior change introduced recently which has been documented in https://github.com/gazebosim/gz-sim/pull/1888. Can you see if that helps?

laughlinbarker, 2023-03-10:
Thanks @azeey. Appropriate naming of those parameters appears (in my test run of N=1) to prevent the runaway condition.

Parameters used:
```
<plugin
        filename="gz-sim-hydrodynamics-system"
        name="gz::sim::systems::Hydrodynamics">
        <link_name>base_link</link_name>
        <xDotU>-4.876161</xDotU>
        <yDotV>-126.324739</yDotV>
        <zDotW>-126.324739</zDotW>
        <kDotP>0</kDotP>
        <mDotQ>-33.46</mDotQ>
        <nDotR>-33.46</nDotR>

        <xUabsU>-6.2282</xUabsU>
        <yVabsV>-601.27</yVabsV>
        <zWabsW>-601.27</zWabsW>
        <kPabsP>-0.1916</kPabsP>
        <mQabsQ>-632.69</mQabsQ>
        <nRabsR>-632.69</nRabsR>

        <xU>0</xU>
        <yV>0</yV>
        <zW>0</zW>
        <kP>0</kP>
        <mQ>0</mQ>
        <nR>0</nR>
      </plugin>
```

The mis-naming of the second-order (and in the case of the demo, the only non-zero) damping terms, appears to be at the heart of the issue.

Seems like appropriate course of action is:
 - (Minimum) Submit PR to rename Hydro params appropriately in auv_controls.sdf
 - (Better) Above + address this potential for instability at the plugin level. This could be by providing warnings for uninitialized first/second order drag terms, or even by providing a default (but small?) drag term so as to remain stable.
 - (Best) Above + Add some checks in the appropriate unit tests to catch something of this nature?

laughlinbarker, 2023-03-10:
Submitted minimal PR as described above, don't have time at the moment to investigate other options I enumerated.

azeey, 2023-03-11:
/cc @arjo129

arjo129, 2023-03-13:
This was fixed in  https://github.com/gazebosim/gz-sim/pull/1888. The said fix has not yet been forward ported to Garden.

mateusmenezes95, 2023-08-02:
We have experienced the same problem in our project simulation, even using the correct naming for the coefficients of the Damping matrix. After deep investigation, I discovered that numerical computation instability can exist if the condition $M_A \prec M_{RB}$ is unsatisfied, where $M_A$ is the Added Mass matrix, $M_{RB}$ is the system inertia matrix, and $\prec$ is the element wise less than comparison. The authors of the uuv_simulator already reported this error in section II.D of their paper [UUV Simulator: A Gazebo-based package for underwater intervention and multi-robot simulation](https://ieeexplore.ieee.org/document/7761080). Based on the uuv_simulation, an older colleague has also reported this instability in his master thesis [Parameters Identification of Dynamic Model and Simulation of Autonomous Underwater Vehicles], section 4.2.

I tested this criterion using an adapted world file used in the hydrodynamics plugins unit tests in the ign gazebo 6.14.0.
To reproduce the error, first, simulate with the `hydrodynamics.sdf` containing the SDF snippet below using the command `ign gazebo hydrodynamics.sdf`. This first run shall execute without problems. Now, change the value of zDotU (line 108) to -26.0 kg and rerun the simulation. Using this mass value forces the criterion to be unsatisfied because the sphere's mass is 24 kg, and the left superior block of the Added Mass matrix $M_{A_{11}} \in R^{3 \times 3}$ equals $mI_{3 \times 3}$; thus -zDotU > m and $M_A \prec M_{RB}$ is unsatisfied.
In my execution, the simulation crashed after some seconds. Even with zDotU -25.0 kg, my simulation crashed.

Thus, if someone could reproduce the crash and this criterion must be satisfied and is a simple element-wise comparison, my first suggestion is to put a warning or even a fatal error informing the user about the problem.

`hydrodynamics.sdf`
```XML
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="hydrodynamics">

   <physics name="1ms" type="ode">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>

    <gravity>0 0 9.8</gravity>

    <plugin
      filename="ignition-gazebo-physics-system"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="ignition-gazebo-user-commands-system"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
      filename="ignition-gazebo-scene-broadcaster-system"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    <light type="directional" name="sun">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 0 0</pose>
      <diffuse>1 1 1 1</diffuse>
      <specular>0.5 0.5 0.5 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
    </light>

    <model name="sphere1">
      <link name="sphere1_link">
        <pose>1 0.25 0 0 0 0</pose>
        <inertial>
          <mass>25</mass>
          <inertia>
            <ixx>0.4</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>0.4</iyy>
            <iyz>0</iyz>
            <izz>0.4</izz>
          </inertia>
        </inertial>

        <visual name='sphere1_visual'>
          <geometry>
            <sphere>
              <radius>0.2</radius>
            </sphere>
          </geometry>
        </visual>
        <collision name='sphere1_collision'>
          <geometry>
            <sphere>
              <radius>0.2</radius>
            </sphere>
          </geometry>
        </collision>
      </link>
    </model>

    <model name="sphere2">
      <link name="sphere2_link">
        <pose>1 -0.25 0 0 0 0</pose>
        <inertial>
          <mass>25</mass>
          <inertia>
            <ixx>0.4</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>0.4</iyy>
            <iyz>0</iyz>
            <izz>0.4</izz>
          </inertia>
        </inertial>

        <visual name='sphere2_visual'>
          <geometry>
            <sphere>
              <radius>0.2</radius>
            </sphere>
          </geometry>
        </visual>
        <collision name='sphere2_collision'>
          <geometry>
            <sphere>
              <radius>0.2</radius>
            </sphere>
          </geometry>
        </collision>
      </link>

      <plugin
        filename="ignition-gazebo-hydrodynamics-system"
        name="ignition::gazebo::systems::Hydrodynamics">
        <link_name>sphere2_link</link_name>
        <water_density>1030</water_density>
        <xDotU>0</xDotU>
        <yDotV>0</yDotV>
        <zDotW>-24</zDotW> <!-- Change here for -26 -->
        <kDotP>0</kDotP>
        <mDotQ>0</mDotQ>
        <nDotR>0</nDotR>
        <xUabsU>0</xUabsU>
        <xU>0</xU>
        <yVabsV>0</yVabsV>
        <yV>0</yV>
        <zWabsW>-10.5359</zWabsW>
        <zW>0.211869</zW>
        <kPabsP>0</kPabsP>
        <kP>0</kP>
        <mQabsQ>0</mQabsQ>
        <mQ>0</mQ>
        <nRabsR>0</nRabsR>
        <nR>0</nR>
      </plugin>
    </model>
  </world>
</sdf>
```

[Parameters Identification of Dynamic Model and Simulation of Autonomous Underwater Vehicles]: https://repositorio.ufba.br/bitstream/ri/29458/1/Disserta%C3%A7%C3%A3o%20Final%20-%20Jo%C3%A3o%20da%20Costa%20Britto%20Neto%20-%202018.pdf

arjo129, 2023-08-02:
Hi @mateusmenezes95 yes that is correct. I could add a check to the hydrodynamics plugin. This a is a great suggestion.

mateusmenezes95, 2023-08-02:
Very nice @arjo129. Thanks for the fast reply 😄!  Another suggestion may be to use the compensation method (named Method 1) suggested by @joaobrittoneto in his thesis that guarantees stability even when the criterion is unsatisfied. However, as reported in the results, this method may degrade the simulation result when there are body rotations. An option is to turn this compensation on/off by a parameter.

arjo129, 2023-08-02:
I don't have time to add the compensation method but PRs are welcome!

mateusmenezes95, 2023-08-03:
Sure! I will try to submit a pull request with the suggestions.

mateusmenezes95, 2024-06-05:
I think that https://github.com/gazebosim/gz-sim/pull/1592 solves this issue for Gazebo Harmonic. What do you think @azeey?

joaobrittoneto, 2024-07-23:
>I think that https://github.com/gazebosim/gz-sim/pull/1592 solves this issue for Gazebo Harmonic. What do you think @azeey?

I agree with @mateusmenezes95. The [fluid_added_mass](http://sdformat.org/tutorials?tut=added_mass_proposal&cat=pose_semantics_docs) present in the inertial (which leverages [DART's](https://github.com/dartsim/dart/blob/main/dart/dynamics/Inertia.hpp#L123) ability to handle 6x6 spatial inertia tensors) should be favored. It leads to a simpler and more robust solution. The only downturn is the reliance on DART itself (it's not supported by other physics engines).

The added mass effect of the hydrodynamics plugin, IMHO, could be deprecated.
The current implementation, as known, is brittle. Doing [`a[k] = -Mrb^-1 * Ma * a[k-1]`](https://github.com/gazebosim/gz-sim/blob/206def391d6ba23259ce8596e7256c2d705a9e53/src/systems/hydrodynamics/Hydrodynamics.cc#L492) leads to instability when `Ma` > `Mrb`.
Trying to use the compensated terms, as mentioned in the references [here](https://github.com/gazebosim/gz-sim/issues/1923#issuecomment-1661282262), leads to a convoluted solution, vide a custom [gazebo_underwater plugin](https://github.com/rock-gazebo/simulation-gazebo_underwater/blob/master/src/GazeboUnderwater.cpp) I helped to create where those terms are used.

arjo129, 2024-07-24:
I agree with you too. We should deprecate the added mass added by the
hydrodynamics plugin and rely ont the physics engine based one.