gazebosim/gz-sim #1934: GUI-Err: currentGLContext was specified with no current GL context in GLXWindow::create

PSandro, 2023-03-18:
## Environment
* OS Version: Ubuntu 22.04.2 LTS
* Source or binary build? binary through apt: Gazebo Sim, version 7.4.0
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2] tried with both ogre2 and ogre. default ogre2
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [x] running on real hardware
      - [x] GPU is concurrently used for other tasks
        - [?] desktop acceleration
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          06:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Renoir [1002:1636] (rev d1)

          echo "$DISPLAY"
          :0

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require           installing mesa-utils package
          direct rendering: Yes
              Preferred profile: core (0x1)
              Max core profile version: 4.6
              Max compat profile version: 4.6
              Max GLES1 profile version: 1.1
              Max GLES[23] profile version: 3.2
          OpenGL vendor string: AMD
          OpenGL renderer string: RENOIR (renoir, LLVM 15.0.6, DRM 3.47, 5.19.0-35-generic)
          OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.2.5
          OpenGL core profile shading language version string: 4.60
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.2.5
          OpenGL shading language version string: 4.60
          OpenGL context flags: (none)
          OpenGL profile mask: compatibility profile
          OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.2.5
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager

          I'm using sway v1.7 on wayland
          ```
    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
16:33:58: Creating resource group General
16:33:58: Creating resource group Internal
16:33:58: Creating resource group Autodetect
16:33:58: SceneManagerFactory for type 'DefaultSceneManager' registered.
16:33:58: Registering ResourceManager for type Material
16:33:58: Registering ResourceManager for type Mesh
16:33:58: Registering ResourceManager for type Mesh2
16:33:58: Registering ResourceManager for type OldSkeleton
16:33:58: MovableObjectFactory for type 'ParticleSystem' registered.
16:33:58: ArchiveFactory for archive type FileSystem registered.
16:33:58: ArchiveFactory for archive type Zip registered.
16:33:58: ArchiveFactory for archive type EmbeddedZip registered.
16:33:58: DDS codec registering
16:33:58: FreeImage version: 3.18.0
16:33:58: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
16:33:58: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
[... 155 lines cut ...]
#14   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fc2bc20029e, in gz::rendering::v7::RenderEngineManagerPrivate::Engine(EngineInfo, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#13   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fc2bc20ab21, in gz::rendering::v7::BaseRenderEngine::Init()
#12   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7fc1c3dcaac2, in gz::rendering::v7::Ogre2RenderEngine::InitImpl()
#11   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7fc1c3dd06f7, in gz::rendering::v7::Ogre2RenderEngine::InitAttempt()
#10   Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c38eb228, in Ogre::ResourceGroupManager::initialiseAllResourceGroups(bool)
#9    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c38eac9e, in Ogre::ResourceGroupManager::parseResourceGroupScripts(Ogre::ResourceGroupManager::ResourceGroup*)
#8    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c392abac, in Ogre::ScriptCompilerManager::parseScript(Ogre::SharedPtr<Ogre::DataStream>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#7    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c39225b2, in Ogre::ScriptCompiler::compile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#6    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c39222b2, in Ogre::ScriptCompiler::compile(Ogre::SharedPtr<std::__cxx11::list<Ogre::SharedPtr<Ogre::ConcreteNode>, Ogre::STLAllocator<Ogre::SharedPtr<Ogre::ConcreteNode>, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#5    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c3933bdf, in Ogre::MaterialTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#4    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c39349a7, in Ogre::TechniqueTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#3    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c393a2ce, in Ogre::PassTranslator::translate(Ogre::ScriptCompiler*, Ogre::SharedPtr<Ogre::AbstractNode> const&)
#2    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c399e9d9, in Ogre::Technique::createPass()
#1    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c38a6cd0, in Ogre::Pass::Pass(Ogre::Technique*, unsigned short)
#0    Object "/usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1", at 0x7fc1c37d9e3d, in Ogre::Hlms::createDatablock(Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::HlmsMacroblock const&, Ogre::HlmsBlendblock const&, std::vector<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
Segmentation fault (Address not mapped to object [0x220])

```

</details>

I also tried following the steps from [the troubleshooting docs](https://github.com/gazebosim/docs/blob/master/garden/troubleshooting.md#unable-to-create-the-rendering-window) for 'Unable to create the rendering window'. I could not notice any change in behaviour.

It might be of importance that I'm running sway-wm (wayland).
Running `glxgears` or `glxdemo` works fine.

PSandro, 2023-03-22:
Thanks for the reply, I tried running gz with EGL but it gave me the same error. I cannot get vulkan running for now, maybe there's something wrong with my installation: 
```
[GUI] [Msg] Loading plugin [vulkan]
Library [/usr/lib/x86_64-linux-gnu/libvulkan.so] does not export any plugins. The symbol [GzPluginHook] is missing, or it is not externally visible.
[GUI] [Err] [RenderEngineManager.cc:534] Failed to load plugin [vulkan] : couldn't load library on path [/usr/lib/x86_64-linux-gnu/libvulkan.so].
```

I experimented some more and tried the following:
- logging out oft sway
- logging in with standard gnome (wayland)
- now I can run gazebo with ogre/ogre2 without any problems
- logging back in to sway
- now ogre/ogre2 also works in sway :thinking: 

Now it seems less like a bug with gz-sim and more like an issue with my installation - so I guess we can close this issue(?). I don't understand much about graphics and rendering. My guess is that something like the `currentGLContext` get's initialized by GNOME but not by my sway installation.. Maybe we could add a check for this in the code and print an error message in this case.

I still have no clue yet how to fix this so that I don't need to login to gnome everytime. Any ideas?

iche033, 2023-03-23:
hmm I don't know why that happens. If you do figure it out, please post back and we can add it to the troubleshooting guide, thanks.

PSandro, 2023-03-27:
The error message is saying that there's no current GL context. GNOME must be initializing some GL context wich gz-sim then uses. I have the suspicion that sway behaves differently. Luckily in the code there is an option to disable using the current GL Context and creating a new context instead.

In line 658 of gz-gui file `src/plugins/minimal_scene/MinimalScene.cc` the option to use the current GL Context is hardcoded to 1. Maybe by removing this line or explicitly changing it to "0" the issue can be resolved.

```
  if (_graphicsAPI == rendering::GraphicsAPI::OPENGL)
  {
    gzdbg << "Creating gz-rendering interface for OpenGL" << std::endl;
    this->dataPtr->rhiParams["useCurrentGLContext"] = "1";
    this->dataPtr->rhi = std::make_unique<GzCameraTextureRhiOpenGL>();
  }
#ifdef __APPLE__
```
I first need to setup a development environment in order to test that but for now it looks promising.

iche033, 2023-03-27:
> this->dataPtr->rhiParams["useCurrentGLContext"] = "1";

I remember that's had to be done so that the render engine (ogre2) can operate in the GL context created by Qt. This is mainly so that it can pass the render texture to Qt to be displayed in the GUI window. I suspect if you set it to 0, the GUI window may become black.

There was another issue about missing gl context reported in https://github.com/gazebosim/gz-rendering/issues/526 and was fixed by https://github.com/gazebosim/gz-rendering/pull/794. I don't know if it's related or not but the version you are using (7.4.0) should have that fix already.

iche033, 2023-03-27:
> the version you are using (7.4.0) should have that fix already.

actually, The gl context fix in released in **gz-rendering 7.3.0**. It might be worth checking to see if you have the latest version

```
dpkg -l | grep gz-rendering7
```

PSandro, 2023-03-28:
Thanks for the reply, I checked my versions of `libgz-rendering7-*`; they are all on `7.3.0-1~jammy`.

After some more trying I came across a workaround to make sure that gz is started with XWayland. By removing the env `WAYLAND_DISPLAY` it falls back to XWayland which results in gazebo starting just fine.
`env -u WAYLAND_DISPLAY gz sim -v 4 shapes.sdf --render-engine ogre`

iche033, 2023-03-29:
Great glad you found a solution to this. I created a PR to update the troubleshooting page with this info: https://github.com/gazebosim/docs/pull/356. Thanks for reporting back.

closed 2023-04-08 by PSandro

victorcarvesk, 2024-02-06:
I encountered the same issue recently, and none of the solutions provided here worked for me.

## Environment

```
OS: Ubuntu 22.04.3 LTS x86x64 (with Wayland)
Kernel: 6.5.0-15-generic
Gazebo: Ignition Fortress 6.16.0 (also tested with Garden 7.7.0 and both from ROS repo and OSRF repo).
ROS: 2 "Humble" (I work with Robotics)
```

## Expected behavior
Launch Gazebo successfully from both the desktop icon and the terminal.

## Observed behavior
Launching from the desktop icon worked, but attempting to launch from terminals other than gnome-terminal (in my case, [Hyper terminal](https://hyper.is/) and VS Code's terminal) produced the following error:

terminal with "-v 4" option outputs:
[image]

The window created quickly closes after the error (the window borders are different from the Ubuntu native style):
[image]

## Immediate solution

- unset XDG_SESSION_TYPE variable on terminal session where Gazebo will be launched:
```bash
unset XDG_SESSION_TYPE
```

## Discussion

While I understand that everything works fine on Xorg, I'm using Ubuntu on a laptop, and only Wayland recognizes touchpad gestures that significantly improve my productivity (such as opening the app gallery and switching workspaces). Therefore, it's crucial for me to have Gazebo working on Wayland.

After extensive analysis, I discovered that **unsetting the XDG_SESSION_TYPE, which is by default set to "wayland", in the Hyper terminal and VS Code's terminal fixes this issue**. The peculiar thing is that XDG_SESSION_TYPE is also "wayland" in gnome-terminal, where everything works fine, albeit with the following output:

`Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
`

So, unsetting this environment variable in the Hyper/VS Code terminal allows Gazebo to launch perfectly, without the message about ignoring XDG_SESSION_TYPE, since it no longer has a value.

I decided to investigate what's different between gnome-terminal and Hyper/VS Code terminal. Thanks to neofetch, I found a subtle difference. In gnome-terminal, neofetch indicates that the DE is GNOME 42.9, while in Hyper/VS Code, it's Unity.

"neofetch" on gnome-terminal
[image]

"neofetch" on hyper-terminal
[image]

"neofetch" on VS Code terminal
[image]

## Suggestion
- Make Gazebo ignore the XDG_SESSION_TYPE for any unknow DE.

_Ps.: I'm not an expert, so this suggestion could be wrong._

victorcarvesk, 2024-02-07:
I encountered the same issue recently, and none of the solutions provided here worked for me.

## Environment

```
OS: Ubuntu 22.04.3 LTS x86x64 (with Wayland)
Kernel: 6.5.0-15-generic
Gazebo: Ignition Fortress 6.16.0 (also tested with Garden 7.7.0 and both from ROS repo and OSRF repo).
ROS: 2 "Humble" (I work with Robotics)
```

## Expected behavior
Launch Gazebo successfully from both the desktop icon and the terminal.

## Observed behavior
Launching from the desktop icon worked, but attempting to launch from terminals other than gnome-terminal (in my case, [Hyper terminal](https://hyper.is/) and VS Code's terminal) produced the following error:

terminal with "-v 4" option outputs:
[image]

The window created quickly closes after the error (the window borders are different from the Ubuntu native style):
[image]

## Immediate solution

- define XDG_CURRENT_DESKTOP with Gnome default value on terminal session where Gazebo will be launched:
```bash
export XDG_CURRENT_DESKTOP=ubuntu:GNOME
```
*OR*
- define XDG_CURRENT_DESKTOP with Gnome default value on .bashrc (if you are never going to switch to Xorg):
```bash
echo "export XDG_CURRENT_DESKTOP=ubuntu:GNOME" >> ~/.bashrc
```

> _Note: it is necessary to investigate the impact of this on other system applications._

*OR*
- unset XDG_SESSION_TYPE on terminal session where Gazebo will be launched:
```bash
unset XDG_SESSION_TYPE
```

## Discussion

While I understand that everything works fine on Xorg, I'm using Ubuntu on a laptop, and only Wayland recognizes touchpad gestures that significantly improve my productivity (such as opening the app gallery and switching workspaces). Therefore, it's crucial for me to have Gazebo working on Wayland.

After extensive analysis, I discovered using `printenv | grep XDG_CURRENT_DESKTOP` that XDG_CURRENT_DESKTOP has a different value ("Unity") in Hyper/VS Code terminal sessions, instead of "ubuntu:GNOME" present in gnome-terminal, where everything works fine. I believe this makes Gazebo work like it does on Xorg, which will obviously cause fatal errors when on Wayland.

It is important to note that there appears to be a relationship between the values in XDG_CURRENT_DESKTOP and XDG_SESSION_TYPE, since either setting the value in the first to ubuntu:GNOME or resetting the second makes Gazebo work correctly.

Ps.: XDG_SESSION_TYPE has the same value ("wayland") on all terminals, but its value is ignored when XDG_CURRENT_DESKTOP is defined with the value "ubuntu:GNOME":
`
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
`

I decided to investigate a little more what's different between gnome-terminal and Hyper/VS Code terminal. Thanks to neofetch, I found the same difference. In gnome-terminal, neofetch indicates that the DE is GNOME 42.9, while in Hyper/VS Code, it's Unity.

"neofetch" on gnome-terminal
[image]

"neofetch" on hyper-terminal
[image]

"neofetch" on VS Code terminal
[image]

## Suggestion
- Treat "Unity" as "ubuntu:GNOME" when the value in XDG_SESSION_TYPE is "wayland".
- Reevaluate the need to use the XDG_SESSION_TYPE variable, since the gazebo worked fine without a value in it, even when XDG_CURRENT_DESKTOP is set to "Unity" in Wayland.

> _Ps.: I'm not an expert, so this suggestion could be wrong._