gazebosim/gz-sim #1946: INTEGRATION_recreate_entities fails on Ubuntu

scpeters, 2023-03-28:
## Environment
* OS Version: Ubuntu 20.04 and 22.04
* Source or binary build? source build in CI of `gz-sim7` branch

## Description
* Expected behavior: the `INTEGRATION_recreate_entities` test should pass
* Actual behavior: it seg-faults. It has been failing quite frequently on the following CI jobs:
  - https://build.osrfoundation.org/view/ign-garden/job/ignition_gazebo-ci-gz-sim7-focal-amd64/85/testReport/junit/(root)/INTEGRATION_recreate_entities/test_ran/history/
  - https://github.com/gazebosim/gz-sim/actions/workflows/ci.yml?query=branch%3Agz-sim7

## Steps to reproduce

1. Run https://build.osrfoundation.org/view/ign-garden/job/ignition_gazebo-ci-gz-sim7-focal-amd64/ or an Ubuntu CI GitHub actions workflow on the `gz-sim7` branch.
2. Review test results to see if `INTEGRATION_recreate_entities` passed or not
3.

## Output

The earliest failure that I see is from the following job after https://github.com/gazebosim/gz-sim/pull/1868 was merged. The test seg-faults and the console output shows a complaint about a double linked list:

* https://github.com/gazebosim/gz-sim/actions/runs/4357709264/jobs/7617354393

~~~
191/235 Test #191: INTEGRATION_recreate_entities ..........................Child aborted***Exception:   2.09 sec
...
[       OK ] RecreateEntitiesFixture.RecreateEntities (1370 ms)
[ RUN      ] RecreateEntitiesFixture.RecreateEntities_Joints
...
[       OK ] RecreateEntitiesFixture.RecreateEntities_Joints (248 ms)
[ RUN      ] RecreateEntitiesFixture.RecreateEntities_WorldJoint
...
[Msg] Serving world names on [/gazebo/worlds]
  [Msg] Resource path add service on [/gazebo/resource_paths/add].
  [Msg] Resource path get service on [/gazebo/resource_paths/get].
  [Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
  [Msg] Resource paths published on [/gazebo/resource_paths].
  [Msg] Server control service on [/server_control].
  [Msg] Found no publishers on /stats, adding root stats topic
  [Msg] Found no publisherscorrupted double-linked list
~~~

azeey, 2023-03-28:
Should be fixed by https://github.com/gazebosim/gz-physics/pull/498

scpeters, 2023-03-29:
looks fixed to me

closed 2023-03-29 by scpeters