gazebosim/gz-sim #1948: Light visuals do not honor SDF `<visualize>` tag value at startup

sea-bass, 2023-03-30:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? Binary bulild of Fortress

## Description
When you create a world in SDF format, there is the option of not visualizing the light frames by setting the `<visualize>` tags to false.
However, these options do not take effect at startup. You have to toggle them in the GUI to get the setting correct.

## Steps to reproduce
* Create an SDF world with a light which has the `<visualize>false</visualize>` tag on it.
* Launch Gazebo with that world
* You will find that the "Show gizmo" option is off, but the light still visualizes. If you turn it on and off again from the GUI, it will go away properly.

## Output

Notice the "Show gizmo" option is off but the visuals for the lights are still there:
[image]

ahcorde, 2023-05-11:
Hi @sea-bass, I tried the `examples/worlds/lights.sdf` file provided and it's working for me. 

The point visual is not visualized. 

https://github.com/gazebosim/gz-sim/blob/b559eb6c62dd9589890721acde071e9cfc528aa0/examples/worlds/lights.sdf#L55

Can you check this?

sea-bass, 2023-05-11:
@ahcorde we were using spot lights in our case -- does that work for you?

ahcorde, 2023-05-11:
I tried this world and it's working. Can you try this world? Point and spot gizmos are deactivated.

<details>
<summary>light.sdf</summary>
<?xml version="1.0" ?>

<sdf version="1.6">
  <world name="lights">
    <physics name="1ms" type="ignored">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>
    <plugin
      filename="ignition-gazebo-physics-system"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="ignition-gazebo-user-commands-system"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
      filename="ignition-gazebo-scene-broadcaster-system"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>

    <scene>
      <ambient>0.4 0.4 0.4</ambient>
      <background>0.8 0.8 0.8</background>
      <grid>true</grid>
    </scene>

    <light type="directional" name="directional">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>100</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>0.5 0.2 -0.9</direction>
    </light>

    <light type="point" name="point">
      <pose>0 -1.5 3 0 0 0</pose>
      <diffuse>1 0 0 1</diffuse>
      <specular>.1 .1 .1 1</specular>
      <attenuation>
        <range>4</range>
        <linear>0.5</linear>
        <constant>0.2</constant>
        <quadratic>0.01</quadratic>
      </attenuation>
      <cast_shadows>false</cast_shadows>
      <visualize>false</visualize>
    </light>

    <light type="spot" name="spot">
      <pose>0 0.0 3 0 0 0</pose>
      <diffuse>0 1 0 1</diffuse>
      <specular>.2 .2 .2 1</specular>
      <attenuation>
        <range>4</range>
        <linear>0.5</linear>
        <constant>0.2</constant>
        <quadratic>0.01</quadratic>
      </attenuation>
      <direction>0 0 -1</direction>
[... 142 lines cut ...]
          <material>
            <ambient>0.3 0.3 0.3 1</ambient>
            <diffuse>0.3 0.3 0.3 1</diffuse>
            <specular>0.3 0.3 0.3 1</specular>
          </material>
        </visual>

        <light type="point" name="link_light_point">
          <pose>0 0 0.8 0 0 0</pose>
          <diffuse>0 0 1 1</diffuse>
          <specular>.1 .1 .1 1</specular>
          <attenuation>
            <range>2</range>
            <linear>0.02</linear>
            <constant>0.05</constant>
            <quadratic>0.01</quadratic>
          </attenuation>
          <cast_shadows>false</cast_shadows>
        </light>
      </link>
    </model>
  </world>
</sdf>

</details>

ahcorde, 2023-06-09:
Hi @sea-bass, 

can you try this https://github.com/gazebosim/gz-rendering/pull/862 ?

sea-bass, 2023-06-11:
> Hi @sea-bass,
> 
> can you try this [gazebosim/gz-rendering#862](https://github.com/gazebosim/gz-rendering/pull/862) ?

Haha -- always the subtle things. PR LGTM, though I see it already was merged.

Thank you!

closed 2023-06-12 by ahcorde