gazebosim/gz-sim #1955: Installation Older Gazebo from Source

lucarei, 2023-04-01:
Hi. I am sorry for the simple question but i am newbie. I am on Ubuntu 20.04.
I need to install a specific version of gz-sim with a given commit hash (2938ede): so i must download and compile the sources (not apt get).

I used the following command to clone it (i am not sure they are right):
`git clone https://github.com/gazebosim/gz-sim/`
`git reset --hard 2938ede`

Then i installed package dependencies (as shown in [:](https://gazebosim.org/api/gazebo/6.9/install.html) and i compile.

Unfortunately, it doesn't work properly. Could you help me to do it correctly?

mjcarroll, 2023-04-03:
Can you be more specific about what doesn't work properly?  Console outputs or detailed information allow us to help diagnose.

azeey, 2023-10-16:
Closing since there hasn't been any activity. Please feel free to reopen.

closed 2023-10-16 by azeey