gazebosim/gz-sim #1958: Custom lidar sensor

rsc9421, 2023-04-04:
Hi, i want to simulate livox lidars (xyz) but they have a special "rosette" type pattern. I've read the documentation but I can't find how to simulate individual GpuRayPlugin as it could be done in gazebo classic.

I have the .csv files that indicate where each lidar should point, so all I need to simulate it is:
* set the inclination angles of each ray
* that each ray returns the distance so I can calculate the point (x,y,z)
* I need to use the gpu because it is 100k points per second

If you can help me with a little example I could make a contribution

mjcarroll, 2023-04-06:
> Also I think it would've been better if you had created the issue in the [gz-sensors](https://github.com/gazebosim/gz-sensors) project so that it would reach the experts for sensor implementations directly wink

I think just about everybody on the gazebo-dev team is subscribed to all the repos, so it will get seen no matter what :)

mjcarroll, 2023-04-06:
In addition to what @Bi0T1N said above, from a rendering perspective, the actual implementation changes that will be needed are:

The underlying gz::rendering::Ogre2GpuRays sensor will need to be adjusted to take a new set of "direction vectors" for each ray of your custom pattern.  
    * The place that this is expanded out is here: https://github.com/gazebosim/gz-rendering/blob/c2e72ee51a7e4dba5156faa96c972c63ca5ab437/ogre2/src/Ogre2GpuRays.cc#L827-L950  
    * Right now, this assumes a uniform distribution of points over a window (min -> max, stepping with hStep).  There is a little bit more work because this is actually handled with multiple rendering cameras arranged in a box, so you have to associate with the correct face of the "cube"
    

You may want to experiment a bit by implementing a custom sensor before going through the effort of getting it all "standardized" through sdformat and the rest of the libraries.

ntfshard, 2024-10-14:
> I came across following plugin. Posting here as I assume it could be a solution, or a relevant example: https://robotec.ai/gazebo-lidar-rgl-plugin/

Seems link is outdated:

We Couldn’t Find This Page
Check out some of the other great posts in this blog.

jrutgeer, 2024-10-14:
@ntfshard Well maybe you could [google it]( https://www.google.com/search?q=robotec+gazebo+lidar+rgl+plugin) and post the link to the GitHub repository here?