gazebosim/gz-sim #1961: ignition gazebo is not simulating the dynamics correctly

xibeisiber, 2023-04-10:
## Environment
* OS Version: Ubuntu 22.04, 
* Source or binary build?
  binary: ignition gazebo (Fortess) 6.11.0, gazebo 11.10.2.

## Description
in the sdf, there are two boxes. I changed the contact attribute as follows:
```
             <contact>
              <ode>
                <min_depth>0.3</min_depth>
                <kp>1e+03</kp>
                <kd>0</kd>
                <max_vel>100</max_vel>
              </ode>
            </contact>
```
* Expected behavior: there will be some bouncing and penetration when the small box falls upon the bigger one.
* Actual behavior: the bouncing and penetration appears in class gazebo 11.10.2, but not in ignition gazebo (Fortess) 6.11.0

## Steps to reproduce
load the sdf ([file link](https://github.com/xibeisiber/pics/blob/master/gazebo/t.sdf)) by:
ign gazebo t.sdf      ([video link](https://github.com/xibeisiber/pics/blob/master/gazebo/ign_gazebo.mp4))
gazebo t.sdf            ([video link](https://github.com/xibeisiber/pics/blob/master/gazebo/gazebo.mp4))

azeey, 2023-04-11:
Gazebo classic uses the ODE physics engine by default whereas the new Gazebo uses DART. The parameters you mentioned are not supported in DART, AFAIK, so I don't know if it can be fixed. We definitely need to document that they are not supported though.

azeey, 2023-04-12:
The new Gazebo does not support ODE at all, so your change won't enable ODE. If you're interested only in bouncing dynamics, you'll want to set the `restitution_coefficient` (http://sdformat.org/spec?ver=1.6&elem=collision#bounce_restitution_coefficient)

xibeisiber, 2023-04-14:
> The new Gazebo does not support ODE at all, so your change won't enable ODE. If you're interested only in bouncing dynamics, you'll want to set the `restitution_coefficient` (http://sdformat.org/spec?ver=1.6&elem=collision#bounce_restitution_coefficient)

I am simulating the hybrid force/position control and want the object surface to be less stiff (like a spring) so that the contact force will not be too large.

I'm not sure whether adjusting the restitution_coefficient will produce moderate contact force (will try later). 

Since ignition gazebo is using DART, I use the "soft_contact" tag and set the "stiffness" and "damping" tag in [t_dart.sdf](https://github.com/xibeisiber/pics/blob/master/gazebo/t_dart.sdf) following the instruction in http://sdformat.org/spec?ver=1.6&elem=collision#surface_soft_contact. Why these two parameters still do not work?

Mads-MA, 2023-04-17:
> I am simulating the hybrid force/position control and want the object surface to be less stiff (like a spring) so that the contact force will not be too large.

I have been looking into this since I need to do something similar to the quote above.
AFAIK, SDFormat does not parse or even attempt to read the soft_contact element. Furthermore, gz-physics does not use the soft_contact either.

I attempted to add support for soft contact/soft body myself. 
However, even though my addition to SDFormat seems to parse the soft_contact element correctly, gz-physics does not. I will gladly do a pull request once it is done, if someone is willing to help/guide me a bit.

xibeisiber, 2023-04-25:
Is there any progress or schedule on this?

azeey, 2023-05-01:
> Since ignition gazebo is using DART, I use the "soft_contact" tag and set the "stiffness" and "damping" tag in [t_dart.sdf](https://github.com/xibeisiber/pics/blob/master/gazebo/t_dart.sdf) [...]

There's an open issue (https://github.com/gazebosim/gz-physics/issues/222) for this, but there hasn't been any progress. There's also a related issue (https://github.com/gazebosim/gz-physics/issues/44) for bullet. If you are interested in contributing, there's some guidance [here](https://github.com/gazebosim/gz-physics/issues/44#issuecomment-1337904151)

> Is there any progress or schedule on this?

This is not currently on our [roadmap](https://gazebosim.org/docs/garden/roadmap), but we'd be glad to review contributions from the community.