gazebosim/gz-sim #1970: Low respond from maintainers at gazebo-classic repo.

talregev, 2023-04-25:
3 weeks ago I created a PR that compile classic gazebo in windows via cmake with vcpkg tool.
It compile on the dependencies of gazebo from source, then generate with cmake a compilation for windows.
You can take a look on the PR: https://github.com/gazebosim/gazebo-classic/pull/3312
My issue is that even I tag the maintainer, waited patiently, also we talk on other Issue, and he say he will look at my PR it didn't happen yet. And it not just me, you can look all other PRs, there is a very low respond.
Yes classic gazbo is the older version, but I manage to create a system (with help of vcpkg repo and their work) that will able to compile classic gazebo in windows without to install anything, and hopefully bring it to other os like linux and osx. I am also thinking to create the same for this repo that developer can easily can compile and debug and even run their result and able to contribute to this community. 
My will is not just review my PR. I hope you do. Is to create higher respond from maintainers in gazebo-classic repo. At least that they review PRs once per week.
I already have another continue PR and I don't want to create another issue here just to get notice for my PRs.
Also there is other PRs in the classic gazebo that waiting. 
I am tag here a random maintainers from here and the maintainers from classic gazebo:
@azeey @mjcarroll @iche033 @scpeters @traversaro

azeey, 2023-04-27:
Hi @talregev, please refrain from randomly tagging people on an issue. It's quite disruptive and unproductive. The gazebo team is focused on the new Gazebo and with the very limited resources we have to invest in Gazebo-classic, we prioritize critical bug fixes. If the team deems your PR critical, they will review it in due time. In the mean time, would you be able to review/test some of the other open pull requests?

closed 2023-04-27 by azeey

talregev, 2023-04-28:
Are you asking me a question and close this issue?
Do you want my replay to the things you write here?
If you want to make a discussion with me, please open this issue.