gazebosim/gz-sim #1987: Gazebo-Garden Vanilla installation both from biary as well as source ,GUI is crashing with segmentation fault in MAcOS BigSur

bhagathkumar, 2023-05-09:
## Environment
* OS Version:  MacOS Bigsur 11.7.1
* Source or binary build?
Tried both binary as well as source build as mentioned in the below links

https://gazebosim.org/docs/garden/install_osx
https://gazebosim.org/docs/garden/install_osx_src

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      ─ system_profiler SPDisplaysDataType                                        ─╯
Graphics/Displays:

    Intel UHD Graphics 630:

      Chipset Model: Intel UHD Graphics 630
      Type: GPU
      Bus: Built-In
      VRAM (Dynamic, Max): 1536 MB
      Vendor: Intel
      Device ID: 0x3e9b
      Revision ID: 0x0000
      Automatic Graphics Switching: Supported
      gMux Version: 5.0.0
      Metal Family: Supported, Metal GPUFamily macOS 2
      Displays:
        Color LCD:
          Display Type: Built-In Retina LCD
          Resolution: 3072 x 1920 Retina
          Framebuffer Depth: 24-Bit Color (ARGB8888)
          Main Display: Yes
          Mirror: Off
          Online: Yes
          Automatically Adjust Brightness: Yes
          Connection Type: Internal

    AMD Radeon Pro 5300M:

      Chipset Model: AMD Radeon Pro 5300M
      Type: GPU
      Bus: PCIe
      PCIe Lane Width: x8
      VRAM (Total): 4 GB
      Vendor: AMD (0x1002)
      Device ID: 0x7340
      Revision ID: 0x0043
      ROM Revision: 113-D3220E-190
      VBIOS Version: 113-D32207P1-020
      Option ROM Version: 113-D32207P1-020
[... 165 lines cut ...]
#15   Object "libsystem_pthread.dylib", at 0x7fff2078d8fc, in _pthread_start + 224
#14   Object "QtCore", at 0x10e570da0, in QThread::qt_metacall(QMetaObject::Call, int, void**) + 1168
#13   Object "QtQuick", at 0x10c38a17f, in QSGOpenGLLayer::qt_metacall(QMetaObject::Call, int, void**) + 21951
#12   Object "QtQuick", at 0x10c388f3c, in QSGOpenGLLayer::qt_metacall(QMetaObject::Call, int, void**) + 17276
#11   Object "QtQuick", at 0x10c3e1a8a, in QQuickWindowPrivate::renderSceneGraph(QSize const&, QSize const&) + 954
#10   Object "QtQuick", at 0x10c319d7a, in QSGRenderer::renderScene(unsigned int) + 58
#9    Object "QtQuick", at 0x10c319fab, in QSGRenderer::renderScene(QSGBindable const&) + 539
#8    Object "QtQuick", at 0x10c332fd6, in QSGBatchRenderer::Renderer::render() + 1958
#7    Object "QtQuick", at 0x10c330c65, in QSGBatchRenderer::Renderer::renderBatches() + 1125
#6    Object "QtQuick", at 0x10c32e444, in QSGBatchRenderer::Renderer::prepareRenderMergedBatch(QSGBatchRenderer::Batch*, QSGBatchRenderer::Renderer::PreparedRenderBatch*) + 1076
#5    Object "QtQuick", at 0x10c32d80a, in QSGBatchRenderer::Renderer::updateMaterialDynamicData(QSGBatchRenderer::ShaderManagerShader*, QSGMaterialRhiShader::RenderState&, QSGMaterial*, QVarLengthArray<QRhiShaderResourceBinding, 8>*, QSGBatchRenderer::Batch const*, int, int) + 122
#4    Object "QtQuick", at 0x10c36fa0f, in QSGRenderLoop::metaObject() const + 16815
#3    Object "QtQuick", at 0x10c36f870, in QSGRenderLoop::metaObject() const + 16400
#2    Object "libsystem_platform.dylib", at 0x7fff207d2d7c, in _sigtramp + 28
#1    Object "libgz-tools2-backward.dylib", at 0x109ab5fad, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 13
#0    Object "libgz-tools2-backward.dylib", at 0x109ab6016, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 70
[1]    26969 segmentation fault 

## Steps to reproduce

1.
2.
3.

## Output

fenil1608, 2023-05-17:
Hi, I'm also facing similar issue when trying to use gz-garden in MACOS Monterey Version 12.3, Apple M1 pro silicon. 

After installation via binaries, I'm facing below errors & the simulation won't run. 
[image]

And after installation via building from source, I'm getting similar segmentation fault. 

[image]

srmainwaring, 2023-05-17:
> Hi, I'm also facing similar issue when trying to use gz-garden in MACOS Monterey Version 12.3, Apple M1 pro silicon.

@fenil1608 Gazebo will run on a M1 MacBook Pro using macOS Monterey (I run 12.6.2 but the earlier version should also be ok).

The problem you are seeing is different to the OP, and seems to be an issue with resolving library paths. Some suggestions:

1. Ensure  you have uninstalled all the brew binaries for Gazebo including `sdformat`.
2. You need to build with RPATH disabled. I also enable a couple of other flags to ensure that my Gazebo build works with ROS Humble.

```bash
colcon build --symlink-install --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=ON -DCMAKE_CXX_STANDARD=17 -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib -DGZ_PYTHON_VERSION=3.10 --event-handlers=desktop_notification- --packages-up-to gz-sim7
```

- `--merge-install` as otherwise the libs will not be found event though `./install/setup.zsh` does point to the correct places. This is all because of the `DYLD_LIBRARY_PATH` sanitisation enforced by macOS when spawning child shells. 
- `-DCMAKE_MACOSX_RPATH` is needed to ensure absolute install paths
- `-DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib` to force the install location
- `-DGZ_PYTHON_VERSION=3.10` because brew installs multiple version of Python and you want to be sure which version you're going to get
- `--event-handlers=desktop_notification-` is to suppress warnings from `colcon`

bhagathkumar, 2023-05-17:
Hi @srmainwaring ,

wether above solution will fix my initial problem statement? Bigsur with i7 config? Though the error output is different for me

srmainwaring, 2023-05-17:
> wether above solution will fix my initial problem statement? Bigsur with i7 config? Though the error output is different for me

Hi @bhagathkumar, maybe not. Your problem appears to not be so much with library resolution but with the rendering set up and possible the interaction with Qt. 

Which model of mac are you using - I have previously built and run `gz-sim` on a 2013 MacBook Pro, macOS Big Sur 11.6.1 with an Intel Iris 1536MB GPU and Retina display. I haven't tried building on this machine since `gz-sim` updated to used `ogre2.3`, so there may be some changes causing issue on older machines.

srmainwaring, 2023-05-17:
> @srmainwaring Have you had any luck running Gazebo installed through homebrew?

I haven't tried recently as I am always using from source builds for dev and updating brew tends to wreck my ROS Humble stack every time. Could try my M1 machine as I am planning to pilot Ventura on that soon.

bhagathkumar, 2023-05-18:
@srmainwaring  I have macbook pro 2019 edition with I7 processor .current version BigSur 11.7.1 .Gpu Iris 1536 MB and Radeon 5300M 4GB.

Hope ogre2.3 have compatibility issues with the version.

i tried with --render-engine ogre .. Still it didn't work.

fenil1608, 2023-05-22:
> > Hi, I'm also facing similar issue when trying to use gz-garden in MACOS Monterey Version 12.3, Apple M1 pro silicon.
> 
> @fenil1608 Gazebo built from source will run on a M1 MacBook Pro using macOS Monterey (I run on Monterey 12.6.2 but the earlier version should also be ok).
> 
> The problem you are seeing is different to the OP, and seems to be an issue with resolving library paths. Some suggestions if you build from source:
> 
> 1. Ensure  you have uninstalled all the brew binaries for Gazebo including `sdformat`.
> 2. You need to build with RPATH disabled. I also enable a couple of other flags so I don't need to disable SIP.
> 
> ```shell
> colcon build --symlink-install --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=ON -DCMAKE_CXX_STANDARD=17 -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib -DGZ_PYTHON_VERSION=3.10 --event-handlers=desktop_notification- --packages-up-to gz-sim7
> ```
> 
> * `--merge-install` as otherwise the libs will not be found event though `./install/setup.zsh` does point to the correct places. This is all because of the `DYLD_LIBRARY_PATH` sanitisation enforced by macOS when spawning child shells.
> * `-DCMAKE_MACOSX_RPATH` is needed to ensure absolute install paths
> * `-DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib` to force the install location
> * `-DGZ_PYTHON_VERSION=3.10` because brew installs multiple version of Python and you want to be sure which version you're going to get
> * `--event-handlers=desktop_notification-` is to suppress warnings from `colcon`

Hi @srmainwaring 
After cleaning up the binaries for gazebo installed by homebrew & rebuilding the RPATH disabled & other options you've provided. 
The gui starts up, but there still seems to be some issue with rendering wherein there's nothing shown up on the gui screen & there's this Error on the Server screen " NodeShared::RecvSrvRequest() error sending response: Host unreachable " 

Attaching screenshots below, 
[image]

[image]

bhagathkumar, 2023-05-24:
Hi @srmainwaring 

Did reset the homebrew. Reinstalled the dependencies. dartsim 6.13.0_3 and qt 5.15.8_3.
Did the Source build

Gazebo Sim Server v7.5.0
Gazebo Sim GUI    v7.5.0

with default gui.config ,It crashed as in Original issue description.

1. After removing component inspector plugin . It worked as expected .with shapes.sdf 

when quitting 
it exited with below message
[GUI] [Wrn] [Application.cc:859] [QT] file::qml/GzCard.qml:139: TypeError: Cannot read property 'pluginToolBarTextColorLight' of null
Stack trace (most recent call last) in thread 123145422659584:
#2    Object "libsystem_platform.dylib", at 0x7fff2089ed7c, in _sigtramp + 28
#1    Object "libgz-tools2-backward.dylib", at 0x106605fad, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 13
#0    Object "libgz-tools2-backward.dylib", at 0x106606016, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 70
[1]    31737 segmentation fault  gz sim -v 4 -g

2.When using minimal gui.config .It works as expected with shapes.sdf   

So yea it might be a problem with component Inspector plugin

srmainwaring, 2023-05-25:
@azeey is there a Qt/QML expert who might take a look at this issue or be available to discuss? I don't think this is a threading error after all as the objects are all on the UI thread and this problem happens even when the `MinimalScene` is unloaded - so no Gz render thread conflicts.

I suspect there is something going on with the QML - specifically where `Loader` is used to dynamically load the `GzSpinBox` into `Items`. I've narrowed the fault down to a couple of QML documents in the `ComponentInspector`: `Light.qml`, `Physics.qml`, `Pose3d.qml` all where they directly or indirectly attempt to load `GzSpinBox`.

The heap corruption error is occurring in the QSG render thread (entirely managed by Qt), and at first I thought that there might be an issue because different Qt control libraries are in use: `QtQuick.Controls 1.4` and `QtQuick.Controls 2.2` are both in the global namespace in many QML files. After scoping the `QtQuick.Controls 1.4` using `import as` and updating the relevant files the issue persists, so that may not be the entire problem (although scoping the imports is probably a good idea in any case).

While the app will run if the `ComponentInspector` is not loaded, at present macOS users running `gz-sim 7.5.0` and brew latest (`qt 5.15.8_3`) may experience a hard seg fault when starting the GUI using the defaults - whether from source or binary.

azeey, 2023-05-25:
Wow, great job investigating this! https://doc.qt.io/qt-5/qtquickcontrols1-index.html says QtQuick Controls 1 is deprecated as of qt 5.12, so we should just update these to use version 2 anyways.

NDevDrone, 2023-05-26:
> > So yea it might be a problem with component Inspector plugin

Hey @srmainwaring , thank you for everything you do. Hard to understate how valuable you are. I reproduced this as well.

I recently tried to install garden from binary (brew) like you suggested a while back and I encounter the same issue as @fenil1608 , and I ensured that SIP was turned off ahead of time. Very stuck. https://github.com/gazebosim/gz-sim/issues/1990 . I am on Ventura.

srmainwaring, 2023-05-31:
> so we should just update these to use version 2 anyways

This is not straight forward. The version of SpinBox in in QtQuick.Controls 2.2 is for ints rather than doubles. The suggested adaptation in https://doc.qt.io/qt-5/qml-qtquick-controls2-spinbox.html#validator-prop for dealing with floating point numbers is not a practical solution as it overflows / underflows for the full range of doubles / decimals points currently used.

Some further discussion here: https://stackoverflow.com/questions/43406830/how-to-use-float-in-a-qml-spinbox in the second comment.

A custom control is probably required.

srmainwaring, 2023-06-02:
> Very stuck. #1990 . I am on Ventura.

@NDevDrone - have just upgraded to Ventura 13.4, Xcode and CommandLineTools 14.3.1, updated brew and rebuilt gz-sim7 - all working ok. Will see if I can get the binary install to run.

[image]

NDevDrone, 2023-09-05:
> 

Hi Rhys, 

Thanks for reviewing it. Some new work has come up involving simulation, which has lead me to try once again. Building from source on Ventura with updated tools yields the following:

[image]

srmainwaring, 2023-09-05:
> Building from source on Ventura with updated tools yields the following:

The `colcon notifier` message is not the main problem, although should be fixed by updating `brew` and `Python` packages.

For some reason `gz-msgs9` is not building, but the log for that is missing. Can you paste the full log as text (easier to read than screen shots).

NDevDrone, 2023-09-05:
Sorry about that, here is the full logs: 

First, running graph gives: 

`colcon graph
gz-cmake3       +***************
gz-tools2        +              
gz-utils2         +**.*******.**
gz-math7           +*****.******
gz-common5          +  ** ******
gz-msgs9             + * *  ****
sdformat13            +   *  **.
gz-fuel_tools8         +      *.
gz-plugin2              + ***.**
gz-transport12           +  ****
gz-physics6               +   *.
gz-rendering7              +***.
gz-gui7                     + **
gz-sensors7                  +*.
gz-sim7                       +*
gz-launch6                     +`

I then use either of the colcon build commands, the first one OR the one that has arguments for apple sillicon macs. Running the second yields the following log: 

`USERNAME@USERNAME-MacBook-Pro-2 workspace % colcon build --cmake-args -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib --merge-install
Starting >>> gz-cmake3
--- stderr: gz-cmake3                               
warning: Tag 'OUTPUT_TEXT_DIRECTION' at line 102 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLANG_ASSISTED_PARSING' at line 1072 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_OPTIONS' at line 1080 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_DATABASE_PATH' at line 1090 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'HTML_TIMESTAMP' at line 1237 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'FORMULA_TRANSPARENT' at line 1529 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_SOURCE_CODE' at line 1837 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_TIMESTAMP' at line 1853 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'RTF_SOURCE_CODE' at line 1927 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOCBOOK_PROGRAMLISTING' at line 2032 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLASS_DIAGRAMS' at line 2219 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTNAME' at line 2261 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTSIZE' at line 2268 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_TRANSPARENT' at line 2494 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'OUTPUT_TEXT_DIRECTION' at line 102 of file '/Users/USERNAME/workspace/build/gz-cmake3/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLANG_ASSISTED_PARSING' at line 1072 of file '/Users/USERNAME/workspace/build/gz-cmake3/api.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
[... cut ...]
Aborted  <<< sdformat13 [2min 4s]                                                                                         
Aborted  <<< gz-msgs9 [2min 23s]                                          

Summary: 4 packages finished [2min 38s]
  1 package failed: gz-common5
  2 packages aborted: gz-msgs9 sdformat13
  7 packages had stderr output: gz-cmake3 gz-common5 gz-math7 gz-msgs9 gz-tools2 gz-utils2 sdformat13
  9 packages not processed
[158.328s] ERROR:colcon.colcon_notification.desktop_notification.terminal_notifier:Could not find the colcon-terminal-notifier.app in the install prefix '/opt/homebrew'
`

NDevDrone, 2023-09-05:
and here is the output from the standard colcon merge install: 

USERNAME@USERNAME-MacBook-Pro-2 workspace % colcon build --merge-install
Starting >>> gz-cmake3
--- stderr: gz-cmake3                               
warning: Tag 'OUTPUT_TEXT_DIRECTION' at line 102 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLANG_ASSISTED_PARSING' at line 1072 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_OPTIONS' at line 1080 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_DATABASE_PATH' at line 1090 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'HTML_TIMESTAMP' at line 1237 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'FORMULA_TRANSPARENT' at line 1529 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_SOURCE_CODE' at line 1837 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'LATEX_TIMESTAMP' at line 1853 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'RTF_SOURCE_CODE' at line 1927 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOCBOOK_PROGRAMLISTING' at line 2032 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLASS_DIAGRAMS' at line 2219 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTNAME' at line 2261 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_FONTSIZE' at line 2268 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'DOT_TRANSPARENT' at line 2494 of file '/Users/USERNAME/workspace/build/gz-cmake3/api_tagfile.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'OUTPUT_TEXT_DIRECTION' at line 102 of file '/Users/USERNAME/workspace/build/gz-cmake3/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'CLANG_ASSISTED_PARSING' at line 1072 of file '/Users/USERNAME/workspace/build/gz-cmake3/api.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_OPTIONS' at line 1080 of file '/Users/USERNAME/workspace/build/gz-cmake3/api.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'CLANG_DATABASE_PATH' at line 1090 of file '/Users/USERNAME/workspace/build/gz-cmake3/api.dox' belongs to an option that was not enabled at compile time.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u", or recompile doxygen with this feature enabled.
warning: Tag 'HTML_TIMESTAMP' at line 1237 of file '/Users/USERNAME/workspace/build/gz-cmake3/api.dox' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
[... cut ...]
Failed   <<< gz-common5 [5.48s, exited with code 2]
Aborted  <<< gz-msgs9 [8.87s]                                  

Summary: 5 packages finished [14.1s]
  1 package failed: gz-common5
  1 package aborted: gz-msgs9
  7 packages had stderr output: gz-cmake3 gz-common5 gz-math7 gz-msgs9 gz-tools2 gz-utils2 sdformat13
  9 packages not processed
[14.181s] ERROR:colcon.colcon_notification.desktop_notification.terminal_notifier:Could not find the colcon-terminal-notifier.app in the install prefix '/opt/homebrew'

srmainwaring, 2023-09-05:
Looks like you may be missing / need to upgrade some deps:

```bash
brew update
brew upgrade
brew install gts swscale uuid
```

There may be others.

Also check you're Python packages are up to date

```bash
python3 -m pip list --outdated 
```

and fix as necessary.

It's worth patching `gz-cmake3` and a couple of other repos to suppress the doxygen warnings that obscure what is going on. Search for `DOT_TRANSPARENT` etc and remove from the doxygen config files. It will save pages of screen space.

My intel mac workstation is current with brew latests and Python as of the beginning of this week, and gz garden is building and running. My M1 MacBook has not been updated for a few months, so there may be inconsistencies...

NDevDrone, 2023-09-05:
> ```shell
> brew update
> brew upgrade
> brew install gts swscale uuid
> ```

Hey Rhys, 

Checked all this to be sure, still no luck. All packages are up to date. 

I am not sure how to go about patching gz-cmake3 on my own, this is beyond me in terms of level of developer skill. 

Also sorry for the massive wall of text in prior messages, I will delete them now to fix this issue's comment section to retain its readability.

srmainwaring, 2023-09-05:
@NDevDrone - ok, next step will be for me to carry out the same upgrade to a M1 machine as applied to the intel one to verify the build is good.

Are all your branches of gz-* current? To replicate I'll need the branch and hash for each of the gz-* libraries. Could you run the following from your workspace and upload the files.

```bash
vcs export ./src > gz-garden.yaml
```

```bash
vcs export --exact ./src > gz-garden-exact.yaml
```

NDevDrone, 2023-09-05:
Oh my, somehow I've managed to get it to build from source!!!

In my frustration, I uninstalled (wiped) homebrew entirely. This time around, I first set the export dependencies for ogre in advance within my system shell, then I ran YOUR suggested colcon build command, i.e.

``` colcon build --symlink-install --merge-install --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DBUILD_TESTING=ON -DCMAKE_CXX_STANDARD=17 -DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib -DGZ_PYTHON_VERSION=3.10 --event-handlers=desktop_notification- --packages-up-to gz-sim7 ```

This then appeared TO BUILD EVERYTHING PROPERLY (!)

After that, I tried to then run the main build command as is to see if it would fail like before, ``` colcon build --merge-install ```

And it built again properly!

I have no idea why your suggested colcon build arguments set was required to fix it, but it did.

NDevDrone, 2023-09-05:
> Could you run the following from your workspace and upload the files.

Attached are the requested files AFTER I managed to get it to work thanks to your help :) 

[RequestedYaml.zip](https://github.com/gazebosim/gz-sim/files/12528778/RequestedYaml.zip)

[image]

I also noticed a few other quirks with the MacOS build from source instructions that we ought to update: 

1. wget isn't installed by default w/ hombrew
2. There is something funky about the colcon build commands on a Apple Silicoin that needs to be identified, but your suggested arguments did the fix so I figure you may see it, and we could update the instructions to reflect this sage information. 
3. Perhaps setting up dependencies in bash script for the likes of Ogre may need to be done in advance of the set up to set up paths correctly?

NDevDrone, 2023-09-05:
@srmainwaring I'll send you a message on ArduPlane community forums, I'm now going to work to finalize the aircraft model I've been working on so I can contribute it to your project, thanks to all this!

srmainwaring, 2023-09-05:
Some background on the flags:

- `-DBUILD_TESTING=ON`, not required but useful when developing
- `-DCMAKE_BUILD_TYPE=RelWithDebInfo`, not required but essential if you want to step through the code in `lldb`
- `-DCMAKE_CXX_STANDARD=17`, should be default for most `gz` libs, but force
- `--merge-install`, `-DCMAKE_MACOSX_RPATH=FALSE -DCMAKE_INSTALL_NAME_DIR=$(pwd)/install/lib`, get round issues with macOS SIP - this force copies the install/lib location into plugins etc so the run time lib loader can find things
-  `-DGZ_PYTHON_VERSION=3.10`, not needed, floating round because of a python project I have in the workspace
-  `--event-handlers=desktop_notification-`, suppress an annoying message which seems to have been resolved with recent versions of brew, python, colcon and other deps.

srmainwaring, 2023-09-05:
> @srmainwaring I'll send you a message on ArduPlane community forums, I'm now going to work to finalize the aircraft model I've been working on so I can contribute it to your project, thanks to all this!

Super! Look forward to seeing that.