gazebosim/gz-sim #1990: Problems using "gz sim" command in MacOS terminal app

FedeCata, 2023-05-12:
After I followed the source installation guide for macOS Monterey step by step, I have a problem using the command "gz sim" in  the terminal app. It appears the following error message: 

```
Library error for [/Users/federicocatalini/workspace/install/lib/libgz-sim7-gz.7.4.0.dylib]: dlopen(/Users/federicocatalini/workspace/install/lib/libgz-sim7-gz.7.4.0.dylib, 0x0009): Symbol not found: (__ZN2gz3sim2v710components7Factory8InstanceEv)
  Referenced from: '/Users/federicocatalini/workspace/install/lib/libgz-sim7-gz.7.4.0.dylib'
  Expected in: '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/libgz-sim7.7.4.0.dylib'

If this script was executed with /usr/bin/ruby, this error may be caused by
macOS System Integrity Protection. One workaround is to use a different
version of ruby, for example:
    brew install ruby
and add the following line to your shell profile:
    export PATH=/usr/local/opt/ruby/bin:$PATH
If you are using a colcon workspace, please ensure that the setup script
has properly set the DYLD_LIBRARY_PATH environment variables.
```

I tried the suggested workarounds but the problem persists. How can I solve it?

scpeters, 2023-05-15:
From the error message, it looks like you have both from-source and binary installations of gz-sim 7.4.0 and they appear to be conflicting. Specifically, I'm noticing the following files that appear to be in conflict:

* /Users/federicocatalini/workspace/install/lib/libgz-sim7-gz.7.4.0.dylib
* /usr/local/Cellar/gz-sim7/7.4.0_2/lib/libgz-sim7.7.4.0.dylib

For a quick test, you can try unlinking the binary package with `brew unlink gz-sim7`, which should avoid that conflict.

You should also take care with which `gz` instance is invoked. If you built the entire Garden workspace in your colcon workspace, then its `gz` instance should be able to find the `gz sim` command files without trouble. However, if the `brew` version of `gz-tools2` is in use, then you would need to set the `GZ_CONFIG_PATH` to ensure it is finding the commands in your workspace.

FedeCata, 2023-05-16:
Thank you for your reply! Anyway, I had noticed the conflict and I proceeded uninstalling both gazebo and gazebo garden and reinstalling only gazebo garden, following the binary installation guide for MacOS on Gazebo Garden Docs website. After the installation, the command `gz` works well, but when I use `gz sim -s` to start the server, I have another issue, which I show below:
```
Error while loading the library [/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: '@rpath/libgz-sim7.7.dylib'
  Referenced from: '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.4.0.dylib'
  Reason: tried: '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::Physics]
- Requested library name: [gz-sim-physics-system]
- Resolved library path: [/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]
Error while loading the library [/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]: dlopen(/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib, 0x0085): Library not loaded: '@rpath/libgz-sim7.7.dylib'
  Referenced from: '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim7-user-commands-system.7.4.0.dylib'
  Reason: tried: '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::UserCommands]
- Requested library name: [gz-sim-user-commands-system]
- Resolved library path: [/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]
Error while loading the library [/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]: dlopen(/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib, 0x0085): Library not loaded: '@rpath/libgz-sim7.7.dylib'
  Referenced from: '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim7-scene-broadcaster-system.7.4.0.dylib'
  Reason: tried: '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::SceneBroadcaster]
- Requested library name: [gz-sim-scene-broadcaster-system]
- Resolved library path: [/usr/local/Cellar/gz-sim7/7.4.0_2/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]
```
The same type of issue appears when I use in a second terminal the command `gz sim -g` to start the GUI. The GUI opens but it's empty without any plugin inserted.
How can I solve it?

NDevDrone, 2023-05-26:
Hi all, I would like to bump this issue with an additional data point. Using M1 running Ventura, same issue as above. I turned off MacOS SIP as well to try to resolve this, with no luck.

Building from binary seems to work fine with no errors thrown, but then I try to run the shapes.sdf file as normal and the following output shows when running the server command:
 
`Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <52FAE64F-D319-352D-9EF0-52CEA1134F9F> /opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.5.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::Physics]
- Requested library name: [gz-sim-physics-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <6FA08166-BFBF-3097-8D0C-D559966B91F9> /opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim7-user-commands-system.7.5.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::UserCommands]
- Requested library name: [gz-sim-user-commands-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <24D6853F-FA7A-3780-AD28-06CE0E360FE0> /opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim7-scene-broadcaster-system.7.5.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/local/lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::SceneBroadcaster]
- Requested library name: [gz-sim-scene-broadcaster-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.5.0/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]`

The GUI also throws a set of similar errors with the library path. Here is what the GUI looks like.

[image]

srmainwaring, 2023-06-02:
This is an issue with the `DYLD_LIBRARY_PATH` for the brew install location `/opt/homebrew/lib`, which is not being searched.

If you have disabled SIP using `csrutils disable` in Recovery Mode, then this will work:

Run the server:

```bash
DYLD_LIBRARY_PATH=/opt/homebrew/lib:$DYLD_LIBRARY_PATH gz sim -v4 -s -r shapes.sdf
```

Run the GUI

```bash
DYLD_LIBRARY_PATH=/opt/homebrew/lib:$DYLD_LIBRARY_PATH gz sim -v4 -g
```

Tested on macOS Ventura, Xcode and CommandLineTools 14.3.1. MacBookPro M1.

FedeCata, 2023-06-02:
Thank you for your reply! Following your instructions, the server works, while the GUI (that seems to open correctly initially) crashes, showing this messages on the terminal:

`ruby(985,0x700003e5c000) malloc: Heap corruption detected, free list is damaged at 0x600005b8f500
*** Incorrect guard value: 1065353216
ruby(985,0x700003e5c000) malloc: *** set a breakpoint in malloc_error_break to debug
Stack trace (most recent call last) in thread 123145367699456:
#31   Object "QtQuick", at 0x10455b056, in QSGBatchRenderer::Renderer::render() + 1846
#30   Object "QtQuick", at 0x104558d35, in QSGBatchRenderer::Renderer::renderBatches() + 1109
#29   Object "QtQuick", at 0x1045565eb, in QSGBatchRenderer::Renderer::prepareRenderMergedBatch(QSGBatchRenderer::Batch*, QSGBatchRenderer::Renderer::PreparedRenderBatch*) + 1131
#28   Object "QtQuick", at 0x1045554f9, in QSGBatchRenderer::Renderer::ensurePipelineState(QSGBatchRenderer::Element*, QSGBatchRenderer::ShaderManagerShader const*) + 569
#27   Object "QtGui", at 0x106720c8b, in QRhiGles2InitParams::newFallbackSurface(QSurfaceFormat const&) + 130587
#26   Object "QtGui", at 0x106720336, in QRhiGles2InitParams::newFallbackSurface(QSurfaceFormat const&) + 128198
#25   Object "Metal                               0x00007ff8258b1a77", at 0x7ff8258b1a77, in newLibraryWithSource:options:error:] + 177
#24   Object "Metal", at 0x7ff8258e375a, in MTLLibraryBuilder::newLibraryWithSource(id<MTLDevice>, NSString*, MTLCompileOptions*, bool, void (id<MTLLibrary>, NSError*) block_pointer) + 6106
#23   Object "Metal", at 0x7ff8258e1d32, in MTLLibraryBuilder::newLibraryWithRequestDataAndHash(MTLLibraryRequestData, MTLUINT256_t const&, void (id<MTLLibrary>, NSError*) block_pointer) + 126
#22   Object "Metal", at 0x7ff8258e0a83, in MTLLibraryBuilder::initLibraryContainerWithRequestData(MTLLibraryContainer*, MTLLibraryRequestData const&, void () block_pointer) + 1639
#21   Object "Metal", at 0x7ff82586cdac, in MTLCompilerConnectionManagerPrivate::buildRequest(unsigned int, MTLCompilerRequest*, bool, void (MTLCompilerError, NSObject<OS_dispatch_data>*, char const*) block_pointer) + 226
#20   Object "libdispatch.dylib", at 0x7ff81ccf27ee, in _dispatch_lane_barrier_sync_invoke_and_complete + 60
#19   Object "libdispatch.dylib", at 0x7ff81cce5317, in _dispatch_client_callout + 8
#18   Object "Metal", at 0x7ff8258bb247, in invocation function for block in MTLCompilerConnectionManagerPrivate::buildRequest(unsigned int, MTLCompilerRequest*, bool, void (MTLCompilerError, NSObject<OS_dispatch_data>*, char const*) block_pointer) + 243
#17   Object "Metal", at 0x7ff82586cedf, in XPCCompilerConnection::XPCCompilerConnection(int) + 75
#16   Object "Metal", at 0x7ff8258b9745, in XPCCompilerConnection::setupConnection() + 73
#15   Object "libdispatch.dylib", at 0x7ff81ccea20b, in _dispatch_lane_create_with_target + 415
#14   Object "libdispatch.dylib", at 0x7ff81cce4a31, in _os_object_alloc_realized + 25
#13   Object "libobjc.A.dylib", at 0x7ff81cd2f1dd, in class_createInstance + 64
#12   Object "libsystem_malloc.dylib", at 0x7ff81ccd3b75, in _malloc_zone_calloc + 60
#11   Object "libsystem_malloc.dylib", at 0x7ff81ccb968f, in nanov2_calloc + 126
#10   Object "libsystem_malloc.dylib", at 0x7ff81ccb7677, in nanov2_allocate + 130
#9    Object "libsystem_malloc.dylib", at 0x7ff81ccb80e8, in nanov2_allocate_from_block + 582
#8    Object "libsystem_malloc.dylib", at 0x7ff81ccd9308, in malloc_zone_error + 178
#7    Object "libsystem_malloc.dylib", at 0x7ff81ccc5357, in has_default_zone0 + 0
#6    Object "libsystem_c.dylib", at 0x7ff81cde7d24, in abort + 123
#5    Object "libsystem_pthread.dylib", at 0x7ff81ce9c1ff, in pthread_kill + 263
#4    Object "libsystem_kernel.dylib", at 0x7ff81ce65fff, in __pthread_kill + 11
#3    Object "libsystem_platform.dylib", at 0x7ff81ceb1dfc, in _sigtramp + 28
#2    Object "libgz-tools2-backward.dylib", at 0x1030b2b1b, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 13
#1    Object "libgz-tools2-backward.dylib", at 0x1030b2b89, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 79
#0    Object "libgz-tools2-backward.dylib", at 0x1030b2d17, in backward::StackTraceImpl<backward::system_tag::darwin_tag>::load_from(void*, unsigned long, void*, void*) + 33
zsh: abort      DYLD_LIBRARY_PATH=/opt/homebrew/lib:$DYLD_LIBRARY_PATH gz sim -v4 -g`

I think that it could be a problem with the heap memory but I have no idea to solve this issue. How can I solve it?

srmainwaring, 2023-06-02:
> while the GUI (that seems to open correctly initially) crashes, showing this messages on the terminal:

@FedeCata - the GUI issue is #1987 - the work-around fix is to comment out the component inspector plug-in in `~/.gz/sim/7/gui.config` and be aware that any GUI plugins that load a SpinBox will crash.

srmainwaring, 2023-06-02:
@scpeters a fix for this that doesn't require disabling SIP is to update the brew formula for the various `gz-*` libs:

```diff
- cmake_args << "-DCMAKE_INSTALL_RPATH=#{rpath}"
+ cmake_args << "-DCMAKE_INSTALL_RPATH=#{rpath};/opt/homebrew/lib"
```

I reinstalled `gz-garden` after this edit and everything is working correctly (bar Qt5 SpinBox conflicts). Can post a PR if you like?

FedeCata, 2023-06-02:
> @FedeCata - the GUI issue is #1987 - the work-around fix is to comment out the component inspector plug-in in `~/.gz/sim/7/gui.config` and be aware that any GUI plugins that load a SpinBox will crash.

I have just comment out the component inspector plug-in in `~/.gz/sim/7/gui.config` and the first time I opened the GUI it worked. Anyway, when I closed it and I tried to restart it together with the server, the same error messages listed above appeared

scpeters, 2023-06-03:
In my experience, the SIP issues could be worked around by using a non-system version of Ruby, as suggested in the following troubleshooting section:

* https://gazebosim.org/docs/garden/troubleshooting#unable-to-load-dylib-file

Though the actual path suggested is not generic enough; I've proposed a fix in https://github.com/gazebosim/docs/pull/361

There may still be an issue with the `RPATH` as @srmainwaring has suggested; I need to look into this further

srmainwaring, 2023-06-03:
In my testing I found using the brew version of ruby was not enough. The plugins failed to load on M1 machines because `/opt/homebrew/lib` is not on the system library search path, whereas on Intel machines `/usr/local/lib` is.

When using the brew version of ruby I still needed to prefix the `gz` commands with `DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/opt/homebrew/lib`, and this is only possible if SIP is disabled.

Adding `/opt/homebrew/lib` to the plugin RPATH when building resolved the issue and negated the need to disable SIP (and use the brew version of ruby).

FedeCata, 2023-06-03:
Disabling the SIP and prefixing the `gz` command with `DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/opt/homebrew/lib`, I have the same issues listed above (`malloc: Heap corruption detected`). Having an Intel machine on my MacBook, I also try to replace `/opt/homebrew/lib` with `/usr/local/lib` but nothing change and the issue persists.

srmainwaring, 2023-06-03:
> I have the same issues listed above (`malloc: Heap corruption detected`).

This is the Qt scene graph issue. You should not see this with a minimal `gui.config` that only includes a couple of the plugins, for example:

```xml
<?xml version="1.0"?>
<window>
  <width>1200</width>
  <height>800</height>
  <style
    material_theme="Light"
    material_primary="#9e9e9e"
    material_accent="#03a9f4"
    toolbar_color_light="#f3f3f3"
    toolbar_text_color_light="#111111"
    toolbar_color_dark="#414141"
    toolbar_text_color_dark="#f3f3f3"
    plugin_toolbar_color_light=""
    plugin_toolbar_text_color_light=""
    plugin_toolbar_color_dark=""
    plugin_toolbar_text_color_dark=""
  />
  <menus>
    <drawer default="false">
    </drawer>
  </menus>
</window>

<plugin filename="MinimalScene" name="3D View">
  <gz-gui>
    <title>3D View</title>
    <property type="bool" key="showTitleBar">false</property>
    <property type="string" key="state">docked</property>
  </gz-gui>
  <engine>ogre2</engine>
  <scene>scene</scene>
  <ambient_light>0.4 0.4 0.4</ambient_light>
  <background_color>0.8 0.8 0.8</background_color>
  <camera_pose>-6 0 6 0 0.5 0</camera_pose>
</plugin>
<plugin filename="GzSceneManager" name="Scene Manager">
  <gz-gui>
    <property key="resizable" type="bool">false</property>
    <property key="width" type="double">5</property>
    <property key="height" type="double">5</property>
    <property key="state" type="string">floating</property>
    <property key="showTitleBar" type="bool">false</property>
  </gz-gui>
</plugin>

<plugin filename="EntityTree" name="Entity tree">
  <gz-gui>
    <property type="bool" key="showTitleBar">true</property>
    <property type="string" key="state">docked</property>
  </gz-gui>
</plugin>
```

As you add back gui plugins you'll eventually get a conflict between QtQuick Controls 1.4 and 2.2. The SpinBox causes issues, there may be others.

> Having an Intel machine on my MacBook, I also try to replace /opt/homebrew/lib with /usr/local/lib but nothing change and the issue persists

This is unrelated to the Qt heap corruption issue. On an intel mac brew installs to `/usr/local/lib`. Including `/opt/homebrew/lib` is only relevant to macs with the M1 chips.

scpeters, 2023-09-23:
I made a bunch of RPATH fixes in homebrew formulae for harmonic, garden and fortress (though the gz-sim gui doesn't yet work on fortress on macOS) recently, and I think the Library errors should be fixed

closed 2024-01-19 by scpeters