gazebosim/gz-sim #1995: Third-party/external plugin repository

azeey, 2023-05-20:
## Desired behavior

It would be great if there was a centralized/recommended place for third party plugin contributions without being merged into upstream. This would speed up the rate of development of new plugins by the community since there will potentially be a less stringent code review process. 

To make this work well, we would provide binary distributions for all supported platforms, but we would also need to develop some infrastructure to make it easy to build that repository and use it with Gazebo seamlessly . This could include things like making sure plugins are installed in the right location or providing colcon hooks so paths are set up correctly.

If we develop the necessary infrastructure to make this work, it could also make plugins developed outside of this central repository (e.g. plugins with incompatible licenses) work well with Gazebo even if we wouldn't provide binary distributions for such repositories.

To speed up the PR review process, we would need help from the community, so perhaps we can have a two phase review process where the initial review would be done by members of the community and once that is complete, it'd be reviewed by the core simulation team.

## Alternatives considered
- Put third-party plugins in a staging area inside the gz-sim repo
- A wiki page with a list of third-party plugins

jrutgeer, 2023-05-26:
Obviously great, but I think you really should also consider the documentation aspect from the start.

It would be great if each plugin would have some standardized description, e.g.:
- General overview of the functionality it provides,
- The components it defines and/or uses,
- If it expects interaction with other plugins to create, initialize, update these components, 
- Which entitities it expects,
- Etc.

arjo129, 2023-05-31:
I'd also add we should provide templates that make it easy for end users to write tests to verify that their plugins behave as expected. These tests shouldn't be run with gazebo's core tests but whenever the developers push their plugins.

We could require that developers provide some form of manifest with there plugin describing what parameters exist and what types they should be.