gazebosim/gz-sim #1997: JointPositionController in ABS mode has first-order behavior rather than setting the position

jrutgeer, 2023-05-25:
Hi,

The JointPositionController can be used in [absolute mode](https://github.com/gazebosim/gz-sim/blob/3bde4b5318f9bed37e2fc8dbe5877bbdc9b73ae1/src/systems/joint_position_controller/JointPositionController.cc#L86-L93). Based on below description, I interprete this as "if enabled, a velocity will be applied to the joint so its position matches the position command":
https://github.com/gazebosim/gz-sim/blob/3bde4b5318f9bed37e2fc8dbe5877bbdc9b73ae1/src/systems/joint_position_controller/JointPositionController.hh#L86-L88

</br>

However, the applied velocity is equal to the position error:

https://github.com/gazebosim/gz-sim/blob/3bde4b5318f9bed37e2fc8dbe5877bbdc9b73ae1/src/systems/joint_position_controller/JointPositionController.cc#L440

So instead of instantly reaching the target position in each timestep, the behavior will rather be a first order motion, i.e. v = Kp * (p_des - p_meas) with Kp = 1. This implies that the position will always have some lag wrt the position command.

Is this intended behavior, or is this rather a bug?

</br>

To effectively apply an instant position (i.e. reach the target position in each timestep), I changed it to:

`targetVel = -error / dt;`

I.e. the velocity to move the joint in timestep `dt` to its desired position.

This yields following behavior:
- If the simulation is not started yet, and I send a command, then the joint instantly jumps to that position upon start of the simulation,
- However, if the simulation is started first, commands have no effect at all, without any error messages.

Any ideas on why this does not work?

</br>

An alternative could be to use the Pgain from the SDF:

`targetVel = -Pgain * error;`

so you can alter the gain. Would that be a relevant change?

azeey, 2023-05-30:
I think it's a bug. `targetVel = -error / dt;` makes more sense for the intended effect. @arjo129 can you verify?

> An alternative could be to use the Pgain from the SDF:
> targetVel = -Pgain * error;
> so you can alter the gain. Would that be a relevant change?

I like this approach. I think we might be able to fix the issue without breaking backward compatibility by setting `Pgain` to 1 by default, but still allow the user to set it to 1/dt.

> However, if the simulation is started first, commands have no effect at all, without any error messages.

Hmm, that's strange. Do you know what `targetVel` it's computing?

arjo129, 2023-05-31:
Yeah, this really should be `-error/dt;`. Seems like our tests are also wrong with regard to this, the intent is that in absolute mode we immediately jump to a position, you can cap the velocity via `CMD_MAX`.

> If the simulation is not started yet, and I send a command, then the joint instantly jumps to that position upon start of the simulation,
> However, if the simulation is started first, commands have no effect at all, without any error messages.

Will investigate whats going on.

arjo129, 2023-05-31:
I opened PR #2001 which addresses this. To the best of my knowledge there is no need to address backwards compatibility because even after changing `-error/dt;`, the system does not work unless a maximum command velocity is set.

azeey, 2023-08-09:
This has been fixed by #2001

PR #2001 (2023-08-03): Fix Joint Position Controller Behaviour Described in #1997
# 🦟 Bug fix

Fixes #1997

## Summary
This commit addresses #1997

There are several issues at play. First the target velocity calculation was wrong as described in #1997. Secondly, even if we corrected that there was still incorrect behavior. This is due to the fact that we use the PID's CmdMax to determine what the maximum velocity for the joint is. However, in the event a user does not set a `<cmd_max>` this defaults to zero and the joint does not move. Finally this PR updates the tests. Previously our tests were only testing the cas [... cut]
closed 2023-08-09 by azeey