gazebosim/gz-sim #1999: Update debian testing repo add bookworm

shaynababe, 2023-05-28:
http://packages.osrfoundation.org/gazebo/debian-testing/dists/
add bookworm
## Desired behavior

sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/debian-testing `lsb_release -cs` main" /etc/apt/sources.list.d/gazebo-stable.list'
wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update

## Alternatives considered

install sid , testing is now bookworm i do not have a method to install alt OS at the moment it seem add bookworm is not hard to do

## Implementation suggestion
 a general linux install would be good Dart did install and compiled protobuf compiled via brazen

## Additional context

shaynababe, 2023-05-28:
ase) z@spin7:~/tools/gymfc/gz-sim$ wget http://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update
--2023-05-28 11:17:18--  http://packages.osrfoundation.org/gazebo.key
Resolving packages.osrfoundation.org (packages.osrfoundation.org)... 52.52.171.73
Connecting to packages.osrfoundation.org (packages.osrfoundation.org)|52.52.171.73|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1755 (1.7K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                          100%[========================================================================================>]   1.71K  --.-KB/s    in 0s      

2023-05-28 11:17:18 (244 MB/s) - written to stdout [1755/1755]

Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
OK
Ign:1 http://packages.osrfoundation.org/gazebo/debian-stable bookworm InRelease
Err:2 http://packages.osrfoundation.org/gazebo/debian-stable bookworm Release                                                                                              
  404  Not Found [IP: 52.52.171.73 80]
Hit:3 http://packages.microsoft.com/repos/code stable InRelease                                                                                                            
Hit:4 http://deb.debian.org/debian bookworm InRelease                                                                                                             
Hit:5 http://dl.google.com/linux/webdesigner/deb stable InRelease                                                                           
Hit:6 https://dl.google.com/linux/chrome/deb stable InRelease                                                                               
Hit:7 https://deb.nodesource.com/node_18.x bookworm InRelease                                 
Hit:8 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:9 http://deb.debian.org/debian bookworm-updates InRelease
Hit:10 http://deb.debian.org/debian bookworm-backports InRelease
Hit:11 http://deb.debian.org/debian unstable InRelease
Reading package lists... Done
E: The repository 'http://packages.osrfoundation.org/gazebo/debian-stable bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Repository 'Debian bookworm' changed its 'non-free component' value from 'non-free' to 'non-free non-free-firmware'
N: More information about this can be found online in the Release notes at: https://www.debian.org/releases/bookworm/amd64/release-notes/ch-information.html#non-free-split
(base) z@spin7:~/tools/gymfc/gz-sim$ sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/debian-testing `lsb_release -cs` main" /etc/apt/sources.list.d/gazebo-stable.list'
deb http://packages.osrfoundation.org/gazebo/debian-testing bookworm main /etc/apt/sources.list.d/gazebo-stable.list
(base) z@spin7:~/tools/gymfc/gz-sim$

bmaxdk, 2023-08-12:
It seems compilation error related to `Google Protobuf`. You mentioned a compilation error due to the missing header `google/protobuf/stubs/casts.h`. It's likely that the code you're working on is using an older version of Protobuf, and you've encountered an API change. Since the newer version of Protobuf removed `google::protobuf::down_cast`, you have a couple of options to fix this. Either reinstall `protobuf` or try correct version of protobuf for your environemnt. Since I have similar issue, try this method. 

## Troubleshooting `protocol buffer`
Issue appeared when `protoc version` is not supported to current `turtlebot_gazebo` setup.First need to check **protocol buffer** version:

```bash
$ protoc --version
```

if `protoc` version is higher, try libprotoc 3.6.1:

Download `protobuf-cpp-3.6.1.tar.gz` in [Protocol Buffers v3.6.1](https://github.com/protocolbuffers/protobuf/releases/tag/v3.6.1)

```bash
$ tar -xzf protobuf-cpp-3.6.1.tar.gz
$ cd protobuf-3.6.1
# Install Dependencies for prtobuf
$ sudo apt-get install autoconf automake libtool curl make g++ unzip
$ ./autogen.sh

# Compile and install:
$ ./configure
$ make
$ sudo make install
# Update the Shared Libarary Cache:
$ sudo ldconfig

# Varify protoc version
$ protoc --version

```

### Gazebo Installation

**Check current Gazebo Version:**

```bash
$ gazebo --version
```
```text
Gazebo multi-robot simulator, version 11.13.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

**Install Gazebo:**

[Install Gazebo from source on Ubuntu](https://classic.gazebosim.org/tutorials?tut=install_from_source&cat=install)

```bash
$ sudo apt-get install gazebo11 libgazebo11-dev

# Locate the Gazebo Config Files.
# Try:
$ sudo find / -name gazebo-config.cmake
$ colcon build --cmake-args -Dgazebo_DIR=/usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake

# Source Gazebo Setup Script
$ sudo find / -name setup.sh | grep gazebo
$ source /path from above/setup.sh

# If you see multiple Gazebo Script such as:
#/usr/share/gazebo-11/setup.sh
#/usr/share/gazebo/setup.sh
# Try gazebo in /usr/share/

# Install gazebo_ros_pkgs for ROS2 Foxy
$ sudo apt-get install ros-foxy-gazebo-ros-pkgs
$ source /opt/ros/foxy/setup.bash
$ colcon build

```

shaynababe, 2023-09-24:
thank you it been some time 
the install is debian bookworm so protoc was not installed 
(base) z@spin7:~/tools/proto/protobuf-all-3.6.1$ protoc --version
libprotoc 3.6.1(base) z@spin7:~/tools/proto/protobuf-all-3.6.1$ protoc --version
libprotoc 3.6.1

stable is now bookworm

sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/debain-test `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-test.list' \
    && wget http://packages.osrfoundation.org/gazebo.key -O - | apt-key add - && apt-get update 

```code 
Hit:12 https://dl.google.com/linux/chrome/deb stable InRelease
Hit:13 http://deb.debian.org/debian unstable InRelease
Reading package lists... Done
E: The repository 'http://packages.osrfoundation.org/gazebo/debian-testing bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: The repository 'http://packages.osrfoundation.org/gazebo/debain-test bookworm Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
N: Repository 'Debian bookworm' changed its 'non-free component' value from 'non-free' to 'non-free non-free-firmware'
N: More information about this can be found online in the Release notes at: https://www.debian.org/releases/bookworm/amd64/release-notes/ch-information.html#non-free-split
dart
./install_dependencies.sh: line 43: cd: too many arguments
(base) z@spin7:~/tools/gymfc$ ^C
(base) z@spin7:~/tools/gymfc$ 
```

booo, 2024-01-08:
Is it possible to install a version of gazebo on debian bookworm? Can someone point me in the right direction?

azeey, 2024-01-08:
We currently do not provide debs for bookworm. You'd have to install from source.