gazebosim/gz-sim #2002: wideanglecamera crashes gazebo

Mathosmls, 2023-05-31:
## Environment
* Ubuntu 22.04 
*  Gazebo Garden, binary

    - Rendering plugin: ogre
      - [ ] Sensor rendering error. No errors of this type
      - [ ] GUI rendering error. A lot of warnings but no errors

    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU) (not sure about this one)
      - [ ] running on real hardware
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          0000:00:02.0 VGA compatible controller [0300]: Intel Corporation TigerLake-LP GT2 [Iris Xe Graphics] [8086:9a49] (rev 01)

          echo "$DISPLAY"
          :0

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          direct rendering: Yes
          Preferred profile: core (0x1)
          Max core profile version: 4.6
          Max compat profile version: 4.6
          Max GLES1 profile version: 1.1
          Max GLES[23] profile version: 3.2
          OpenGL vendor string: Intel
          OpenGL renderer string: Mesa Intel(R) Xe Graphics (TGL GT2)
          OpenGL core profile version string: 4.6 (Core Profile) Mesa 22.2.5
          OpenGL core profile shading language version string: 4.60
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.6 (Compatibility Profile) Mesa 22.2.5
          OpenGL shading language version string: 4.60
          OpenGL context flags: (none)
          OpenGL profile mask: compatibility profile
          OpenGL ES profile version string: OpenGL ES 3.2 Mesa 22.2.5
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

          ps aux | grep Xorg
          mathis    343302  0.0  0.0  18052  2412 pts/0    S+   11:25   0:00 grep --color=auto Xorg

          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```X.Org X Server 1.21.1.4
          X Protocol Version 11, Revision 0
          Current Operating System: Linux mathis-Inspiron-5402 5.19.0-42-generic #43~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Apr 21 16:51:08 UTC 2 x86_64
          Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.19.0-42-generic root=UUID=62a3d99b-5393-4e07-ada3-7ec054600be2 ro quiet splash vt.handoff=7
          xorg-server 2:21.1.4-2ubuntu1.7~22.04.1 (For technical support please see http://www.ubuntu.com/support) 
          Current version of pixman: 0.40.0
	          Before reporting problems, check http://wiki.x.org
	          to make sure that you have the latest version.

<details>

my code : 
<?xml version="1.0"?>
<sdf version="1.9">

    <world name="buddy_world">

        <physics
            name="1ms" type="ignored">
            <max_step_size>0.001</max_step_size>
            <real_time_factor>1</real_time_factor>
        </physics>
        <plugin
            filename="gz-sim-physics-system"
            name="gz::sim::systems::Physics">
        </plugin>
[... 118 lines cut ...]
#19   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764fbf30c, in gz::rendering::v7::OgreScene::PreRender()
#18   Object "/lib/x86_64-linux-gnu/libgz-rendering7.so.7", at 0x7fb766ae094d, in gz::rendering::v7::BaseScene::PreRender()
#17   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f248ba, in 
#16   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f2d6f1, in 
#15   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f248ba, in 
#14   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f2d6f1, in 
#13   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f248ba, in 
#12   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f2d6f1, in 
#11   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f2a92e, in 
#10   Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f84bbe, in 
#9    Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f8980a, in 
#8    Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f7d7db, in gz::rendering::v7::OgreMaterial::UpdateShaderParams()
#7    Object "/usr/lib/x86_64-linux-gnu/gz-rendering-7/engine-plugins/libgz-rendering-ogre.so", at 0x7fb764f7d4d4, in gz::rendering::v7::OgreMaterial::UpdateShaderParams(std::shared_ptr<gz::rendering::v7::ShaderParams const>, Ogre::SharedPtr<Ogre::GpuProgramParameters>)
#6    Object "/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0", at 0x7fb7649c0a7a, in Ogre::GpuProgramParameters::_writeRawConstant(unsigned long, int)
#5    Object "/lib/x86_64-linux-gnu/libOgreMain.so.1.9.0", at 0x7fb7649c0a55, in Ogre::GpuProgramParameters::_writeRawConstants(unsigned long, int const*, unsigned long)
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb796639e95, in __assert_fail
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb79662871a, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb7966287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb796642475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7fb796696a7c, in pthread_kill
Aborted (Signal sent by tkill() 344092 1000)
Warning [Utils.cc:130] [/sdf/plugin[@name="ignition::gazebo::systems::ShaderParam"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].
Warning [Utils.cc:130] [/sdf/plugin[@name="ignition::gazebo::systems::ShaderParam"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].

```

iche033, 2023-05-31:
> the sim crashes if I include both models into the sim, but works fine if I only include one of them.

are you able to run the wide angle camera by itself?

I can reproduce the issue. I think the problem is that the waves model uses shaders that only work in ogre2 and the wide angle camera is currently only available in ogre1. For me I can run wide angle camera in ogre1 fine but not with models that the shaders (e.g. waves, deformable sphere)

Mathosmls, 2023-05-31:
Yes I was able to run only the wide angle camera but I was also able to run only the waves with ogre1. It's when I put them together that the sim crashes. 
And I don't think that my robot model or my map have shaders, they are just collada files directly exported from blender.
Also I added in the details of my issue that my sim crashes when I use a normal camera sensor with ogre1, does ogre1 not support it ?

Mathosmls, 2023-06-01:
When I run `gz sim -v 4 camera_sensor.sdf --render-engine ogre`, gazebo opens but my laptop freeze and after few seconds gazebo crashes. The last outputs in my shell  : 
```
Stack trace (most recent call last) in thread 3345288:
#30   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#29   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f80333269ff, in 
#28   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8033294b42, in 
#27   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802caccca0, in 
#26   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802cacbaf1, in QThread::exec()
#25   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802ccb875a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#24   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802cd130b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#23   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f802d3193e2, in g_main_context_iteration
#22   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f802d3706c7, in 
#21   Object "/lib/x86_64-linux-gnu/libglib-2.0.so.0", at 0x7f802d31bd3a, in g_main_context_dispatch
#20   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802cd13a66, in 
#19   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802ccbcf26, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#18   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802ccb9e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#17   Object "/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f802c36c712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#16   Object "/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f802cce741d, in QObject::event(QEvent*)
#15   Object "/usr/lib/x86_64-linux-gnu/gz-gui-7/plugins/libMinimalScene.so", at 0x7f7ff801fef8, in gz::gui::plugins::RenderThread::RenderNext(gz::gui::plugins::RenderSync*)
#14   Object "/lib/x86_64-linux-gnu/libQt5Gui.so.5", at 0x7f802af8495e, in QOpenGLContext::doneCurrent()
#13   Object "/usr/lib/x86_64-linux-gnu/qt5/plugins/xcbglintegrations/libqxcb-glx-integration.so", at 0x7f7ffa46bdfc, in 
#12   Object "/lib/x86_64-linux-gnu/libGLX.so.0", at 0x7f802a0bc125, in 
#11   Object "/lib/x86_64-linux-gnu/libGLX.so.0", at 0x7f802a0b7640, in 
#10   Object "/lib/x86_64-linux-gnu/libGLX_mesa.so.0", at 0x7f7ffa3dc2c5, in 
#9    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7f7ff88b6168, in 
#8    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7f7ff88b24dd, in 
#7    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7f7ff8988aea, in 
#6    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7f7ff8a167ff, in 
#5    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7f7ff89727a6, in 
#4    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7f7ff955de5e, in 
#3    Object "/usr/lib/x86_64-linux-gnu/dri/iris_dri.so", at 0x7f7ff889c8b6, in 
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f80332287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8033242475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f8033296a7c, in pthread_kill
Aborted (Signal sent by tkill() 3345216 1000)
[Dbg] [SignalHandler.cc:141] Received signal[2].
[Dbg] [ServerPrivate.cc:115] Server received signal[2]
[Dbg] [Sensors.cc:428] SensorsPrivate::Stop
[Dbg] [Sensors.cc:414] SensorsPrivate::RenderThread stopped
```
It does more or less the same thing when I try to run the example world wideanglecamera :/

> 

>  Want to double check on this. The gui could still be using ogre2 (and only the camera sensor on the server backend is using ogre1). Can you try running with --render-engine ogre arg? This ensures that both the gui and server are using ogre1.

But you were right for that, thanks ! I thought that `<render_engine>ogre</render_engine>` was enough, sorry.

iche033, 2023-06-05:
The ogre log does not reveal any errors. I thought it would then be Qt issue but it seems like Qt + ogre2 works fine for you. It's not ideal but you may have to go with ogre2 and use a normal camera + large horizontal FOV.

Mathosmls, 2023-06-06:
Yeah that's what I did, at least users understand that is supposed to be a wide angle camera. Anyway, Thank you very much for your help and your time ! Do I close this issue as not planned or do I let it open ?

iche033, 2023-06-06:
ok I'll close this issue. By the way, wide angle camera support for ogre2 will be available in the next version of Gazebo (Harmonic) which will be released in Sept. Related PR: https://github.com/gazebosim/gz-rendering/pull/733

closed 2023-06-06 by iche033

Dat-Bois, 2023-09-13:
> I can reproduce the issue. I think the problem is that the waves model uses shaders that only work in ogre2 and the wide angle camera is currently only available in ogre1. For me I can run wide angle camera in ogre1 fine but not together with models that use shaders (e.g. waves, deformable sphere).
> 
> In fact, this example world crashes with ogre
> 
> ```
> gz sim -v 4 shader_param.sdf --render-engine ogre
> ```
> 
> The underlying problem here is that shaders are not working in ogre1. Those models use glsl version 330. Maybe we just need to convert them to versions compatible with ogre1, e.g. 150

I am running into this exact issue. It is a shaders issue and running the waves plugin or the command you suggested give the same ```void Ogre::GpuProgramParameters::_writeRawConstants(size_t, const int*, size_t): Assertion `physicalIndex + count <= mIntConstants.size()' failed``` error but anything without those shaders works (ie wide angle camera). I can't switch to Ogre2 since that introduces a lot of other issues (I am running in a VM with 3d acceleration on ARM64 and with Ogre2 its just a blank or flickering screen)

Looking at the syntax changes between 330 and 150 I can't find incompatibility in the waves shader files. The ones I am using are here: https://github.com/srmainwaring/wave_sim_vrx

In another forum someone wrote that the error message means that "It says that Ogre is trying to write more constants that the shader has" https://forums.ogre3d.org/viewtopic.php?t=24961&start=425

Any ideas?

iche033, 2023-09-13:
The [waves](https://app.gazebosim.org/OpenRobotics/fuel/models/Waves) model uses the shader param system. In the model.sdf file, you'll see blocks like 

```xml
  <param>                                                                   
       <shader>vertex</shader>                                              
       <name>Nwaves</name>                                                  
       <type>int</type>                                                     
       <value>3</value>                                                     
   </param>                                                                 
   <param>                                                                  
       <shader>vertex</shader>                                              
       <name>rescale</name>                                                 
       <type>float</type>                                                   
       <value>0.5</value>                                                   
   </param>                                                                 
   <param>                                                                  
       <shader>vertex</shader>                                              
       <name>bumpScale</name>                                               
       <type>float_array</type>                                             
       <value>25 25</value>                                                 
   </param>                                                                 

```

These are for setting variables defined in the shader script. So in the above example, we expect variables like `Nwaves`, `rescale`, `bumpScale`, etc to exist in the vertex shader. Same goes for the fragrment shader.

I'm assuming you took the shaders from [here](https://github.com/srmainwaring/wave_sim_vrx/tree/master/wave_gazebo/world_models/ocean_waves/materials/programs). Looking at the shaders there, you can probably just remove `<param>` blocks for variables that are not used in the glsl 150 version e.g. `Nwaves`, `worldviewproj_matrix`. You can find a list of variables needed by the 150 version in the [material script file](https://github.com/srmainwaring/wave_sim_vrx/blob/4b115672435ae7b317abeb5fa8c30a89cb39c15d/wave_gazebo/world_models/ocean_waves/materials/scripts/waves.material#L33-L54)

One special case is `time`. We'll need to keep this `<param>` block:

```
  <param>                                    
      <shader>vertex</shader>                                               
      <name>t</name>                                                        
      <value>TIME</value>                                                   
  </param> 
```

and change the variable in vertex shader from `time` to `t`.

Dat-Bois, 2023-09-14:
Hi @iche033, thanks for the quick response.

I am fairly new to all this so thanks for the detailed explanation. The project I am trying to run is the one found here:
https://github.com/osrf/vrx

This is the model.sdf file for the waves in the project (can be found under vrx_gz/models/coast_waves/model.sdf):
```
<?xml version="1.0"?>
<sdf version="1.6">
  <model name="waves">
    <static>true</static>
    <link name="link">
      <visual name= "wave_visual">
        <geometry>
          <mesh>
            <uri>meshes/waterlow.dae</uri>
          </mesh>
        </geometry>
        <plugin filename="libWaveVisual.so" name="vrx::WaveVisual">
          <shader>
            <vertex>materials/programs/GerstnerWaves_vs_330.glsl</vertex>
            <fragment>materials/programs/GerstnerWaves_fs_330.glsl</fragment>
            <parameters>
              <rescale>0.125</rescale>
              <bumpScale>75 75</bumpScale>
              <bumpSpeed>0.01 0.0</bumpSpeed>
              <hdrMultiplier>0.4</hdrMultiplier>
              <fresnelPower>5.0</fresnelPower>
              <shallowColor>0 0.1 0.2 1.0</shallowColor>
              <deepColor>0 0.05 0.2 1.0</deepColor>
            </parameters>
          </shader>
          <wavefield>
            <topic>/vrx/wavefield/parameters</topic>
          </wavefield>
        </plugin>
        <visibility_flags>8</visibility_flags>
      </visual>
    </link>
  </model>
</sdf>

```

> you can probably just remove <param> blocks for variables that are not used in the glsl 150 version e.g. Nwaves, worldviewproj_matrix.

How do I know which ones are not used in the glsl 150 version? I tried commenting them all out but the error still occurred.

> You can find a list of variables needed by the 150 version in the material script file

In the same coast_waves directory of the project I cannot find a script file for the waves, am I looking in the wrong place?

iche033, 2023-09-15:
ok I see. VRX is using a different system to render the waves so my previous instructions are less relevant.

I did some tests. To have waves working in ogre 1.x, you'll need these changes from https://github.com/gazebosim/gz-rendering/pull/908. In that pull request, I ported the vertex and fragment shaders to glsl version 150 so I think you should be able to use those and replace the ones in VRX.

This will only get the the waves working with VRX though. There could be other things in VRX that don't work properly in ogre 1.x.

Dat-Bois, 2023-09-16:
Unfortunately, replacing the respective fragment and vertex files from the ones in the pull request did not seem to fix the issue. It still has the exact same error. 

I can verify it is only the waves causing the issues since removing the shaders lines from the coast waves sdf file allows the rest of the environment to load in without issues (just of course there are no waves anymore).