gazebosim/gz-sim #2008: Tracked vehicles go backwards.

rodericktaylor, 2023-06-08:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
  Source 3bde4b5318f9bed37e2fc8dbe5877bbdc9b73ae1 (gz-sim7 branch on garden)
## Description
* Expected: tracked_vehicle_simple.sdf, pressing W should make the robot go forward, pressing D should make the robot turn clockwise.
* Actual: pressing W makes the robot go backwards and pressing D makes the robot go anti-clockwise.

The TrackedVehicle plugin does not take into account that TrackController's moving in the positive direction make the part of the track that touches ground go the opposite direction.

## Steps to reproduce

1. `gz sim -r tracked_vehicle_simple.sdf`
2. Press W, then press S to stop. The robot moves backwards.
3. Press D, then press S to stop. The robot turns anti-clockwise.

Bi0T1N, 2023-06-10:
I've an environment setup as written in #1662 for Focal (@jrutgeer for Garden) and there it works as expected. `W` moves the robot forward and `D` moves it clockwise. Thus this must be a recent regression in gz-sim or DART if it is reproducible by others.

peci1, 2023-06-12:
I have no idea why this is happening.

Could you compare the contents of https://github.com/gazebosim/gz-sim/blob/a65c4c91aff3fddb5e7241b777af9894cdfce28b/include/gz/sim/physics/Events.hh#L47 in Fortress and Garden? I.e. to figure out whether the problem lies on Gazebo side or DART side?

azeey, 2023-06-13:
I'm using just Garden but with DART 6.10 (fork) vs DART 6.13.0 from upstream.

Here's the print out from setting `<debug>1</debug>` on one of the `TrackController` plugins and pressing `W`:
<details><summary>With DART 6.13</summary>
<p>

```
[Dbg] [TrackController.cc:512] Link: left_track
[Dbg] [TrackController.cc:513] - is collision 1 track 1
[Dbg] [TrackController.cc:515] - velocity cmd         1
[Dbg] [TrackController.cc:516] - limited velocity cmd 1
[Dbg] [TrackController.cc:517] - friction direction   -1 1.5088e-11 0
[Dbg] [TrackController.cc:518] - surface motion       -1
[Dbg] [TrackController.cc:519] - contact point             2.58091
    0.247139
-0.000164322
[Dbg] [TrackController.cc:520] - contact normal       0 0 1
[Dbg] [TrackController.cc:521] - track rot            -1.66882e-08 -9.13289e-07 -1.50728e-11
[Dbg] [TrackController.cc:522] - track Y              1.5088e-11 1 -1.66882e-08
[Dbg] [TrackController.cc:523] - belt direction       -1 1.5088e-11 0

```
</p>
</details>

<details><summary>With DART 6.10 from our fork</summary>
<p>

```
[Dbg] [TrackController.cc:512] Link: left_track
[Dbg] [TrackController.cc:513] - is collision 1 track 1
[Dbg] [TrackController.cc:515] - velocity cmd         1
[Dbg] [TrackController.cc:516] - limited velocity cmd 1
[Dbg] [TrackController.cc:517] - friction direction   -1 -1.50255e-11 0
[Dbg] [TrackController.cc:518] - surface motion       -1
[Dbg] [TrackController.cc:519] - contact point             3.25509
    0.247139
-3.71957e-05
[Dbg] [TrackController.cc:520] - contact normal       0 0 1
[Dbg] [TrackController.cc:521] - track rot            -3.08715e-08 -1.52132e-06 1.50724e-11
[Dbg] [TrackController.cc:522] - track Y              -1.50255e-11 1 -3.08715e-08
[Dbg] [TrackController.cc:523] - belt direction       -1 -1.50255e-11 0
```
</p>
</details> 

They look very similar to me, so I'm inclined to think it's a difference in implementation between our fork of DART and upstream.

azeey, 2023-06-13:
I've tracked this down to the way  the `TangentBasisMatrix` is calculated in DART. When we upstreamed our changes in https://github.com/dartsim/dart/pull/1427, there was a [review comment](
https://github.com/dartsim/dart/pull/1427#discussion_r352811075) that caused us to flip the direction of the tangent. At the time, the change didn't seem to break any behavior, so we didn't think to sync our fork with the upstream version. 

@peci1 Do you think we can simply negate the sign of the `contactSurfaceMotionVelocity` vector in gz-physics for DART versions 6.13.0 and later?

peci1, 2023-06-13:
Good investigation. It seems negating the velocity vector in gz-physics would be the cleanest way.

The only thing I fear is whether it will be sufficient to distinguish the fork and upstream based on the version number. I remember times where there were the same versions upstream and in the fork with different behaviors.

Bi0T1N, 2023-06-13:
I've updated my Fortress version to the new release from 2023-05-31 and the buttons are inverted now. I'm pretty sure this wasn't the case before (working on different robots, not the ones from the Gazebo example) but can't easily verify it...

rodericktaylor, 2023-06-20:
I'm building from source. Is the fork available for me to include in my build?

jrutgeer, 2023-06-20:
@rodericktaylor 

> I'm building from source. Is the fork available for me to include in my build?

Refer to https://github.com/gazebosim/gz-sim/issues/1662#issuecomment-1227400260

azeey, 2023-06-20:
> The only thing I fear is whether it will be sufficient to distinguish the fork and upstream based on the version number. I remember times where there were the same versions upstream and in the fork with different behaviors.

Our fork is currently at 6.10, so we can easily distinguish between the two. If we ever update our fork to be >= 6.13, we'll need to change the `TangentBasisMatrix` to match what's in upstream. In addition to opening an issue on our fork, we should add a test in gz-physics to make sure we don't forget.

> In the interest of accelerating a solution, may I suggest you define something like. `#define DART_HAS_POSITIVE_SURFACE_MOTION_VELOCITY 1.` and negate based on `#ifndef` instead of checking version numbers. When or if your fork is not used, the code will still work... forever.

Where are you suggesting we put this `#define`? If we put it in our fork, and update gz-physics to check for it, I'm concerned that users that aren't following the developments in our fork will have a silent change of behavior.  We already make assumptions that >=6.13 is upstream, so I think fixing it that way is our best option. /cc @scpeters

rodericktaylor, 2024-01-15:
This looks like it is fixed in the harmonic release when compiled from source with the DART 6.13.

azeey, 2024-01-16:
Yes, I closed https://github.com/gazebosim/gz-sim/issues/1662, but forgot to close this.

closed 2024-01-16 by azeey