gazebosim/gz-sim #2016: Track controller with position output

jrutgeer, 2023-06-15:
I am looking into simulation a conveyor that has an encoder output (i.e. publishes the conveyor belt position).

I assume the [track controller system](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/track_controller/TrackController.cc) would be the right place to add that functionality.

Is there any interest that this would be added (I can try and submit a PR)?

I am not sure how to implement this though. AFAIK the track controller does not have the notion of position, it just changes friction parameters to simulate motion?

I had a look at the [tracked vehicle system](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/tracked_vehicle/TrackedVehicle.cc), as it outputs odometry data. But it seems this just [integrates the velocity commands](https://github.com/gazebosim/gz-sim/blob/064f69ed7c441b90abbf0f84eeb0bfaeec628da9/src/systems/tracked_vehicle/TrackedVehicle.cc#L731-L734)? This seems strange, since there might as well be slip between the track and the contacting surface. So if you just interpolate the desired velocity, this will eventually lead to incorrect odometry values?

An alternative could be to implement a new system, similar to the [Ariac conveyor belt plugin](https://github.com/usnistgov/ARIAC/blob/ariac2023/ariac_plugins/src/conveyor_belt_plugin.cpp). This does not change the friction, but if I recall correctly, it models a linear joint between the conveyor and its collision entity. The joint is then moved at the set velocity (so the objects on top of it move with it), and every once in a while the joint position is reset to 0 (in one timestep). But this is in Gazebo classic, not sure if this could be implemented in the same way in Gazebo sim?

jrutgeer, 2023-06-15:
Further thinking of it: in the case of a conveyor, integration of the velocity commands is actually correct, as an ideal conveyor does realize the commanded motion. Obviously objects on the conveyor can slip.

For the tracked vehicle however, I think it would be better to calculate odometry based on the actual pose wrt world.

azeey, 2023-06-15:
> For the tracked vehicle however, I think it would be better to calculate odometry based on the actual pose wrt world.

If the tracks slip, wouldn't the position of the track still change without the corresponding motion of the vehicle? i.e, slipping would introduce an error even if you use the joint position. Also, odometry is not meant to give you the ground truth. Errors due to slip should be included in the odometry output to simulate real odometry measurements. If you want ground truth, you can just get the pose of the vehicle without using odometry.

jrutgeer, 2023-06-15:
> Errors due to slip should be included in the odometry output to simulate real odometry measurements.

Indeed. My line of thought was similar to e.g. a simulated camera, where a 'perfect' image is rendered and noise is added afterwards. So I similarly assumed 'perfect odometry' onto which an error would be applied afterwards. But in this case calculating the error would come down to estimating slip, which inherently is in behavior of the plugin itself. So it makes sense.

Anyway, the conclusion is that track position can be calculated by integrating the velocity commands.

Remain following questions:
1. Is it relevant to be added (i.e. do I submit a PR) or do I just implement this locally?
2. I think ideally the plugin should report track position, velocity, acceleration and jerk. I checked [gz-msgs](https://github.com/gazebosim/gz-msgs/tree/gz-msgs9/proto/gz/msgs) but there doesn't seem to be an existing message proto for this. Is it best to define a new one, or is there some other (combination of?) message(s) to cover this?

PR #2021 (2023-08-30): Odometry topic for the track controller system
# 🎉 New feature

Closes #2016
~~Requires https://github.com/gazebosim/gz-msgs/pull/352~~

## Summary
This PR adds odometry output to the [track controller system](https://github.com/gazebosim/gz-sim/blob/gz-sim7/src/systems/track_controller/TrackController.cc), publishing position and instantaneous velocity. That way a conveyor with an encoder and/or resolver for respectively position and velocity feedback can be simulated.

The default topic is `/model/<model name>/link/<link name>/odometry`.
An alternative can be chosen by specifying the `odomet [... cut]
closed 2023-08-30 by azeey