gazebosim/gz-sim #2022: Rendering failure of cameras on 22.04 with libgazebo6

PeterQFR, 2023-06-20:
## Environment
* OS Version:  Ubuntu 22.04
* Source or binary build?
 libignition-gazebo6-dev
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
I am running the following command:
ign gazebo -r -s -v 4 --headless-rendering world.sdf
[Msg] Ignition Gazebo Server v6.14.0

   - Rendering plugin: ogre2

    - EGL headless mode:
 I am running headless on a ubuntu desktop 22.04 install via ssh. I do get the following warnings. 
 libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen

    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ x] running on a dual GPU machine (integrated GPU + discrete GPU)

      - [ x] running on real hardware

      - [ x] running remotely (e.g. via SSH)

      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:

          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP102 [GeForce GTX 1080 Ti] [10de:1b06] (rev a1)
          echo "$DISPLAY"
Nil $Display

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
Error unable to open display

          ps aux | grep Xorg
ichor-s+    2778  0.0  0.0 25357380 59832 tty2   Sl+  20:48   0:01 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1001/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
ichor-s+    5866  0.0  0.0  18188  2440 pts/0    S+   21:31   0:00 grep --color=auto Xorg

          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
X.Org X Server 1.21.1.4
X Protocol Version 11, Revision 0
Current Operating System: Linux 5.19.0-45-generic #46~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Jun 7 15:06:04 UTC 20 x86_64
xorg-server 2:21.1.4-2ubuntu1.7~22.04.1 (For technical support please see http://www.ubuntu.com/support) 
Current version of pixman: 0.40.0

      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
21:21:29: Creating resource group General
21:21:29: Creating resource group Internal
21:21:29: Creating resource group Autodetect
21:21:29: SceneManagerFactory for type 'DefaultSceneManager' registered.
21:21:29: Registering ResourceManager for type Material
21:21:29: Registering ResourceManager for type Mesh
21:21:29: Registering ResourceManager for type Mesh2
21:21:29: Registering ResourceManager for type OldSkeleton
21:21:29: MovableObjectFactory for type 'ParticleSystem' registered.
21:21:29: ArchiveFactory for archive type FileSystem registered.
21:21:29: ArchiveFactory for archive type Zip registered.
21:21:29: ArchiveFactory for archive type EmbeddedZip registered.
21:21:29: DDS codec registering
[... 613 lines cut ...]
21:21:29: Added resource location '/home/ichor-server/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/base station/8/meshes' of type 'FileSystem' to resource group 'General' with recursive option
21:21:29: Initialising resource group General
21:21:29: Added resource location '/home/ichor-server/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/base station/8/meshes/../materials/textures/' of type 'FileSystem' to resource group 'General'
21:21:29: Can't assign material scene::Material(64951) because this Material does not exist. Have you forgotten to define it in a .material script?
21:21:29: Texture memory budget exceeded. Stalling GPU.
21:21:29: Stalling was not enough. Freeing memory.
21:21:29: Can't assign material scene::Material(64946) because this Material does not exist. Have you forgotten to define it in a .material script?
21:21:29: Added resource location '/home/ichor-server/simulation/models/terrain_street_rh_turn/meshes' of type 'FileSystem' to resource group 'General' with recursive option
21:21:29: Initialising resource group General
21:21:29: Added resource location '/home/ichor-server/simulation/models/terrain_street_rh_turn/meshes/' of type 'FileSystem' to resource group 'General'
21:21:29: OGRE EXCEPTION(2:InvalidParametersException): Unable to load image: Image format is unknown. Unable to identify codec. Check it or specify format explicitly. in Image2::load at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/OgreMain/src/OgreImage2.cpp (line 718)
21:21:29: OGRE EXCEPTION(2:InvalidParametersException): Unable to load image: Image format is unknown. Unable to identify codec. Check it or specify format explicitly. in Image2::load at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/OgreMain/src/OgreImage2.cpp (line 718)
21:21:29: Texture memory budget exceeded. Stalling GPU.
21:21:29: Stalling was not enough. Freeing memory.

```

</details>

## Description
Expected the gazebo server to run headless. It does work with the same model file on 20.04 and gazebo server v6.10.0

This same error occurs even when I am running in the desktop environment of the 22.04 computer. 

The system does work when I comment out only my camera sensors from the model. Other sensors such as imu, altimeter etc do not seem to cause the fault.

PeterQFR, 2023-06-20:
Adding a camera such as:
```
<sensor name="front_camera" type="rgbd_camera">
                <pose>0.09753187 -0.00160207 -0.03329037 1.57079632 0 0.0</pose>
                <always_on>1</always_on>
                <update_rate>20</update_rate>
                <camera name="front_camera">
                    <horizontal_fov>1.5768</horizontal_fov>
                    <lens>
                        <intrinsics>
                            <fx>141.14</fx>
                            <fy>141.14</fy>
                            <cx>142.5</cx>
                            <cy>80.5</cy>
                            <s>0</s>
                        </intrinsics>
                    </lens>
                    <distortion>
                        <k1>0.0</k1>
                        <k2>0.0</k2>
                        <k3>0.0</k3>
                        <p1>0.0</p1>
                        <p2>0.0</p2>
                        <center>0.5 0.5</center>
                    </distortion>
                    <image>
                        <width>284</width>
                        <height>160</height>
                        <format>R8G8B8</format>
                    </image>
                    <clip>
                        <near>0.01</near>
                        <far>100</far>
                    </clip>
                    <depth_camera>
                      <clip>
                        <near>0.1</near>
                        <far>10</far>
                      </clip>
                    </depth_camera>
                    <noise>
                        <type>gaussian</type>
                        <mean>0</mean>
                        <stddev>0.007</stddev>
                    </noise>
                </camera>
            </sensor>

```
will cause the fault. Commenting it out will avoid the fault. This is also for rgbd cameras

PeterQFR, 2023-06-20:
Running the sensors_demo.sdf works.

iche033, 2023-06-20:
looking at the ogre log and the backtrace, there may be a texture in your world that's causing the crash. 

```
21:21:29: OGRE EXCEPTION(2:InvalidParametersException): Unable to load image: Image format is unknown. Unable to identify codec. Check it or specify format explicitly. in Image2::load at /build/ogre-next-UFfg83/ogre-next-2.2.5+dfsg3/OgreMain/src/OgreImage2.cpp (line 718)
```

The texture is probably used by a mesh in a model so try commenting out models in the world and see if you can isolate the crash.

PeterQFR, 2023-06-20:
Okay I checked I found a single mesh that is causing the crash but the textures are correctly png files. In the ogre log its also complaining about being out of memory, and being unable to identify the correct codec.  How do I give it more memory, and manually specify the codec?

PeterQFR, 2023-06-20:
Here is the the model that is causes grief.
[terrain_street_intersection.zip](https://github.com/gazebosim/gz-sim/files/11809260/terrain_street_intersection.zip)