gazebosim/gz-sim #2029: 3D screen flickering and Ogre2 crashes

Qqww4599, 2023-07-11:
## Environment
* OS Version: Ubuntu 22.04(WSL) on Windows 10
* binary build (Garden). Install from https://gazebosim.org/docs/garden/install_ubuntu

    - Rendering plugin: ogre and ogre2.
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [x] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:

          ftc@DESKTOP-ARO2VPH:~$ LANG=C lspci -nn | grep VGA
          ftc@DESKTOP-ARO2VPH:~$ echo "$DISPLAY"
          :0
          ftc@DESKTOP-ARO2VPH:~$ LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'
          direct rendering: Yes
          Preferred profile: core (0x1)
          Max core profile version: 4.1
          Max compat profile version: 4.1
          Max GLES1 profile version: 1.1
          Max GLES[23] profile version: 3.0
          OpenGL vendor string: Microsoft Corporation
          OpenGL renderer string: D3D12 (Intel(R) UHD Graphics 770)
          OpenGL core profile version string: 4.1 (Core Profile) Mesa 22.2.5-0ubuntu0.1~22.04.3
          OpenGL core profile shading language version string: 4.10
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.1 (Compatibility Profile) Mesa 22.2.5-0ubuntu0.1~22.04.3
          OpenGL shading language version string: 4.10
          OpenGL context flags: (none)
          OpenGL profile mask: compatibility profile
          OpenGL ES profile version string: OpenGL ES 3.0 Mesa 22.2.5-0ubuntu0.1~22.04.3
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
          ftc@DESKTOP-ARO2VPH:~$ ps aux | grep Xorg
          ftc        19581  0.0  0.0   4024  1924 pts/0    S+   13:28   0:00 grep Xorg
          
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

```
ogre1
# paste log here
13:14:27: Creating resource group General
13:14:27: Creating resource group Internal
13:14:27: Creating resource group Autodetect
13:14:27: SceneManagerFactory for type 'DefaultSceneManager' registered.
13:14:27: Registering ResourceManager for type Material
13:14:27: Registering ResourceManager for type Mesh
13:14:27: Registering ResourceManager for type Skeleton
13:14:27: MovableObjectFactory for type 'ParticleSystem' registered.
13:14:27: ArchiveFactory for archive type FileSystem registered.
13:14:27: ArchiveFactory for archive type Zip registered.
13:14:27: ArchiveFactory for archive type EmbeddedZip registered.
[... 766 lines cut ...]
10:18:49: Finished parsing scripts for resource group Internal
10:18:49: Creating resources for group Internal
10:18:49: All done
10:18:49: Added resource location '/usr/share/gz/gz-rendering7/media/materials/textures/' of type 'FileSystem' to resource group 'General'
10:18:49: OGRE EXCEPTION(9:UnimplementedException):  in GL3PlusTextureGpu::copyTo at ./RenderSystems/GL3Plus/src/OgreGL3PlusTextureGpu.cpp (line 685)

```

</details>

## Description
I want to simulate VTOL taking off and landing on the ship, hoping to have the same effect as https://www.youtube.com/watch?v=rJCEN5Htu2s
But now I have some problems:

1.There will be no waves when using Ogre1 and software rendering, and the 3D screen will flicker frequently
After executing the `gz sim car.world --render-engine ogre` command, the screen will flicker at high frequency. 
car.world is the world I created, including a VTOL, a ship and waves
[image]

2.Executing `LIBGL_ALWAYS_SOFTWARE=1 gz sim car.world` can be started successfully, but no waves are generated
[image]

3.The window will be closed immediately after execution with Ogre2. The command is `gz sim car.world` 
This problem is a bit similar to https://github.com/gazebosim/gz-sim/issues/1841.
I have tried `LIBGL_DRI3_DISABLE=1 gz sim car.world` so far it will flicker

iche033, 2023-07-12:
custom shaders support in ogre 1.x (used by waves) is not working well. As for ogre2 support on WSL, there are a few know issues, please see https://github.com/gazebosim/gz-sim/issues/920. You'll likely need a few patches mentioned in https://github.com/gazebosim/gz-sim/issues/920#issuecomment-1556206263

mixtah, 2023-07-19:
I have the same issue and similar setup (using Fortress), I found setting LIBGL_ALWAYS_SOFTWARE=1 does fix the immediate crash and flickering but it will still easily crash on some features. 
Only just starting a project in this setup so I'll see how far I get with this, and will keep an eye on this thread.