gazebosim/gz-sim #2068: custom plugins causing ignition fortress gazebo6 to fail to load world

Sebastian-Schroder, 2023-08-03:
I have created a [forum post](https://answers.gazebosim.org/question/29317/gazebo-fortress-custom-plugins-causing-scene-to-not-load/) with relevant information. but I've put the most pressing info here.

## Environment
* OS Version: 22.04
* Source or binary build?
  binary ignition fortress

## Description
* Expected behavior: building and running the example system plugin causes gazebo to fail to load and hang on a blank screen
* Actual behavior: gazebo should load the selected world and run the plugins alongside it 

## Steps to reproduce
1. install gazebo fortress
2. `git clone https://github.com/gazebosim/gz-sim.git -b ign-gazebo-5 `
3.  `cd gz-sim/examples/plugin/system_plugin`
4.  update CMakeList.txt by swapping calls from ignition-gazebo5 to ignition gazebo6
5. update sample_system.sdf to include a model i just added an include tag linking to a fuel model. `<include><uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Ground Plane</uri></include>`
6.  `mkdir build && cd build && cmake .. && make`
7. add build folder to environment variable `IGN_GAZEBO_SYSTEM_PLUGIN_PATH `
8. and finally run the sample system `cd .. && ign gazebo sample_system.sdf` 

notes: 
- I cloned the ign-gazebo5 branch rather than the ign-gazebo6 despite having gazebo6 installed is because ign-gazebo6 was designed after the name change back to gazebo from ignition, so all the header files reference that naming scheme. 

## Output

When the plugins are commented out of the world, loading the world results in the following simulation
[image]

when the plugins are included. the world looks like this:
[image]

gazebo versions:
[image]

Verbose output when running with plugins:
```
[Msg] Ignition Gazebo GUI    v6.11.0
[Dbg] [ign.cc:161] Subscribing to [/gazebo/starting_world].
[Dbg] [ign.cc:163] Waiting for a world to be set from the GUI...
[Dbg] [Gui.cc:253] Waiting for subscribers to [/gazebo/starting_world]...
[Msg] Received world [sample_system.sdf] from the GUI.
[Dbg] [ign.cc:167] Unsubscribing from [/gazebo/starting_world].
[Msg] Ignition Gazebo Server v6.11.0
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[Msg] Loading SDF world file[/home/sperospace/Git/gz-sim/examples/plugin/system_plugin/sample_system.sdf].
[Dbg] [Application.cc:92] Initializing application.
[GUI] [Dbg] [Application.cc:555] Create main window
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [SystemManager.cc:58] Loaded system [sample_system::SampleSystem] for entity [1]
[Dbg] [SystemManager.cc:58] Loaded system [sample_system::SampleSystem2] for entity [1]
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/default/control], [/world/default/control/state] and [/world/default/playback/control]
[Msg] Serving GUI information on [/world/default/gui/info]
[Msg] World [default] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/default/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:112:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:105:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/StyleDialog.qml:98:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:797] [QT] qrc:qml/Main.qml:102:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:797] [QT] qrc:/qml/PluginMenu.qml:27:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:241:3: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
[GUI] [Wrn] [Application.cc:797] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[... 136 lines cut ...]
[Msg] SampleSystem2::PostUpdate
[Msg] SampleSystem2::PreUpdate
[Msg] SampleSystem2::Update
[Msg] SampleSystem::PostUpdate
[Msg] SampleSystem2::PostUpdate
[Msg] SampleSystem2::PreUpdate
[Msg] SampleSystem2::Update
[Msg] [Msg] [Msg] SampleSystem2::PostUpdate
SampleSystem::PostUpdate
[Msg] SampleSystem2::PreUpdate
[Msg] SampleSystem2::Update
[Msg] SampleSystem::PostUpdate
[Msg] SampleSystem2::PostUpdate
[Msg] SampleSystem2::PreUpdate
[Msg] SampleSystem2::Update
[Msg] SampleSystem::PostUpdate
SampleSystem::PostUpdate[Msg] SampleSystem2::PostUpdate
[Msg] SampleSystem2::PreUpdate
[Msg] SampleSystem2::Update
[Msg] SampleSystem::PostUpdate
[Msg] SampleSystem2::PostUpdate
[Msg] SampleSystem2::PreUpdate

```
it continues running the sample system PreUpdate and PostUpdate methods until the simulation is closed

Sebastian-Schroder, 2023-08-04:
I further tested this by trying out downloading gazebo garden and following the same steps on the gz-sim7 branch, resulting in the same errors.
I also tried building all of the example plugins in gazebo garden, of which the only ones that didn't work were:

- [system_plugin](examples/plugin/system_plugin)
- [hello world](examples/plugin/hello_world)
- [custom component](examples/plugin/custom_component)
which all hung with a blank simulation as shown above.
-[rendering_plugins](example/plugin/rendering_plugin) this just couldn't find the shared library, which is probably my fault but i still want to raise it, in case it means anythings.

azeey, 2023-08-04:
If you are loading a custom world plugin, none of the default world plugins will be loaded. You'll have to tell Gazebo to load them yourself. Is that what's happening? Also see https://gazebosim.org/api/sim/7/server_config.html

Sebastian-Schroder, 2023-08-06:
@azeey thanks, I'll test that when I've got some free time. btw would you happen to know which plugins are loaded by default? thank you for the insight

azeey, 2023-08-08:
> @azeey thanks, I'll test that when I've got some free time. btw would you happen to know which plugins are loaded by default? thank you for the insight

You should be able to find the default plugins in the default config created in ` $HOME/.gz/sim/<#>/server.config`

Sebastian-Schroder, 2023-08-09:
yep works, thank you for the solution

closed 2023-08-09 by Sebastian-Schroder