gazebosim/gz-sim #2076: DetachableJoint doesn't attach the expected links with Bullet Featherstone physics

shameekganguly, 2023-08-14:
For the attached sdf, the expected behavior is that links `cylinder1` and `cylinder2` will fall together to the ground without any offset. Instead, `cylinder1` falls on top of `cylinder2` and `cylinder2` stays fixed in air. Note that the scene behaves as expected for `dart` physics, but the buggy behavior occurs for `bullet-featherstone` physics.
[detachable_joint.sdf.txt](https://github.com/gazebosim/gz-sim/files/12338303/detachable_joint.sdf.txt)

shameekganguly, 2023-08-14:
Is it possible that we are adding the fixed joint constraint to the wrong links at https://github.com/gazebosim/gz-physics/blob/871bab79146e7000fe9f4a9f3a0e80f944dcdcca/bullet-featherstone/src/JointFeatures.cc#L327?

mjcarroll, 2023-08-15:
I have confirmed this behavior.

azeey, 2023-08-15:
The issue seems like constraint is set between the root links of each model as indicated by the `-1` as the index used for each link in https://github.com/gazebosim/gz-physics/blob/871bab79146e7000fe9f4a9f3a0e80f944dcdcca/bullet-featherstone/src/JointFeatures.cc#L327-L329.

Also, `indexInModel` does not seem to be set anywhere, so fixing this might require fixing `indexInModel`.

mjcarroll, 2023-08-15:
> Also, indexInModel does not seem to be set anywhere, so fixing this might require fixing indexInModel.

It seems to be set for the parent link, but not the child.

closed 2023-10-23 by mjcarroll