gazebosim/gz-sim #2082: Libraries are not found

AreaScout, 2023-08-20:
\## Environment  
\* OS Version: Windows 11  
\* Binary build 7.5.0  
   - Rendering plugin: ogre2  
     - \[ \] running on a dual GPU machine (integrated GPU + discrete GPU)  
     - \[ \] running on real hardware  
     - \[ \] GPU is concurrently used for other tasks  
       - \[ \] desktop acceleration  
       - \[ \] video decoding (i.e. a playing Youtube video)  
       - \[ \] video encoding  
       - \[ \] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)  
   - Rendering system info:  
\[DxDiag.txt\](https://github.com/gazebosim/gz-sim/files/12388736/DxDiag.txt)

\## Description  
\* Expected behavior: gz sim should start with world emulation  
\* Actual behavior: It does not find the gz-sim library although the search path is right and the file exists

\## Output  
\`\`\`gz sim -s -r --iterations 2 worlds\\example\_photo\_shoot.sdf -v 3\`\`\`  
Library error for \[D:/gazebo\_sim-main/workspace/install/bin/gz-sim7-gz.dll\]: No such file or directory

AreaScout, 2023-08-24:
Ok I will close this as this doesn't happen anymore with latest git, despite from that working on Windows with the Gazebo GUI would be great, I hope this will get fixed soon.

traversaro, 2023-08-24:
Hello @AreaScout, what error do you have in running `gz sim -g`? I opened an issue dedicated to fix `gz sim -g` on Windows at https://github.com/gazebosim/gz-sim/issues/2089, it could be helpful if you provided the error you are obtaining there, thanks!

AreaScout, 2023-08-26:
@traversaro Ok I have to reopen this, again I have troubles with the library not found error message, Importer.dlload fails every time inside gz script, no matter what path I add to it, even if I directly copy the dll to the location gz starts, it does not dlopen it, what could that be? On my PC in my company this does not happen but that environment is screwed up with an unresolvable package conflict. I still got this

```(gz-test) D:\gazebo_sim-main\ws\install\bin>gz sim -s --verbose
Library error for [D:/gazebo_sim-main/ws/install/bin/gz-sim7-gz.dll]: No such file or directory

(gz-test) D:\gazebo_sim-main\ws\install\bin>```

and if I start **_simple_demo_qml.exe_** a window opens and closes immediately with this console output:

 ```
(gz-test) D:\gazebo_sim-main\ws\install\bin>D:\gazebo_sim-main\ws\src\gz-rendering\examples\simple_demo_qml\build\Release\simple_demo_qml.exe
[Msg] Loading plugin [gz-rendering-ogre2]
Error while loading the library [D:\gazebo_sim-main\ws\install\lib\gz-rendering-7\engine-plugins\gz-rendering-ogre2.dll]: "D:\gazebo_sim-main\ws\install\lib\gz-rendering-7\engine-plugins\gz-rendering-ogre2.dll": Das angegebene Modul wurde nicht gefunden.
[Err] [D:\gazebo_sim-main\ws\src\gz-rendering\src\RenderEngineManager.cc:534] Failed to load plugin [gz-rendering-ogre2] : couldn't load library on path [D:\gazebo_sim-main\ws\install\lib\gz-rendering-7\engine-plugins\gz-rendering-ogre2.dll].
Engine 'ogre2' is not supported
[Err] [D:\gazebo_sim-main\ws\src\gz-rendering\examples\simple_demo_qml\GzRenderer.cc:272] No cameras found. Scene will not be rendered```

traversaro, 2023-08-26:
How did you installed gz-sim ? Can you report the output of `conda info` and `conda list` of your environment? Thanks.

traversaro, 2023-08-26:
Also, it would be useful to know how the `simple_demo_qml` you are trying to compile was produced.

AreaScout, 2023-08-26:
I didn't install it, I build the whole project from source

```
(gz-test) D:\gazebo_sim-main\ws\install\bin>conda info

     active environment : base
    active env location : C:\Users\Area\miniconda3\envs\gz-test
            shell level : 2
       user config file : C:\Users\Area\.condarc
 populated config files : C:\Users\Area\.condarc
          conda version : 23.1.0
    conda-build version : not installed
         python version : 3.7.9.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __win=0=0
       base environment : C:\Users\Area\miniconda3\envs\gz-test  (writable)
      conda av data dir : C:\Users\Area\miniconda3\envs\gz-test\etc\conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\Area\miniconda3\envs\gz-test\pkgs
                          C:\Users\Area\.conda\pkgs
                          C:\Users\Area\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Area\miniconda3\envs\gz-test\envs
                          C:\Users\Area\.conda\envs
                          C:\Users\Area\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.1.0 requests/2.28.1 CPython/3.7.9 Windows/10 Windows/10.0.22621
          administrator : False
             netrc file : None
           offline mode : False
```

```
(gz-test) D:\gazebo_sim-main\ws\install\bin>conda list
# packages in environment at C:\Users\Area\miniconda3\envs\gz-test:
#
# Name                    Version                   Build  Channel
aom                       3.4.0                h0e60522_1    conda-forge
assimp                    5.0.1                h5510ddc_6
atomicwrites              1.4.0                      py_0
attrs                     22.1.0           py37haa95532_0
benchmark                 1.4.1                he025d50_3    conda-forge
blas                      1.0                         mkl
blosc                     1.21.0               h19a0ad4_1
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotlipy                  0.7.0           py37h2bbff1b_1003
bullet-cpp                3.21                 h3182a2c_3    conda-forge
bzip2                     1.0.8                he774522_0
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0               he04af86_2
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py37h2bbff1b_3
cfitsio                   4.1.0                h5a969a9_0    conda-forge
charset-normalizer        2.0.4              pyhd3eb1b0_0
cmake                     3.24.1               h39d44d4_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py37h03978a9_0    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.14           py37h03978a9_0    conda-forge
[... 162 lines cut ...]
urdfdom                   2.3.5                h2d74725_0    conda-forge
urdfdom_headers           1.1.0                h5362a0b_0    conda-forge
urllib3                   1.26.14          py37haa95532_0
vc                        14.2                 h21ff451_1
vcstool                   0.3.0                    pypi_0    pypi
vs2015_runtime            14.27.29016          h5e58377_2
websockets                10.3             py37hcc03f2d_0    conda-forge
wheel                     0.38.4           py37haa95532_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2              py37haa95532_2
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.4                hd77b12b_0
xz                        5.2.10               h8cc25b3_1
yaml                      0.2.5                he774522_0
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.11.0           py37haa95532_0
zlib                      1.2.12               h8cc25b3_2
zstandard                 0.19.0           py37h2bbff1b_0
zstd                      1.5.2                h19a0ad4_0
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

Regards,
Daniel

traversaro, 2023-08-26:
Apparently your conda environment gz-test has a mixture of conda-forge and anaconda/defaults packages, that is not a supported workflow (see https://conda-forge.org/docs/user/tipsandtricks.html#why-does-that-happen). Which command did you use to create that environment? A possible solution is to create a new environment passing also the -c nodefaults option to make sure that the defaults channel is not used.

AreaScout, 2023-08-26:
Well some packages needed I only found from anaconda packages, but that was not much or even wrong, but I don't think that this is the reason I have this path problems, what else could that cause ?

AreaScout, 2023-08-26:
I just followed this instructions 

https://gazebosim.org/docs/garden/install_windows_src

traversaro, 2023-08-26:
> Well some packages needed I only found from anaconda packages, but that was not much or even wrong, but I don't think that this is the reason I have this path problems, what else could that cause ?
> 

I am not sure if the problem are independent. Anyhow, probably you can debug the loading problems by installing  https://github.com/lucasg/Dependencies, launching it in the environment, load the dll that is causing loading problems and check if more information are printed.

traversaro, 2023-08-26:
> I just followed this instructions 
> 
> https://gazebosim.org/docs/garden/install_windows_src

Good point, I will double check, but probably -c nodefaults should be added there as well, probably it went unnoticed as most gz users actually use miniforge or mambaforge .

AreaScout, 2023-08-26:
Well I did a new environment and even with the **_-c nodefaults_** switch it added non conda-forge packages 

```
# packages in environment at C:\Users\Area\miniconda3\envs\gz-bin:
#
# Name                    Version                   Build  Channel
aom                       3.5.0                h63175ca_0    conda-forge
blosc                     1.21.4               hdccc3a2_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0            h63a05c6_1001    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
cfitsio                   3.470                h2bbff1b_7
cmake                     3.26.4               h1537add_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h1ea47a8_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.15          py311h1ea47a8_1    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
coverage                  7.3.0           py311ha68e1ae_0    conda-forge
cppzmq                    4.10.0               h449d27f_0    conda-forge
curl                      8.2.1                hd5e4a3a_0    conda-forge
dav1d                     1.2.0                hcfcfb64_0    conda-forge
distlib                   0.3.7              pyhd8ed1ab_0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
docutils                  0.20.1          py311h1ea47a8_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    6.0.0           gpl_h3a5b8a7_100    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               h33f27b4_0    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
freexl                    1.0.6                h67ca5e6_1    conda-forge
gdal                      3.6.2           py311h4e7b5b2_3
geos                      3.8.0                h33f27b4_0
geotiff                   1.7.0                h4545760_1
git                       2.42.0               h57928b3_0    conda-forge
glib                      2.69.1               h5dc1a3c_2
gst-plugins-base          1.18.5               h9e645db_0
gstreamer                 1.18.5               hd78058f_0
gts                       0.7.6                h69fa6b1_0    conda-forge
hdf4                      4.2.13               h712560f_2
hdf5                      1.12.1          nompi_h57737ce_104    conda-forge
humanfriendly             10.0            py311h1ea47a8_4    conda-forge
icu                       58.2                 ha925a31_3
[... 76 lines cut ...]
svt-av1                   1.5.0                h63175ca_0    conda-forge
tiledb                    2.3.3                hd8964de_3
tinyxml                   2.6.2                h2d74725_2    conda-forge
tinyxml2                  9.0.0                h0e60522_2    conda-forge
tk                        8.6.12               h8ffe710_0    conda-forge
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tomli                     2.0.1              pyhd8ed1ab_0    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
[... cut ...]
xerces-c                  3.2.4                h63175ca_3    conda-forge
xz                        5.4.2                h8cc25b3_0
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.16.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.2                h12be248_7    conda-forge
```

traversaro, 2023-08-27:
> Well I did a new environment and even with the **_-c nodefaults_** switch it added non conda-forge packages

That is quite strange. Can you provide exactly the commands you use to create this environment? From the docs in https://gazebosim.org/docs/garden/install_windows_src it seems that the commands should be:
~~~
conda create -n gz-ws
conda activate gz-ws
conda install cmake git vcstool curl pkg-config colcon-common-extensions eigen freeimage gdal gts glib dlfcn-win32 ffmpeg ruby tinyxml2 tinyxml  protobuf urdfdom zeromq cppzmq ogre jsoncpp libzip qt pybind11 --channel conda-forge
~~~

However, the fact that in your environemnts `pybind11` and `ogre` are missing suggest me that probably another command was used.

AreaScout, 2023-08-27:
pybind11 is installed, ogre and ogre-next as well for the **envs\\gz-test**, for the new one **envs\\gz-bin** I didn't install ogre because I only wanted to test if all packages got installed from conda-forge channel and by the way pybind11 is installed on both environments gz-test and gz-bin

Regards,

Daniel

traversaro, 2023-08-27:
Right, I missed the pybind11, sorry about that . Can you share the commands you used to create the env? I tried with a miniconda install and the instructions from the tutorial and in my case all the packages got installed from conda-forge.

traversaro, 2023-08-28:
Sorry, did you tried specifying `--channel conda-forge --channel nodefaults` ? For me, only `conda-forge` packages get installed by just specifying `--channel conda-forge`.

AreaScout, 2023-08-28:
Ah! sorry didn't report that back, yes I did check already and all dependencies are there, here is the output

```
dumpbin /dependents gz-rendering7-ogre2.dll
Microsoft (R) COFF/PE Dumper Version 14.37.32822.0
Copyright (C) Microsoft Corporation.  All rights reserved.

Dump of file gz-rendering7-ogre2.dll

File Type: DLL

  Image has the following dependencies:

    gz-rendering7.dll
    OgreHlmsPbs.dll
    OgreHlmsUnlit.dll
    OgreMain.dll
    OgreOverlay.dll
    gz-common5-events.dll
    gz-common5-graphics.dll
    gz-common5.dll
    gz-math7.dll
    gz-plugin2.dll
    MSVCP140.dll
    VCRUNTIME140_1.dll
    VCRUNTIME140.dll
    api-ms-win-crt-runtime-l1-1-0.dll
    api-ms-win-crt-string-l1-1-0.dll
    api-ms-win-crt-math-l1-1-0.dll
    api-ms-win-crt-stdio-l1-1-0.dll
    api-ms-win-crt-convert-l1-1-0.dll
    api-ms-win-crt-environment-l1-1-0.dll
    api-ms-win-crt-heap-l1-1-0.dll
    KERNEL32.dll

  Summary

        D000 .data
        9000 .pdata
       94000 .rdata
        5000 .reloc
        1000 .rsrc
       F1000 .text
       
```
And the output from the tool you mentioned ->

```
(gz-bin) D:\gazebo_sim-main\ws\install\bin>Dependencies.exe -chain gz-rendering7-ogre2.dll -depth 1
├ gz-rendering7-ogre2.dll (ROOT) : gz-rendering7-ogre2.dll
|  ├ gz-rendering7.dll (ApplicationDirectory) : gz-rendering7.dll
|  ├ OgreHlmsPbs.dll (Environment) : C:\Users\Area\miniconda3\envs\gz-bin\Library\bin\OgreHlmsPbs.dll
|  ├ OgreHlmsUnlit.dll (Environment) : C:\Users\Area\miniconda3\envs\gz-bin\Library\bin\OgreHlmsUnlit.dll
|  ├ OgreMain.dll (Environment) : C:\Users\Area\miniconda3\envs\gz-bin\Library\bin\OgreMain.dll
|  ├ OgreOverlay.dll (Environment) : C:\Users\Area\miniconda3\envs\gz-bin\Library\bin\OgreOverlay.dll
|  ├ gz-common5-events.dll (ApplicationDirectory) : gz-common5-events.dll
|  ├ gz-common5-graphics.dll (ApplicationDirectory) : gz-common5-graphics.dll
|  ├ gz-common5.dll (ApplicationDirectory) : gz-common5.dll
|  ├ gz-math7.dll (ApplicationDirectory) : gz-math7.dll
|  ├ gz-plugin2.dll (ApplicationDirectory) : gz-plugin2.dll
|  ├ MSVCP140.dll (ApplicationDirectory) : MSVCP140.dll
|  ├ VCRUNTIME140_1.dll (ApplicationDirectory) : VCRUNTIME140_1.dll
|  ├ VCRUNTIME140.dll (ApplicationDirectory) : VCRUNTIME140.dll
|  ├ api-ms-win-crt-runtime-l1-1-0.dll (ApiSetSchema) : C:\WINDOWS\system32\ucrtbase.dll
|  ├ api-ms-win-crt-string-l1-1-0.dll (ApiSetSchema) : C:\WINDOWS\system32\ucrtbase.dll
|  ├ api-ms-win-crt-math-l1-1-0.dll (ApiSetSchema) : C:\WINDOWS\system32\ucrtbase.dll
|  ├ api-ms-win-crt-stdio-l1-1-0.dll (ApiSetSchema) : C:\WINDOWS\system32\ucrtbase.dll
|  ├ api-ms-win-crt-convert-l1-1-0.dll (ApiSetSchema) : C:\WINDOWS\system32\ucrtbase.dll
|  ├ api-ms-win-crt-environment-l1-1-0.dll (ApiSetSchema) : C:\WINDOWS\system32\ucrtbase.dll
|  ├ api-ms-win-crt-heap-l1-1-0.dll (ApiSetSchema) : C:\WINDOWS\system32\ucrtbase.dll
|  ├ KERNEL32.dll (WellKnownDlls) : C:\WINDOWS\system32\kernel32.dll
```

AreaScout, 2023-08-29:
> So, passing `--channel conda-forge` to the `conda install` results in exactly the same packages installed in [#2082 (comment)](https://github.com/gazebosim/gz-sim/issues/2082#issuecomment-1695181809) ?

Of course, I did it several times now and the outcome was always the same

traversaro, 2023-08-29:
> > So, passing `--channel conda-forge` to the `conda install` results in exactly the same packages installed in [#2082 (comment)](https://github.com/gazebosim/gz-sim/issues/2082#issuecomment-1695181809) ?
> 
> Of course, I did it several times now and the outcome was always the same

Can you try to also add `--channel conda-forge` do the `create` step?

AreaScout, 2023-08-29:
Still the same:

```
(gz-bin) C:\Users\Area>conda list
# packages in environment at C:\Users\Area\miniconda3\envs\gz-bin:
#
# Name                    Version                   Build  Channel
aom                       3.5.0                h63175ca_0    conda-forge
blosc                     1.21.4               hdccc3a2_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_4    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0            h63a05c6_1001    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
cfitsio                   3.470                h2bbff1b_7
cmake                     3.26.4               h1537add_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h1ea47a8_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-notification       0.2.15          py311h1ea47a8_1    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
coverage                  7.3.0           py311ha68e1ae_0    conda-forge
cppzmq                    4.10.0               h449d27f_0    conda-forge
curl                      8.2.1                hd5e4a3a_0    conda-forge
dav1d                     1.2.0                hcfcfb64_0    conda-forge
distlib                   0.3.7              pyhd8ed1ab_0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
docutils                  0.20.1          py311h1ea47a8_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.3              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    6.0.0           gpl_h3a5b8a7_100    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0               h33f27b4_0    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
freexl                    1.0.6                h67ca5e6_1    conda-forge
gdal                      3.6.2           py311h4e7b5b2_3
geos                      3.8.0                h33f27b4_0
geotiff                   1.7.0                h4545760_1
git                       2.42.0               h57928b3_0    conda-forge
glib                      2.69.1               h5dc1a3c_2
gst-plugins-base          1.18.5               h9e645db_0
gstreamer                 1.18.5               hd78058f_0
gts                       0.7.6                h69fa6b1_0    conda-forge
hdf4                      4.2.13               h712560f_2
hdf5                      1.12.1          nompi_h57737ce_104    conda-forge
humanfriendly             10.0            py311h1ea47a8_4    conda-forge
[... 84 lines cut ...]
tiledb                    2.3.3                hd8964de_3
tinyxml                   2.6.2                h2d74725_2    conda-forge
tinyxml2                  9.0.0                h0e60522_2    conda-forge
tk                        8.6.12               h8ffe710_0    conda-forge
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tomli                     2.0.1              pyhd8ed1ab_0    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
ucrt                      10.0.22621.0         h57928b3_0    conda-forge
urdfdom                   3.1.1                h33150cf_0    conda-forge
urdfdom_headers           1.1.0                h5362a0b_0    conda-forge
vc                        14.3                h64f974e_17    conda-forge
[... cut ...]
xz                        5.4.2                h8cc25b3_0
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.16.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_5    conda-forge
zstd                      1.5.5                h12be248_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```

traversaro, 2023-08-29:
Sorry, I am out of ideas on why this is happening. Probably you can try to create the environment in one shot or similar, in general anyhow mixing defaults and conda-forge can lead to ABI and other linking problems such as the one you are experiencing here.

AreaScout, 2023-08-29:
I will do some more tests, It could also be the way that search path is generated bei cmake maybe there is some clash with some hard coded path, something like here:

```
#define GZ_RENDERING_RESOURCE_PATH _Pragma ("GCC warning \"'GZ_RENDERING_RESOURCE_PATH' macro is deprecated, use gz::rendering::getResourcePath() function instead. \"") "D:/gazebo_sim-main/ws/install/share/gz/gz-rendering7"

#define GZ_RENDERING_ENGINE_INSTALL_DIR _Pragma ("GCC warning \"'GZ_RENDERING_ENGINE_INSTALL_DIR' macro is deprecated, use gz::rendering::getEngineInstallDir() function instead. \"") "D:/gazebo_sim-main/ws/install/lib/gz-rendering-7/engine-plugins"
```

also I will check if I ruby is able to load a dll in general, test some windows API calls from kernel32.dll or so, will report back if I found something

Regards,
Daniel

traversaro, 2023-08-29:
If you are installing from source the hardcoded path should be ok. If you need to have relocatable packages, you can enable the `GZ_ENABLE_RELOCATABLE_INSTALL` CMake option, see https://github.com/gazebosim/gz-sim/issues/626 and the linked PRs.

AreaScout, 2023-08-29:
Hmm I found something interesting, the only dynamic link library able to load from ruby which was generated by the gazebo project was gz-tools2-backward.dll, all others fail to load, so it is proved that it has something to do with the libraries itself, I will try to recompile that project and look into that `GZ_ENABLE_RELOCATABLE_INSTALL` CMake option. Also all other libraries I have testet are able to be loaded 'kernel32.dll' etc.

AreaScout, 2023-08-30:
@traversaro Ok, I got it running and I know now what causes the **_Libraries are not found_** message, at least for my package configuration, I did a recompile yesterday and after this it was working, for a self written plugin I had to install **_opencv_** dependency and this package was downgrading **_protobuf_** after this I had to recompile gazebo, when it was finished the same error popped up again (**_Libraries are not found_**), remove **_opencv_** and install newer **_protobuf_** and recompile it again did make it work again.

So here is a console output from it working:

```
(gz-test) D:\gazebo_sim-main\ws\install>gz sim -s --verbose
[Wrn] [D:\gazebo_sim-main\ws\src\gz-common\src\Filesystem.cc:50] Failed common::createDirectories (ec: system:123 Die Syntax f³r den Dateinamen, Verzeichnisnamen oder die Datentrõgerbezeichnung ist falsch.)
Error opening log file: C:\Users\Area\.gz\sim\log\2023-08-30T08:29:46.276456400\server_console.log
[Msg] Gazebo Sim Server v7.5.0
[Msg] Loading default world.
[Msg] Serving entity system service on [/entity/system/add]
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Wrn] [D:\gazebo_sim-main\ws\src\gz-common\src\Util.cc:380] Reading environment variable with _allowEmpty==true is unsupported on Windows.
[Wrn] [D:\gazebo_sim-main\ws\src\gz-common\src\Util.cc:380] Reading environment variable with _allowEmpty==true is unsupported on Windows.
[Msg] Create service on [/world/default/create]
[Msg] Remove service on [/world/default/remove]
[Msg] Pose service on [/world/default/set_pose]
[Msg] Pose service on [/world/default/set_pose_vector]
[Msg] Light configuration service on [/world/default/light_config]
[Msg] Physics service on [/world/default/set_physics]
[Msg] SphericalCoordinates service on [/world/default/set_spherical_coordinates]
[Msg] Enable collision service on [/world/default/enable_collision]
[Msg] Disable collision service on [/world/default/disable_collision]
[Msg] Material service on [/world/default/visual_config]
[Msg] Material service on [/world/default/wheel_slip]
[Msg] Serving world controls on [/world/default/control], [/world/default/control/state] and [/world/default/playback/control]
[Msg] Serving GUI information on [/world/default/gui/info]
[Msg] World [default] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/default/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Msg] Serving scene information on [/world/default/scene/info]
[Msg] Serving graph information on [/world/default/scene/graph]
[Msg] Serving full state on [/world/default/state]
[Msg] Serving full state (async) on [/world/default/state_async]
[Msg] Publishing scene information on [/world/default/scene/info]
[Msg] Publishing entity deletions on [/world/default/scene/deletion]
[Msg] Publishing state changes on [/world/default/state]
[Msg] Publishing pose messages on [/world/default/pose/info]
[Msg] Publishing dynamic pose messages on [/world/default/dynamic_pose/info]
```
I will now test with GUI, for this I will compile either OGRE1.x or OGRE2.x into Gazebo as both of them compiled in seems to make problems --> Will report back.

Regards,
Daniel

AreaScout, 2023-08-30:
Running the GUI fails for me on both OGRE1.x and OGRE2.x, a window opens but does not respond anymore after a few seconds, sadly there are no more error messages to collect from console.

I will close this now and I hope it will help someone who also has problems with libraries not found.

@traversaro thanks for your help

For anyone wondering, to build gz-rendering7-ogre2.dll I was using this SDK package https://github.com/OGRECave/ogre-next/releases/download/v2.3.1/ogre-sdk-v2.3.1-vc19-x64.7z and installed it manually in my environment, as the package provided by conda seems to be library incomplete but the library naming is the same like OGRE1.x is using so you can eighter build with OGRE1.x or OGRE2.x if you want to do it that way.

And I used VS 2022 to build it, however the supported platform is VS 2019

Regards,
Daniel

traversaro, 2023-08-30:
> Running the GUI fails for me on both OGRE1.x and OGRE2.x, a window opens but does not respond anymore after a few seconds, sadly there are no more error messages to collect from console.
> 
> I will close this now and I hope it will help someone who also has problems with libraries not found.

I guess this is similar to https://github.com/gazebosim/gz-sim/issues/2089, even if nothing is printed on the console probably you can get some useful info from the file `<home>/.gz/rendering/ogre2.log` or `<home>/.gz/rendering/ogre.log`.

AreaScout, 2023-08-30:
> I guess this is similar to #2089, even if nothing is printed on the console probably you can get some useful info from the file `<home>/.gz/rendering/ogre2.log` or `<home>/.gz/rendering/ogre.log`.

I don't have a ogre2 or ogre.log, I have auto_default.log with endless entries of 

```
(2023-08-30T18:06:54.862715500) (2023-08-30T18:06:54.862789100) [GUI] [Dbg] [D:\gazebo_sim-main\ws\src\gz-sim\src\gui\Gui.cc:341] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
```

So that's why my GUI is busy all the time!

AreaScout, 2023-08-31:
Ok finally I did it:

[image]