gazebosim/gz-sim #2089: gz sim -g fails on Windows with error OGRE EXCEPTION(3:RenderingAPIException): wglMakeCurrent failed: The pixel format is invalid.

traversaro, 2023-08-24:
I'll open a new issue for gz sim -g support on Windows as https://github.com/gazebosim/gz-sim/issues/168 is a a bit generic and contains a lot of confusing information.

## Environment
* OS Version: Microsoft Windows [Version 10.0.22000.2176]
* Source or binary build?
  Binary from conda-forge, `mamba list` included in details

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2]. 
      - [ ] Sensor rendering error. 
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [x] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [x] GPU is concurrently used for other tasks
        - [x] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Windows, run `dxdiag` and report the GPU-related information.

    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.gz/rendering` 

<details>

The logs were to big for an issue, see them at https://gist.github.com/traversaro/bb1adcc11ce1a3baf7e1a712c226fb31 .

</details>

## Description
* Expected behavior: `gz sim -s` crashes, while `gz sim -g` crashes
* Actual behavior: `gz sim -s` followed by `gz sim -g`  opens the simulation Windows

## Steps to reproduce

1. Install gz-sim7 with `mamba create -n gzsim7 gz-sim7`
2. Manually comment these lines https://github.com/gazebosim/gz-sim/blob/gz-sim7_7.5.0/src/cmd/cmdsim.rb.in#L497-L501 in the file `%CONDA_PREFIX%\Library\lib\ruby\gz\cmdsim7.rb`
3. Run `gz sim -s` followed by `gz sim -g`
4.

## Output

~~~
(gzsim7) C:\Users\STraversaro>gz sim -g --verbose
[Msg] Gazebo Sim GUI    v7.5.0
[GUI] [Msg] Loading config [C:\Users\STraversaro\.gz\sim\7\gui.config]
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [MinimalScene] from path [C:\Users\STraversaro\AppData\Local\mambaforge\envs\gzsim7\Library\lib\gz-gui-7\plugins\MinimalScene.dll]
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [D:\bld\gz-gui7_1688332124286\work\src\Application.cc:860] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[... 283 lines cut ...]
#21   Object "", at 00007FFD82F3EC0C, in Ogre::ResourceGroupManager::parseResourceGroupScripts
#20   Object "", at 00007FFD82F73FA4, in Ogre::ScriptCompilerManager::parseScript
#19   Object "", at 00007FFD82F6B8D2, in Ogre::ScriptCompiler::compile
#18   Object "", at 00007FFD82F6B71D, in Ogre::ScriptCompiler::compile
#17   Object "", at 00007FFD82F940D2, in Ogre::MaterialTranslator::translate
#16   Object "", at 00007FFD82F82D98, in Ogre::ScriptTranslator::processNode
#15   Object "", at 00007FFD82F9CADF, in Ogre::TechniqueTranslator::translate
#14   Object "", at 00007FFD82F82D98, in Ogre::ScriptTranslator::processNode
#13   Object "", at 00007FFD82F95C55, in Ogre::PassTranslator::translate
#12   Object "", at 00007FFD82FE62C7, in Ogre::Technique::createPass
#11   Object "", at 00007FFD82EFF151, in Ogre::Pass::Pass
#10   Object "", at 00007FFD82E42EB8, in Ogre::Hlms::createDatablock
#9    Object "", at 00007FFDBD8C7BDE, in KiUserExceptionDispatcher
#8    Object "", at 00007FFDBD855BEA, in RtlRestoreContext
#7    Object "", at 00007FFDBD8C8BEF, in _chkstk
#6    Object "", at 00007FFD9CC9EFC0, in _C_specific_handler
Attempt to access invalid address.

#5    Object "", at 00007FF6C0BC1B9C, in  ??
#4    Object "", at 00007FFDBB5B1A82, in exp2f
#3    Object "", at 00007FFD9C9232B6, in ruby_signal_name
#2    Object "", at 00007FFD9C8143CF, in rb_bug_for_fatal_signal
#1    Object "", at 00007FFD9C812C7B, in ruby_stop
#0    Object "", at 00007FFDBB5CDD61, in abort
~~~

peci1, 2023-09-03:
The same happens to me with Intel HD Graphics 4600 and ogre2 (GPU should have OpenGL 4.3) on Harmonic.

traversaro, 2024-04-13:
To progress on this issue, I backtracked down to gz-rendering . I am tring some gz-rendering examples, and while some do not work on Windows, some work. In particular simple_demo and simple_demo_qml do not work, while https://github.com/gazebosim/gz-rendering/tree/gz-rendering8/examples/wide_angle_camera works.

closed 2024-06-26 by azeey