gazebosim/gz-sim #2099: Joint Position Controller Bug with multiple entities.

BA-Utkarsh, 2023-08-30:
## Environment
* OS Version: Ubuntu 22.04.2 LTS

## Description
- Included the Joint Position Controller plugin in our example.
- Found out that while clicking on multiple entities the list gets appended and doesn't get clear for the previous entity.

## Actual behavior:
- Clicked on L_0, we see `pan_joint` `tilt_joint` and `roll_joint`
[image]
- Clicked on L_1, we see additional `pan_joint` `tilt_joint` and `roll_joint` are added.
[image]

- **They are duplicates. And sliders are not working properly.**

## Expected behavior:
- Expectation is, the previous entity should be cleared out.
- Clicked on L_0, we see `pan_joint` `tilt_joint` and `roll_joint` for L_0 only.
[image]
- Clicked on L_2, we see `pan_joint` `tilt_joint` and `roll_joint` for L_2 only.
[image]

## Steps to reproduce
- We have created the minimal testcase which has 4 Logical Cameras (L_0, L_1, L_2 and L_3) with Joint Position Controller.
* Untar the zip file.  
  $ unzip [JPC_bug_testcase.zip](https://github.com/gazebosim/gz-sim/files/12473307/JPC_bug_testcase.zip)
```
Archive:  JPC_bug_testcase.zip
   creating: LogCamera/
  inflating: LogCamera/model.config
  inflating: LogCamera/model.sdf
  inflating: minimal_JPC_bug.sdf
```
* Source the gazebo path.
  $ source /opt/gazebo/install/setup.bash
* Export the model path.  
  $ export GZ_SIM_RESOURCE_PATH="$PWD:/opt/gazebo/models"
* Run.  
  $ gz sim ./minimal_JPC_bug.sdf

## Proposal
- From `steps to reproduce`, we reproduced the actual behavior.
- However, we have made the fix by [JPC_bug_fix.patch](https://github.com/gazebosim/gz-sim/files/12473373/JPC_bug_fix.patch)
- From above patch we can see the expected behavior.

**Let us know your valuable inputs on the same.**

iche033, 2023-08-30:
thanks for the patch. I can confirm that it fixes the bug. I've created https://github.com/gazebosim/gz-sim/pull/2101 based on the patch

BA-Utkarsh, 2023-08-31:
Thank you for the support @iche033

PR #2101 (2023-09-11): Fix duplicate entries in joint position controller GUI plugin

# 🦟 Bug fix

Fixes #2099

## Summary
Prevents duplicate entries from being added to the Joint Position Controller GUI plugin.

Applies the patch from #2099. See #2099 for instructions and sdf file for reproducing the bug.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing- [... cut]
closed 2023-09-11 by iche033