gazebosim/gz-sim #2112: gz-garden , gui process stuck

devashishRaj, 2023-09-01:
Brew Config :
```
HOMEBREW_VERSION: 4.1.7
ORIGIN: https://github.com/Homebrew/brew
HEAD: d4444b563e24ac7c05a93121c464c02dfa04d44f
Last commit: 5 days ago
Core tap JSON: 01 Sep 13:54 UTC
HOMEBREW_PREFIX: /opt/homebrew
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.06JH9bO1cO/org.xquartz:0
HOMEBREW_MAKE_JOBS: 10
Homebrew Ruby: 2.6.10 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: 10-core 64-bit arm_blizzard_avalanche
Clang: 14.0.3 build 1403
Git: 2.42.0 => /opt/homebrew/bin/git
Curl: 8.1.2 => /usr/bin/curl
macOS: 13.5.1-arm64
CLT: 14.3.1.0.1.1683849156
Xcode: 14.3.1
Rosetta 2: false
```

Installation : binary via homebrew :  brew info gz-garden
```
srf/simulation/gz-garden: stable 1.0.0, HEAD
Collection of gazebo simulation software
https://github.com/gazebosim/gz-garden
/opt/homebrew/Cellar/gz-garden/1.0.0_5 (93 files, 605.5KB) *
  Built from source on 2023-09-01 at 19:31:09
From: https://github.com/osrf/homebrew-simulation/blob/HEAD/Formula/gz-garden.rb
License: Apache-2.0
==> Dependencies
Build: cmake ✔, python@3.11 ✔
Required: gz-cmake3 ✔, gz-common5 ✔, gz-fuel-tools8 ✔, gz-gui7 ✔, gz-launch6 ✔, gz-math7 ✔, gz-msgs9 ✔, gz-physics6 ✔, gz-plugin2 ✔, gz-rendering7 ✔, gz-sensors7 ✔, gz-sim7 ✔, gz-tools2 ✔, gz-transport12 ✔, gz-utils2 ✔, pkg-config ✔, sdformat13 ✔
==> Requirements
Required: macOS >= 10.14 (or Linux) ✔
==> Options
--HEAD

```

**Bug : gui process is stuck** 

terminal 1 : gz sim -v 4 shapes.sdf -s

```
[Msg] Gazebo Sim Server v7.5.0
[Msg] Loading SDF world file[/opt/homebrew/Cellar/gz-sim7/7.5.0_8/share/gz/gz-sim7/worlds/shapes.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Msg] Loaded level [3]
[Msg] No systems loaded from SDF, loading defaults
[Dbg] [ServerConfig.cc:1058] Loaded (3) plugins from file [/Users/devashishraj/.gz/sim/7/server.config]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <BDD23500-3E96-3E94-8649-5113C449BFCF> /opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim7-physics-system.7.5.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::Physics]
- Requested library name: [gz-sim-physics-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-physics-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <8DC30481-777D-33F2-96F8-10AC02A2E5D2> /opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim7-user-commands-system.7.5.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
- Requested plugin name: [gz::sim::systems::UserCommands]
- Requested library name: [gz-sim-user-commands-system]
- Resolved library path: [/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-user-commands-system.dylib]
Error while loading the library [/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib]: dlopen(/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim-scene-broadcaster-system.dylib, 0x0085): Library not loaded: @rpath/libgz-sim7.7.dylib
  Referenced from: <72C09491-E96D-35E9-AB4C-FB64EC2A6ACC> /opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/libgz-sim7-scene-broadcaster-system.7.5.0.dylib
  Reason: tried: '/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-sim7/7.5.0_8/lib/gz-sim-7/plugins/../lib/libgz-sim7.7.dylib' (no such file), '/opt/homebrew/Cellar/gz-plugin2/2.0.1/lib/../lib/libgz-sim7.7.dylib' (no such file), '/usr/lib/libgz-sim7.7.dylib' (no such file, not in dyld cache), (security policy does not allow @ path expansion)
[Err] [SystemLoader.cc:140] Failed to load system plugin: (Reason: No plugins detected in library)
[... 61 lines cut ...]
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:54: TypeError: Cannot read property 'exitDialogShowCloseGui' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:53: TypeError: Cannot read property 'exitDialogShowShutdown' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:251: TypeError: Cannot read property 'showDrawer' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:/qml/SideDrawer.qml:93: TypeError: Cannot read property 'showDefaultDrawerOpts' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:/qml/GzSplit.qml:56: TypeError: Cannot read property 'pluginCount' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:41: TypeError: Cannot read property 'toolBarColorLight' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:42: TypeError: Cannot read property 'toolBarTextColorLight' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:43: TypeError: Cannot read property 'toolBarColorDark' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:44: TypeError: Cannot read property 'toolBarTextColorDark' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:45: TypeError: Cannot read property 'pluginToolBarColorLight' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:46: TypeError: Cannot read property 'pluginToolBarTextColorLight' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:47: TypeError: Cannot read property 'pluginToolBarColorDark' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:48: TypeError: Cannot read property 'pluginToolBarTextColorDark' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:50: TypeError: Cannot read property 'defaultExitAction' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:51: TypeError: Cannot read property 'showDialogOnExit' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:55: TypeError: Cannot read property 'exitDialogShutdownText' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:56: TypeError: Cannot read property 'exitDialogCloseGuiText' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:66: TypeError: Cannot read property 'toolBarColorLight' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:76: TypeError: Cannot read property 'toolBarTextColorLight' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:/qml/PluginMenu.qml:126: TypeError: Cannot call method 'PluginListModel' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:219: TypeError: Cannot read property 'showPluginMenu' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:202: TypeError: Cannot read property 'showDrawer' of null
[GUI] [Wrn] [Application.cc:859] [QT] qrc:qml/Main.qml:188: TypeError: Cannot read property 'showDrawer' of null

```

azeey, 2023-09-01:
https://github.com/osrf/homebrew-simulation/pull/2273 might be a potential solution. Would you be able to try it?

devashishRaj, 2023-09-01:
it's for build from source option :  https://gazebosim.org/docs/fortress/install_osx_src ?
 yeah i'll try.

devashishRaj, 2023-09-01:
i did try this one though : https://github.com/gazebosim/gz-sim/issues/1990#issuecomment-1573683249

but still same problem

devashishRaj, 2023-09-01:
i tried this :  **brew install --verbose ./ignition-fortress.rb**
from downloaded repo : https://github.com/srmainwaring/homebrew-simulation/tree/prs/pr-fix-gz-garden-rpath

but this resulted in same error : https://github.com/osrf/homebrew-simulation/issues/2347

and for : **brew install --verbose ./ignition-fortress.rb**

it's the same  gui process is stuck.. 

i tried the wrong method?

azeey, 2023-09-01:
I thought that PR was for Garden and your original issue was also for Garden.

devashishRaj, 2023-09-01:
yeah i tried on both , fortress is getting build error and on garden gui process is stuck .

devashishRaj, 2023-09-18:
I tried harmonic today 
server side : https://gist.github.com/devashishRaj/937e95e95ac24ab400c54fb7c5147ac4
gui : https://gist.github.com/devashishRaj/6045fbb76f1d5b5b8c8724be70240f89

devashishRaj, 2023-09-18:
fortress via brew has ogre2.2 build error : https://gist.github.com/devashishRaj/37546b0719b300043431615b1f9ef1ab

closed 2023-10-01 by devashishRaj