gazebosim/gz-sim #2125: How are models represented in the internal graph structure?

Levi-Armstrong, 2023-09-07:
I was looking the the Graph library in gz-math but did not see a concept for subgraph so was interested to know how models are represented internally?

mjcarroll, 2023-09-11:
Since we are using the entity-component pattern, models aren't stored as a graph, per se.  Instead, each entity has components to determine the type of that entity, (eg `Model`, `Link`, `Joint`), as well as information about parent/child relationships.  Typically, this is expressed with a `ParentEntity` component.

The ECM itself has some helper functions for retrieving these relationships in a gz::math::DirectedGraph structure: https://github.com/gazebosim/gz-sim/blob/5fbfcf7333398d88a346f3e82c93f2dba0343cde/include/gz/sim/EntityComponentManager.hh#L508-L518

I believe that the SDF FrameSemantics also makes use of the DirectedGraph class to store information about coordinate frames, but that doesn't impact simulation so much as loading/storing SDF data.

Levi-Armstrong, 2023-09-12:
Thank you for the information and I have been looking into entity-component pattern and this [repository](https://github.com/skypjack/entt) which has been helpful with understanding the concept if anyone else is interested.

closed 2023-09-12 by Levi-Armstrong

mjcarroll, 2023-09-12:
I would also check out https://github.com/SanderMertens/flecs.  They additionally have support for "Entity Relationships" which is similar to what we implement: https://www.flecs.dev/flecs/md_docs_Relationships.html