gazebosim/gz-sim #2146: linear_battery_demo.sdf service calls timeout although seems to work

j-rivero, 2023-09-14:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
  * Binary: gz-sim 8.0.0~pre1

Following the instructions [in the sdf file](https://github.com/gazebosim/gz-sim/blob/gz-sim8/examples/worlds/linear_battery_demo.sdf#L46) everything seems to work fine but the service calls returns a timeout although I see the expected effects in the logs.

## Description
* Expected behavior: returns nothing/success/true
* Actual behavior: returns a timeout

## Steps to reproduce

1. `gz sim -v 4 linear_battery_demo.sdf -z 10000`
2. Press play
3. `gz topic -t "/model/vehicle_blue/cmd_vel" -m gz.msgs.Twist -p "linear: {x: 6.0} angular: {z: 0.4}"`

## Output

```
gz service -s /model/vehicle_blue/battery/linear_battery/recharge/start --reqtype gz.msgs.Boolean --reptype gz.msgs.Empty --req 'data:true' --timeout 10000
Service call timed out
```

caguero, 2024-01-05:
I can't reproduce this from source. @j-rivero, do you have a chance to test it on your side with gz-sim 8 compiled from source?

j-rivero, 2024-01-11:
> I can't reproduce this from source. @j-rivero, do you have a chance to test it on your side with gz-sim 8 compiled from source?

I can not reproduce it anymore, with the current stable packages, gz-sim `version 8.0.0-1`.

caguero, 2024-01-11:
My explanation to this issue is that this request is a two-way message. The request is sent from the sender to destination (`gz-sim`) in this case. Then, `gz-sim` reacts to the request and sends the response. If for some reason, `gz-sim` is `slow` processing the request, the response that is sent back to the original sender might arrive after the timeout. That could be the explanation for seeing a timeout on the requester and a normal reaction to the request on `gz-sim`.

azeey, 2024-01-11:
I'm able to reproduce this locally with a clean build of Harmonic (gz-sim 82969d554, and gz-transport 764a04bf).

caguero, 2024-02-02:
https://github.com/gazebosim/gz-transport/pull/477