gazebosim/gz-sim #2153: Elevator state is not being publish on the ground floor

Voldivh, 2023-09-18:
## Environment
* OS Version: Ubuntu Jammy
* Source and Binary

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1. Run the elevator.sdf file:
```bash
gz sim elevator.sdf
```
2. Subscribe to the elevator state:
```bash
gz topic -e -t /model/elevator/state
```
3. Publish a floor for the elevator to go. Example:
```bash
gz topic -t "/model/elevator/cmd" -m gz.msgs.Int32 -p "data: 2"
```

## Output
We get the expected behavior when we send the elevator to any of the upper floors:
[image]
However, when we send the elevator to the ground floor (`gz topic -t "/model/elevator/cmd" -m gz.msgs.Int32 -p "data: 0"`), we stop receiving information on the console:
[image]

Voldivh, 2023-09-18:
Interesting. I don't think it is critical to be able to echo it as long as the we still get the correct value on the subscription. My only concern is that it may confuse users that are not aware of that specific behavior. Maybe we could add a little comment at the top of the .sdf file? I can go ahead and create the appropriate PR if you agree on the "solution".

mjcarroll, 2023-09-18:
I think a comment would be helpful.

Or we can 1-index all of our arrays from now on. :wink:

Voldivh, 2023-09-18:
I agree, let's go with the comment for now and keep it in mind from now on.

PR #2227 (2023-11-03): Add note about elevator example
Closes #2153
commit 45233fc3 (2023-11-03): Add note about elevator example (#2227)

Closes #2153
closed 2023-11-03 by mjcarroll