gazebosim/gz-sim #2154: app.gazebosim.org model upload broken?

kaiaai, 2023-09-19:
Hi, I'm not sure where to file this bug, but app.gazebosim.org seems broken, gives me "Error #150000 Unexpected error" when I try uploading a (working and tested) SDF model.

[image]

jarkenau, 2023-09-19:
Not only uploading is broken, downloading assets also doesn't work for me:

```shell
[Err] [FuelClient.cc:717] Failed to download model.
[ruby $(which ign) gazebo-1]   Server: https://fuel.gazebosim.org
[ruby $(which ign) gazebo-1]   Route: openrobotics/models/jersey barrier/tip/jersey barrier.zip
[ruby $(which ign) gazebo-1]   REST response code: 500
```

mjcarroll, 2023-09-19:
Thanks for reporting. It seems the fuel server backend is having some issues.  @nkoenig is going to take a look at it.

azeey, 2023-09-19:
The issue should be resolved now. Feel free to reopen if otherwise.

closed 2023-09-19 by azeey

kaiaai, 2023-09-21:
Thank you, I'm afraid the server is acting up again. It did work fine during the day, but the download seems broken again. That said, manual download from app.gazebosim.org still works, albeit slowly.
```
(1695287193 236285037) [Err] [FuelModelDatabase.cc:258] Unable to download model[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sofa/2/files/meshes/Sofa.obj]
(1695287193 236369357) [Err] [FuelModelDatabase.cc:320] Unable to download model[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sofa/2/files/meshes/Sofa.obj]
(1695287193 236384857) [Wrn] [SystemPaths.cc:459] File or path does not exist [""] [https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sofa/2/files/meshes/Sofa.obj]
(1695287193 236402777) [Err] [Visual.cc:2956] No mesh specified
(1695287253 704461115) [Err] [FuelModelDatabase.cc:258] Unable to download model[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sofa/2/files/meshes/Sofa.obj]
(1695287253 704540625) [Err] [FuelModelDatabase.cc:320] Unable to download model[https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sofa/2/files/meshes/Sofa.obj]
(1695287253 704552555) [Wrn] [SystemPaths.cc:459] File or path does not exist [""] [https://fuel.ignitionrobotics.org/1.0/openrobotics/models/sofa/2/files/meshes/Sofa.obj]
(1695287253 704567425) [Err] [Visual.cc:2956] No mesh specified
```