gazebosim/gz-sim #2160: Include arm64 machines as used in MAC M1 Virtual, Raspberry Pi & Nvidia machine as a gz-sim supported platform

ARLunan, 2023-09-20:
**Required Info**
Operating System: Ubuntu 22.04.3 arm64
ROS2 Version: Iron Binary

**Desired behavior**
Install Gazebo Harmonic and ros-gz on arm64 platforms such as MAC M1 Virtual (NOT native MAC functionality), Raspberry Pi or Nvidia machines used as a ROS 2 Desktop,

azeey, 2023-12-13:
Running `bash terminal-dashboard/table.bash harmonic` shows that arm64 packages are available, so it should be possible to build `ros-gz` from source on arm64.
[image]

ARLunan, 2023-12-23:
After running a rosdep, just attempted to colcon compile on a arm64 Ubuntu 22.04 VM on a MAC M1 - failed unable to locate cmake3. Searching packages.ubunt.com , none of the above libgz are available.

ARLunan, 2023-12-24:
Then tried Source install ros_gz,: rosdep completed. But Build failed due missing sim8, expected as build not completed in gz_sim Build. SO in summary, please consider completing a arm64 Ubuntu 22.04 Binary Install of gz_sim and ros_gz :)

AndromedaHelix, 2023-12-26:
Hi, have there been any fixes to this issue?

ARLunan, 2023-12-26:
Pertaining to gz_sim, yes. Following Installation from Binary ,  (https://gazebosim.org/api/sim/8/install.html)  on my Ubuntu 22.04 MAC M1 arm64 Parallels 18 VM (and I believe would be the case on a native x86-64 also),  successfully installed gz_sim. executing $ gz sim --version displays : Gazebo Sim, version 8.0.0 .
This includes libgz-sim8-dev and the set of gz_sim packages gz-cmake3 gz-tools2 gz-utils2 gz-math7 gz-common5 gz-msgs10 sdformat14 gz-fuel_tools9 gz-plugin2 gz-transport13 gui8 launch7 sim8 sensors8 gz-physics7 gz-harmonic. I actually installed these 16 packages manually with Synaptic (GUI dpkg) .

However, there remains an issue with Source install of ros_gz on Humble (even provided executing $ export GZ_VERSION=harmonic as  necessary dependency ros2/ros2 rosidl_pycommon doesn't exist in Humble. ( [https://github.com/ros2/rosidl/issues/777](https://github.com/ros2/rosidl/issues/777) ). 
I posted a bug issue (https://github.com/gazebosim/ros_gz/issues/474) which was moved to github-project-automation Inbox in [Core development](https://github.com/orgs/gazebosim/projects/7).

AndromedaHelix, 2023-12-27:
Yesterday I installed Gazebo Fortress from this link: https://gazebosim.org/docs/fortress/install_ubuntu

And ran it with `ign gazebo --render-engine ogre` and had no problems. Then I installed ros_gz through `sudo apt install ros-humble-ros-gz` and had no issues.

What's the difference between my method to install gazebo and yours?

ARLunan, 2023-12-28:
To answer your install method question: I have a Ubuntu 22.04 arm64 machine (MAC M1 with Parallels VM), vs your x86-64 Intel machine. I can binary install gz_sim but not ros_gz. A ros_gz Source that I must use now that fails, install requires a ros2 dependency with is not available on Humble. I hope that is fixed by adding an arm64 Binary install to the already available x86-64 install, (https://github.com/gazebosim/ros_gz/issues/474) .

AndromedaHelix, 2023-12-29:
Thanks for replying! 

I have a Mac m1 with Asahi ubuntu, so I'm on a similar situation as you. I am running ros2 humble without any issues. I didn't have any issues downloading ros_gz. I installed through `sudo apt install ros-humble-ros-gz`. Not sure if I'm missing something here. 

Maybe try using Gazebo Fortress with ign_ros_gz ?

ARLunan, 2023-12-29:
Here’s my Ubuntu 22.04/humbe terminal

￼

ARLunan, 2023-12-29:
Andromeda, Here’s a capture of my Ubuntu 22.04 VM Terminal, that has both Gazebo Ignition and Gazebo Harmonic Installed, but no ros_gz.

Note the error that is generated by the non-available humble rosicl-pycommon package.
Ross

***@***.***:~$ uname -a
Linux macvm-ub22h-bv 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 20:26:31 UTC 2 aarch64 aarch64 aarch64 GNU/Linux
***@***.***:~$ ign gazebo --version
Gazebo Sim, version 6.15.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.

***@***.***:~$ gz sim --version
Gazebo Sim, version 8.0.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.

***@***.***:~$ printenv|grep GZ
GZ_VERSION=harmonic
***@***.***:~$ sudo apt install ros-humble-ros-gz
[sudo] password for ubuntutbc: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help resolve the situation:

The following packages have unmet dependencies:
 libdart6.13-collision-bullet-dev : Breaks: libdart-collision-bullet-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
 libdart6.13-collision-ode-dev : Breaks: libdart-collision-ode-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
 libdart6.13-dev : Breaks: libdart-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
 libdart6.13-external-convhull-3d-dev : Breaks: libdart-external-convhull-3d-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
 libdart6.13-external-ikfast-dev : Breaks: libdart-external-ikfast-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
 libdart6.13-external-odelcpsolver-dev : Breaks: libdart-external-odelcpsolver-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
 libdart6.13-utils-dev : Breaks: libdart-utils-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
 libdart6.13-utils-urdf-dev : Breaks: libdart-utils-urdf-dev (< 6.14.0) but 6.12.1+dfsg4-11build2 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
***@***.***:~$

azeey, 2024-01-02:
As mentioned in https://github.com/gazebosim/ros_gz/issues/474#issuecomment-1874364563, you need to use the `humble` branch when building `ros_gz` from source with with ROS 2 Humble. The `ros2` branch is equivalent to `rolling`. 

We currently provide binary packages for the ROS 2 Humble / Gazebo Fortress combination. That's what `ros-humble-ros-gz` is. Other binaries are available (eg. ros-humble-ros-gzgarden, see https://gazebosim.org/docs/harmonic/ros_installation), but these are not available for `arm64`.

ARLunan, 2024-01-03:
> 

Andromeda. As I mentioned in https://github.com/gazebosim/ros_gz/issues/474#issuecomment-1876045861 , Success compiling ros_gz from Source : Summary: 12 packages finished [37 secs] with the allocating 4 Cores CPU /8 GB Memory and export MAKEFLAGS="-j 1". Thank you. 

So the ros_gz Harmonic (Humble branch) Install on a ROS 2 Humble from Source on a suitably powerful arm64 machine works . Thank you (and Azeey) for assistance.

AndromedaHelix, 2024-01-04:
Perfect, I'm glad you've managed to compile it from source

ARLunan, 2024-01-09:
For ros_rz and ros_gz_project_template source build, suggest adding these  parameters to the READ.ME (such as on arm64 machines) , namely use the humble branch, before running colcon build in a Terminal: 
$ export MAKEFLAGS="-j 1"                                                
$ workspace/colcon build —executor sequential —parallel-workers=1, and in
ros_gz_sim_demos/hooks/ros_gz_demos.dev.in , revise IGN_GAZEBO_RESOURCE_PATH to GZ_SIM_RESOURCE_PATH.  

Finally consider release a arm64 Binary Install for ros_gz and ros_gz_project_template.

closed 2024-01-09 by ARLunan