gazebosim/gz-sim #2161: gz-msgs deprecation message when using The Topic Viewer GUI plugin

azeey, 2023-09-21:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? source, binary (Harmonic prereleases)

## Description
* Expected behavior:  No warning message when using the plugin
* Actual behavior: The following warning message is emitted when starting the Topic Viewer GUI plugin

```
Message (gz.msgs.Header) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
Message (gz.msgs.Time) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
Message (gz.msgs.Time) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
Message (gz.msgs.Time) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
Message (gz.msgs.Time) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
Message (gz.msgs.Header) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
Message (gz.msgs.Time) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
Message (gz.msgs.Header) was retrieved with non-fully qualified name. This behavior is deprecated in msgs10
...
```

## Steps to reproduce

1. Run `gz sim -v2 shapes.sdf`
2. Enable the Topic Viewer plugin from the GUI plugin search box
3. The warning should be printed on the console

azeey, 2023-09-21:
@mjcarroll mind taking a look?

mjcarroll, 2023-09-21:
https://github.com/gazebosim/gz-gui/pull/575

closed 2023-09-21 by azeey