gazebosim/gz-sim #2163: [GUI] [Err] [Conversions.cc:1198] Attempting to convert a NavSat SDF sensor, but the sensor pointer is null.

ashBabu, 2023-09-22:
Ubuntu 22.04, ROS2 Humble, Gazebo Fortress. Installed everything from `apt` repos

I have a differential drive robot on to which I have added a [`navsat` sensor](https://github.com/ashBabu/skidy/blob/ros2-ignition-gazebo/skidy_description/urdf/robot_test_diff_drive.urdf#L155). I launch the simulation with `ros2 launch skidy_bringup bringup_sim.launch.py`. As soon as press the play button in gazebo, the following error comes up although I am able to see the msg being published at the topic that I specified.

`[GUI] [Err] [Conversions.cc:1198] Attempting to convert a NavSat SDF sensor, but the sensor pointer is null.`

azeey, 2023-09-26:
It looks like we are missing a `msgs` -> `sdf` conversion for `NavSat` in `gz::sim::convert(const msgs::Sensor)` . Interestingly, it was left as a TODO in the original PR https://github.com/gazebosim/gz-sim/pull/1248. @ashBabu would you be able to add the conversion similar to how https://github.com/gazebosim/gz-sim/blob/f55212eb271a2f2b4d726da45e3aaeebede3b1a2/src/Conversions.cc#L1084-L1116 is done, but in the reverse direction in `gz::sim::convert(const msgs::Sensor)`  https://github.com/gazebosim/gz-sim/blob/f55212eb271a2f2b4d726da45e3aaeebede3b1a2/src/Conversions.cc#L1366

ashBabu, 2023-09-26:
Hi @azeey. I will try to update and create a PR asap

ashBabu, 2023-09-27:
Hi @azeey : I did make a quick code edit as you can see [here](https://github.com/ashBabu/gz-sim/blob/ign-gazebo6/src/Conversions.cc#L1428-L1456). However it doesnt solve the error but added another one 
```
[GUI] [Err] [Conversions.cc:1198] Attempting to convert a NavSat SDF sensor, but the sensor pointer is null.
[Err] [Conversions.cc:1453] Attempting to convert an navsat sensor message, but the message does not have a navsat nested message.
```
Wondering if Im on the right track? Any pointers?

ashBabu, 2023-09-27:
@azeey : Please ignore the above. I found out what I was doing wrong and a PR #2176 was made. Pls review it

23pointsNorth, 2023-10-03:
I am attempting to use GPS/navsat sensor in Gazebo Garden, but I am hitting the same issues. Would the above PR make it in gz-sim7 or 8 as well?

ashBabu, 2023-10-03:
@23pointsNorth : Although the error appears, the navsat sensor still publishes msgs. `ign topic -e -t topic_name`. If you want it in ROS, use the bridge
```
bridge = Node(
        package='ros_gz_bridge',
        executable='parameter_bridge',
        arguments=[
                   '/topic_name@sensor_msgs/msg/NavSatFix@ignition.msgs.NavSat',
                   ],
        output='screen'
    )
```
But to answer your question, someone from gazebo has to step in

mjcarroll, 2023-10-30:
Closed by #2176

closed 2023-10-30 by mjcarroll

23pointsNorth, 2023-10-30:
Is there a schedule on how quickly these changes will get propagated to published 6.x and 7.x?

azeey, 2023-10-31:
> Is there a schedule on how quickly these changes will get propagated to published 6.x and 7.x?

Generally, we wait for PRs that are close to getting in for the version/branch in question before making a release, but we can make a release sooner if necessary.

amar-jay, 2025-02-02:
any update still getting the same error but this time with ardupilot Error [Utils.cc:176] Conversion of sensor type: [navsat] from SDF DOM to Element is not supported yet.navsat_sensor

bastianhjaeger, 2025-02-06:
To me it looks like there is an `else if` missing here
https://github.com/gazebosim/sdformat/blob/sdf15/src/Sensor.cc#L830-L836

azeey, 2025-02-06:
I think you're right. Mind creating a PR?

bastianhjaeger, 2025-02-06:
I sure can try.

I created https://github.com/gazebosim/sdformat/pull/1538 but I only get a static GPS position for a moving actor.

PR #2176 (2023-10-30): Fix for sensor pointer null when navsat plugin in included in sdf
# 🦟 Bug fix

Fixes #2163

## Summary
When the play button is pressed in Gazebo, the terminal window shows an error as described in #2163. 

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/al [... cut]