gazebosim/gz-sim #2165: Removing model with `JointPositionController` plugin leads to 'entity does not exist' error messages.

jrutgeer, 2023-09-22:
When a model that uses the `JointPositionController` plugin is removed from the simulation through the UserCommands service, repeating error messages appear:

```
[Err] [EntityComponentManager.cc:1083] Trying to create a component of type [8319580315957903596] attached to entity [12], but this entity does not exist. This create component request will be ignored.
```

See the attached example files:

Run `gz sim -r world.sdf`
Spawn a model with the spawn script: `./spawn_boxes`
Remove the boxes with the remove script: `./remove_boxes`

--> The error messages start.

Edit the `boxes.sdf` file and remove the `JointPositionController` plugin.
Repeat the test.

--> No error messages after removal.

[example.tar.gz](https://github.com/gazebosim/gz-sim/files/12704007/example.tar.gz)

arjo129, 2023-11-06:
I think if we solve #2217 this will no longer be an issue.

jrutgeer, 2023-11-06:
Closing as duplicate of https://github.com/gazebosim/gz-sim/issues/2217.

closed 2023-11-06 by jrutgeer