gazebosim/gz-sim #2186: :woman_farmer: ActorFixture.ActorTrajectoryNoMesh

claraberendsen, 2023-10-02:
## Environment
* OS Version:  Ubuntu Focal
Reference build: https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-focal-amd64/171/

Test regression: [ActorFixture.ActorTrajectoryNoMesh](https://build.osrfoundation.org/job/ignition_gazebo-ci-gz-sim7-focal-amd64/171/testReport/junit/(root)/ActorFixture/ActorTrajectoryNoMesh/)

## Output

```
255: [Dbg] [RenderUtil.cc:2720] Create scene [scene]
255: [Dbg] [Sensors.cc:277] Rendering Thread initialized
255: Initialization neededInitialization needed
255: [Dbg] [CameraSensor.cc:445] Camera images for [camera::link::camera] advertised on [camera]
255: [Dbg] [CameraSensor.cc:756] Camera info for [camera::link::camera] advertised on [/camera_info]
255: [Msg] Loading plugin [gz-rendering-ogre2]
255: /home/jenkins/workspace/ignition_gazebo-ci-gz-sim7-focal-amd64/ign-gazebo/test/integration/actor_trajectory.cc:161: Failure
255: Expected: (poses[i]) != (poses[i+1]), actual: -1.00398 -0.995952 1 0 0 0 vs -1.00398 -0.995952 1 0 0 0
255: [Dbg] [Sensors.cc:433] SensorsPrivate::Stop
255: [Dbg] [Sensors.cc:419] SensorsPrivate::RenderThread stopped
255: [Dbg] [SimulationRunner.cc:530] Exiting postupdate worker thread (Exiting postupdate worker thread (1)
255: 0)
255: [Dbg] [Sensors.cc:433] SensorsPrivate::Stop
255: [  FAILED  ] ActorFixture.ActorTrajectoryNoMesh (1899 ms)
255: [----------] 1 test from ActorFixture (1899 ms total)
```

#### Job flakiness

job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
ignition_gazebo-ci-gz-sim7-focal-amd64|2023-09-26|2023-09-11|9|4|44.44
ignition_gazebo-ci-ign-gazebo6-focal-amd64|2023-09-16|2023-09-16|4|1|25.0
gz_sim-ci-gz-sim8-jammy-amd64|2023-09-11|2023-09-11|11|1|9.09

iche033, 2023-10-06:
potential fix in #2196

claraberendsen, 2023-11-02:
@azeey we should reopen this issue since the same issue has appeared again. It seems I don't have the permissions to reopen it.

iche033, 2023-11-02:
The fix in https://github.com/gazebosim/gz-sim/pull/2196 was merged into gz-sim7. 
https://github.com/gazebosim/gz-sim/pull/2224 should fix the gz-sim8 failure. We'll need to backport the fix to ign-gazebo6 to fix the other build failure

iche033, 2023-11-02:
> We'll need to backport the fix to ign-gazebo6 to fix the other build failure

https://github.com/gazebosim/gz-sim/pull/2225

Crola1702, 2024-04-08:
This issue hasn't happened again since 2023-11-01. I'll close it for now

PR #2196 (2023-10-10): Relax pose check in flaky ActorTrajectoryNoMesh test

# 🦟 Bug fix

Fixes #2186

## Summary

The test fails if the box does not move every iteration. This likely occurs when the rendering thread is not able to update as fast as the main update thread. This PR now checks pose difference between every other pose instead of consecutive poses.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](h [... cut]
PR #2225 (2023-11-03): Backport ActorTrajectoryNoMesh test fix

# 🦟 Bug fix

Fixes #2186

## Summary
Backport fix from #2196

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a review on your PR, please help review [another [... cut]
closed 2024-04-08 by Crola1702