gazebosim/gz-sim #2194: Requesting scene state fails from the same process as Server

shameekganguly, 2023-10-06:
Requesting ECM state from the state service at `/world/default/state` from the same process that is running the `Server` always fails. See the following snippet for example:
```
gz::sim::ServerConfig server_config;
const std::string sdf_path = "sphere.sdf";
server_config.SetSdfFile(sdf_path);
gz::sim::Server server(server_config);
server.Run(/*_blocking=*/false);

absl::SleepFor(absl::Seconds(5));

const std::string service_name = "/world/default/state";

gz::transport::Node node;
gz::msgs::SerializedStepMap reply;
std::cout << "Request state" << std::endl;
bool request_success = false;
if (!node.Request(service_name, 10000 /*timeout in ms*/, reply,
                  request_success)) {
  std::cerr << "Failed to request state, transport failure" << std::endl;
}
if (!request_success) {
  std::cerr << "Failed to request state, server failure" << std::endl;
}
```
for a simple sdf file which has a sphere and the physics and scene broadcaster plugins, will fail with `Failed to request state, server failure`.

iche033, 2023-10-06:
Would the non-blocking version of `Request` work for you? [example usage](https://github.com/gazebosim/gz-sim/blob/f6efeefb4ab99284d1a053061235f8c2fa02a5da/src/gui/plugins/component_inspector/ComponentInspector.cc#L1116)

iche033, 2023-10-06:
actually I think it still blocks. I see this [workaround](https://github.com/gazebosim/gz-sim/blob/f6efeefb4ab99284d1a053061235f8c2fa02a5da/test/integration/scene_broadcaster_system.cc#L475-L481) in the test.

shameekganguly, 2023-10-06:
Thanks for the pointer Ian, I can use the thread workaround. Am still wondering if this is working as intended. If so, please feel free to close this bug.

iche033, 2023-10-06:
I recall that the `/world/<world_name>/state_async` topic was created due to this issue. Our GUI has also switched to using the async topic as well. It's not very well documented. I'll close this issue since an alternative topic / workaround exists.

closed 2023-10-06 by iche033