gazebosim/gz-sim #2204: :farmer: 10 `INTEGRATION` tests failing in MacOS Monterey and Ventura in gz-sim3

Crola1702, 2023-10-11:
## Environment
* OS Version: MacOS Monterey and Ventura
* Source gz-sim3 homebrew

## Description
* Expected behavior: Tests should pass
* Actual behavior: Intergration tests are segfaulting

Similar issue in: https://github.com/gazebosim/gz-sensors/issues/393

## Steps to reproduce

1. Run a build in https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/
2. See INTEGRATION tests fail

## Output

Reference build: https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/#showFailuresLink

Test regression:
* [INTEGRATION_ackermann_steering_system.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_ackermann_steering_system/test_ran/)
* [INTEGRATION_breadcrumbs.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_breadcrumbs/test_ran/)
* [INTEGRATION_contact_system.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_contact_system/test_ran/)
* [INTEGRATION_detachable_joint.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_detachable_joint/test_ran/)
* [INTEGRATION_diff_drive_system.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_diff_drive_system/test_ran/)
* [INTEGRATION_joint_controller_system.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_joint_controller_system/test_ran/)
* [INTEGRATION_joint_state_publisher_system.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_joint_state_publisher_system/test_ran/)
* [INTEGRATION_multicopter.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_multicopter/test_ran/)
* [INTEGRATION_physics_system.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_physics_system/test_ran/)
* [INTEGRATION_touch_plugin.test_ran](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/330/testReport/junit/(root)/INTEGRATION_touch_plugin/test_ran/)

Log output:

Output is just the SegFault message without any other information
```
 69/178 Test  #69: INTEGRATION_breadcrumbs ..............................***Exception: SegFault  3.17 sec
```

This failure is only happening in gz-sim3 homebrew:
job_name|last_fail|first_fail|build_count|failure_count|failure_percentage
-- | -- | -- | -- | -- | --
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|2023-10-10|2023-10-09|8|3|37.5

azeey, 2023-10-16:
Let's try to backport https://github.com/gazebosim/gz-physics/pull/529.

scpeters, 2023-10-30:
> Let's try to backport [gazebosim/gz-physics#529](https://github.com/gazebosim/gz-physics/pull/529).

back-porting in https://github.com/gazebosim/gz-physics/pull/564

scpeters, 2023-11-01:
the backport didn't help. I can't reproduce the failures on my machine

azeey, 2023-11-10:
Fixed by #2220

PR #2220 (2023-11-01): Fix macOS test failures by registering components in the core library
# 🦟 Bug fix

Fixes #2204
Needs https://github.com/gazebo-tooling/release-tools/pull/1051

## Summary
A better solution to the problem of components being registered by plugins is in https://github.com/gazebosim/gz-sim/pull/1836, but backporting that to citadel was not trivial. The solution in this PR is to force-register all built-in components in the core library by including the `components.hh` header (only for macOS). This fixes all of the gz-sim tests mentioned in #2204 locally.

## Checklist
- [ ] Signed all commits for [... cut]
closed 2023-11-10 by azeey