gazebosim/gz-sim #2206: Python system path installation wrong directory

oysstu, 2023-10-12:
Enabling the USE_SYSTEM_PATHS_FOR_PYTHON_INSTALLATION option installs the python libraries in a folder called gz in the system root. The other libraries, such as gz-msgs and gz-math, installs their libraries in the correct location using this option.

I suppose the following part is the problem
https://github.com/gazebosim/gz-sim/blob/ab0669c87b14faedec69e9f768ad5aea32e8b744/python/CMakeLists.txt#L9-L32

The equivalent part in gz-math. I guess the else statement in USE_DIST_PACKAGES_FOR_PYTHON might be missing?
https://github.com/gazebosim/gz-math/blob/a527e1422c25df73e30ee4f0c38ccbe56f6da4e4/src/python_pybind11/CMakeLists.txt#L75-L100

azeey, 2023-11-10:
Hi @oysstu, thanks for filing an issue! Since you've already identified what needs to be done, would you mind creating a PR?

oysstu, 2024-01-02:
> Hi @oysstu, thanks for filing an issue! Since you've already identified what needs to be done, would you mind creating a PR?

Opened in #2274. With this, python libraries are correctly installed to `/usr/lib/python3.XX/site-packages/gz/` instead of `/gz`.

PR #2274 (2024-01-02): Install the Python libs to system site-packages instead of root

# 🦟 Bug fix

Fixes #2206

## Summary
Changes are copied from gz-math

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a review on your [... cut]
closed 2024-01-02 by azeey