gazebosim/gz-sim #2213: Compiling on Debian Bookworm

tmayoff, 2023-10-17:
## Environment
* OS Version: Debian Bookworm
* Source or binary build?
  

## Description
* Expected behavior: Build completes
* Actual behavior: Build fails

## Steps to reproduce

1. On Debian Bookworm follow install instructions from here, https://gazebosim.org/docs/fortress/install_ubuntu_src

I'm aware those instructions are for Ubuntu however, our system is using debian bookworm and I've been able to get most everything else ROS2 related to compile this is one of the last things.

Is there any way to get this to work on bookworm. 

These are the packages I have installed related to ignition

```
ignition-common-cli/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
ignition-transport-cli/stable,now 11.0.0+ds-4+b1 amd64 [installed,automatic]
libignition-cmake-dev/stable,stable,now 2.11.0-1 all [installed]
libignition-common-av-dev/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-common-core-dev/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-common-dev/stable,now 4.5.1+ds-2+b1 amd64 [installed]
libignition-common-graphics-dev/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-common4-4/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-common4-av4/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-common4-events4/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-common4-graphics4/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-common4-profiler4/stable,now 4.5.1+ds-2+b1 amd64 [installed,automatic]
libignition-math-dev/stable,now 6.10.0+ds3-7+b4 amd64 [installed]
libignition-math6-6/stable,now 6.10.0+ds3-7+b4 amd64 [installed,automatic]
libignition-msgs-dev/stable,now 8.2.0+ds-1+b1 amd64 [installed,automatic]
libignition-msgs8-8/stable,now 8.2.0+ds-1+b1 amd64 [installed,automatic]
libignition-transport-dev/stable,now 11.0.0+ds-4+b1 amd64 [installed]
libignition-transport11-11/stable,now 11.0.0+ds-4+b1 amd64 [installed,automatic]
libignition-utils-dev/stable,now 1.2.0+ds-1 amd64 [installed,automatic]

```

## Output
```
 colcon build --packages-skip-build-finished --cmake-args -DBUILD_TESTING=OFF --merge-install
/usr/lib/python3/dist-packages/colcon_devtools/verb/extension_points.py:4: UserWarning: 'colcon_core.entry_point' has been deprecated, use 'colcon_core.extension_point' instead
  from colcon_core.entry_point import EXTENSION_POINT_GROUP_NAME
Starting >>> ignition-rendering6
--- stderr: ignition-rendering6                           
CMake Warning at /usr/share/cmake-3.25/Modules/CMakeFindDependencyMacro.cmake:47 (find_package):
  By not providing "Findignition-common4-graphics.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "ignition-common4-graphics", but CMake did not find one.

  Could not find a package configuration file provided by
  "ignition-common4-graphics" (requested version 4.7.0) with any of the
  following names:

    ignition-common4-graphicsConfig.cmake
    ignition-common4-graphics-config.cmake

  Add the installation prefix of "ignition-common4-graphics" to
  CMAKE_PREFIX_PATH or set "ignition-common4-graphics_DIR" to a directory
  containing one of the above files.  If "ignition-common4-graphics" provides
  a separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  /home/tyler/src/module-ROS2/ws_ignition/install/lib/cmake/ignition-common4/ignition-common4-config.cmake:187 (find_dependency)
  /home/tyler/src/module-ROS2/ws_ignition/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:197 (find_package)
  CMakeLists.txt:41 (ign_find_package)

CMake Warning at /home/tyler/src/module-ROS2/ws_ignition/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:197 (find_package):
  Found package configuration file:

    /home/tyler/src/module-ROS2/ws_ignition/install/lib/cmake/ignition-common4/ignition-common4-config.cmake

  but it set ignition-common4_FOUND to FALSE so package "ignition-common4" is
  considered to be NOT FOUND.  Reason given by package:

  ignition-common4 could not be found because dependency
  ignition-common4-graphics could not be found.

Call Stack (most recent call first):
  CMakeLists.txt:41 (ign_find_package)

CMake Warning at /home/tyler/src/module-ROS2/ws_ignition/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:55 (message):
   CONFIGURATION WARNINGS:
   -- Skipping component [optix]: Missing dependency [OptiX].
      ^~~~~ Set SKIP_optix=true in cmake to suppress this warning.
   
Call Stack (most recent call first):
  CMakeLists.txt:173 (ign_configure_build)

CMake Error at /home/tyler/src/module-ROS2/ws_ignition/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:59 (message):
  -- BUILD ERRORS: These must be resolved before compiling.
Call Stack (most recent call first):
  CMakeLists.txt:173 (ign_configure_build)

CMake Error at /home/tyler/src/module-ROS2/ws_ignition/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:61 (message):
  -- 	Missing dependency [ignition-common4] (Components: graphics, events)
Call Stack (most recent call first):
  CMakeLists.txt:173 (ign_configure_build)

CMake Error at /home/tyler/src/module-ROS2/ws_ignition/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:63 (message):
  -- END BUILD ERRORS

Call Stack (most recent call first):
  CMakeLists.txt:173 (ign_configure_build)

CMake Error at /home/tyler/src/module-ROS2/ws_ignition/install/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:68 (message):
  Errors encountered in build.  Please see BUILD ERRORS above.
Call Stack (most recent call first):
  CMakeLists.txt:173 (ign_configure_build)

---
Failed   <<< ignition-rendering6 [12.8s, exited with code 1]

Summary: 0 packages finished [14.0s]
  1 package failed: ignition-rendering6
  1 package had stderr output: ignition-rendering6
  4 packages not processed
```

azeey, 2023-10-17:
If you're building from source, I would recommend removing all `ignition` packages you installed from debs. There might be version mismatches that cause build failures that are hard to debug.

tmayoff, 2023-10-18:
Alright that was it, I guess I installed 1 package that caused it to fail so I tried installing more thinking it would fix it. Removing them all let me compile it

closed 2023-10-18 by tmayoff

azeey, 2023-10-18:
Yeah, the clue was
```
 "ignition-common4-graphics" (requested version 4.7.0) with any of the
  following names:
```

but I wish it would print out  the version it found even if it's not the requested version.