gazebosim/gz-sim #2215: Error: The maximum number of open files on this system has been reached. Use `ulimit -n` to increase this limit.

Kakcalu13, 2023-10-21:
## Environment
* OS Version: Mac M2 Ventura
* Source or binary build?
Harmonic 
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
bwuk@KOs-Mac-mini ~ % system_profiler SPDisplaysDataType
Graphics/Displays:

    Apple M2 Pro:

      Chipset Model: Apple M2 Pro
      Type: GPU
      Bus: Built-In
      Total Number of Cores: 16
      Vendor: Apple (0x106b)
      Metal Support: Metal 3
      Displays:
        VN279:
          Resolution: 1920 x 1080 (1080p FHD - Full High Definition)
          UI Looks like: 1920 x 1080 @ 50.00Hz
          Main Display: Yes
          Mirror: Off
          Online: Yes
          Rotation: Supported
        Beyond TV:
          Resolution: 3840 x 2160 (2160p/4K UHD 1 - Ultra High Definition)
          UI Looks like: 1920 x 1080 @ 60.00Hz
          Mirror: Off
          Online: Yes
          Rotation: Supported
          Television: Yes

bwuk@KOs-Mac-mini ~ % 

<details>

Here is what I did step by step
```
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
brew tap osrf/simulation
brew install gz-harmonic
```

It looks like promising until I get error:
```
Error: The maximum number of open files on this system has been reached. Use `ulimit -n` to increase this limit.
```

</details>

I just want to install gazebo in m2, thats all I want. I literally bought m2 for gazebo lol I never had any good luck with mac in past. I really hope this is simplest fix

Kakcalu13, 2023-10-21:
I ran this command without any prior experience or knowledge:
ulimit -n 1000000 

and error shows the same thing

azeey, 2023-10-23:
@scpeters I've also ran into this in the past, but wasn't sure it was just on my machine. Do you know what might be going on?

scpeters, 2023-10-25:
yes I ran into this same thing, and opened https://github.com/Homebrew/brew/pull/14879 to try to improve the error message. On my mac machines `ulimit -n` returns 256, but if I increase it to 1024, it seems to work. I'm not sure why the default is so low

Kakcalu13, 2023-10-25:
At this point, it's not gazebo issue. It's the M2 issue with brew. I will do some research after work. 

So the problem is that ulimit doesn't affect homebrew at all. There are supposed to be few files where you can edit the file such as `/proc/$$/limits` which is not available in m2. M2 doesn't have `/proc/` 

I will close the issue with a solution once I found the fix

Kakcalu13, 2023-10-27:
Sounds like this is Gazebo's end. @azeey @scpeters according to the issue I opened in brew

Steve, what kind of mac do you have?

scpeters, 2023-11-04:
my main development machine has an Intel cpu and its default `ulimit -n` is 256. I often have to increase it to 1024 to install gz-sim

~~~
CPU: dodeca-core 64-bit kabylake
macOS: 13.5.2-x86_64
~~~

I also have access to an m1 machine that I don't use for development very often, though I have used it to debug issues with gazebo on m1. It also has a default `ulimit -n` of 256

~~~
CPU: octa-core 64-bit arm_firestorm_icestorm
macOS: 12.7-arm64
~~~

Kakcalu13, 2023-11-05:
what's even happening?!?! NO WAY! GAZEBO IS FINALLY ON M2!!!! MY VERY FIRST GAZEBO ON MY M2!!! OMG

[image]

Thank you so much!!!

As for this issue, I just did `ulimit -n unlimited` and it worked. Not sure why..I wonder if you need to restart your pc after apply the command. This solved my problem! I hope it helps people with m2 too!

closed 2023-11-05 by Kakcalu13

scpeters, 2023-11-06:
I didn't have a chance to post the logs, but I ran `brew install gz-sim8 --verbose --debug` after getting the "maximunum number of open files" error, and I believe the stack trace came from some code in bullet trying to write a lock file. Perhaps, you had a lock file that wasn't deleted which was blocking all install attempts but was removed after your restart?

Anyway glad it's working!