gazebosim/gz-sim #2216: Creating ball joint between models

rteklewold, 2023-10-23:
Hello 
I would like to know if it is possible to create a ball joint  between two models using ignition gazebo plugin. On gazebo classic , it is possible to connect two models by creating a joint during runtime by adding such a code in a plugin:

` physics::JointPtr j2; 
  j2 = world->Physics()->CreateJoint("ball", d1);
  j2->Attach(l2, lb);
  j2->Load(l2, lb, pose_joint);
  j2->SetModel(cable);
  j2->Init();`

Is there an equivalent feature on ignition gazebo? I have tried detachable joint but using that makes my drone unstable for some reason. I am trying to make a simulation of aerial cable towed system and I need to connect drones with links of another model using a ball joint.

Thank you

azeey, 2023-10-23:
Currently, it is only possible to create "fixed" joints dynamically using detachable joints.

rteklewold, 2023-10-24:
Thank you for your reply. Is there a plan to have this kind of feature soon on ignition gazebo? 
Am I maybe not respecting some constraints when I add a fixed joint between the `base_link`and the `drone_attach` described in this world file?
[https://gitlab.univ-nantes.fr/ls2n-drones/ls2n_drone_armada/-/blob/main/worlds/armada_world.sdf](url)

Link towards the model armada:
[https://gitlab.univ-nantes.fr/ls2n-drones/ls2n_drone_armada/-/blob/main/models/armada/armada.erb](url)

Link towards the model crazy2fly drone:
[https://gitlab.univ-nantes.fr/ls2n-drones/ls2n_drone_sitl/-/blob/master/sitl_px4_simulation/models/crazy2fly/model.sdf](url)

The behaviour is that changing the type of joint in world  from  `fixed`  to `ball` type makes the drone fly as expected. But using fixed joint causes the drone to be unstable and go towards undesired direction. I would like to know why fixed joint causes problem.

Thanks in advance

azeey, 2023-11-13:
That's currently not in the roadmap. We would accept PRs though :) 

> Am I maybe not respecting some constraints when I add a fixed joint between the base_linkand the drone_attach described in this world file?
https://gitlab.univ-nantes.fr/ls2n-drones/ls2n_drone_armada/-/blob/main/worlds/armada_world.sdf

It's difficult to tell. One thing I noticed is you have a link with a mass of `0.001 Kg` as the attachment link between the drones and the armada. Having small masses connected to large masses usually causes instabilities in simulation. Can you use a `<frame>` instead?  Also, Is it possible to demonstrate the problem with a simple example so it's easy to tell where the problem might be?

vbalaji21, 2024-03-13:
@rteklewold I am trying the similar thing on a different application of aerial cables. Did you solve your issue ? Can you let me know how did you solve your problem ?